espressif/esp-idf #11592: python esp-idf-monitor package not installed by install.sh script

lboue, 2023-06-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1-rc1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

pip install esp-idf-monitor

### What is the actual behavior?

I had to install manually install  esp-idf-monitor
`pip install esp-idf-monitor
`

### Steps to reproduce.

```
git clone -b v5.1-rc1 --recursive https://github.com/espressif/esp-idf.git esp-idf-v5.1-rc1
cd ~/esp-idf-v5.1-rc1/
./install.sh
. ./export.sh
```

wqx6, 2023-06-06:
Currently, the suggested IDF version for connectedhomeip repo is v4.4.4, the `activate.sh` will enable the python environment created by Matter repo. But there are some python packages missed in that python environment, which causes this issue. If you are using v5.x IDF for Matter examples, you can use [esp-matter](https://github.com/espressif/esp-matter) sdk.

lboue, 2023-06-07:
Thanks. I will try [esp-matter](https://github.com/espressif/esp-matter).

closed 2023-06-07 by lboue