espressif/esp-idf #11593: Building connectedhomeip/examples/bridge-app: error => Failed to resolve component 'mdns'

lboue, 2023-06-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1-rc1

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Building connectedhomeip bridge-app examples without error.

### What is the actual behavior?

 error => Failed to resolve component 'mdns'

### Steps to reproduce.

```
git clone https://github.com/project-chip/connectedhomeip.git
cd ~/connectedhomeip
source scripts/bootstrap.sh
source scripts/activate.sh

cd ~/connectedhomeip/examples/bridge-app/esp32
gn gen out/debug
ninja -C out/debug
idf.py set-target esp32c3
idf.py build
```

### Build or installation Logs.

```plain
Building connectedhomeip/examples/bridge-app: error => Failed to resolve component 'mdns'

user@matter-vm:~/connectedhomeip/examples/bridge-app/esp32$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/user/connectedhomeip/examples/bridge-app/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/user/connectedhomeip/.environment/pigweed-venv/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/user/connectedhomeip/examples/bridge-app/esp32"...
-- IDF_TARGET is not set, guessed 'esp32c3' from sdkconfig '/home/user/connectedhomeip/examples/bridge-app/esp32/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.34.1")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/user/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/user/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
CMake Warning at /home/user/esp-idf-v5.1-rc1/tools/cmake/build.cmake:565 (message):
  "idf_component.yml" file was found for components:

        /home/user/connectedhomeip/examples/bridge-app/esp32/third_party/connectedhomeip/config/esp32/components/chip

  However, the component manager is not enabled.
Call Stack (most recent call first):
  /home/user/esp-idf-v5.1-rc1/tools/cmake/project.cmake:464 (idf_build_process)
  CMakeLists.txt:31 (project)

-- Found Python3: /home/user/connectedhomeip/.environment/pigweed-venv/bin/python3.10 (found version "3.10.6") found components: Interpreter
CMake Error at /home/user/esp-idf-v5.1-rc1/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'mdns'.
Call Stack (most recent call first):
  /home/user/esp-idf-v5.1-rc1/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /home/user/esp-idf-v5.1-rc1/tools/cmake/build.cmake:309 (__build_expand_requirements)
  /home/user/esp-idf-v5.1-rc1/tools/cmake/build.cmake:595 (__build_expand_requirements)
  /home/user/esp-idf-v5.1-rc1/tools/cmake/project.cmake:464 (idf_build_process)
  CMakeLists.txt:31 (project)

-- Configuring incomplete, errors occurred!
See also "/home/user/connectedhomeip/examples/bridge-app/esp32/build/CMakeFiles/CMakeOutput.log".
HINT: The component mdns could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.
cmake failed with exit code 1, output of the command is in the /home/user/connectedhomeip/examples/bridge-app/esp32/build/log/idf_py_stderr_output_51215 and /home/user/connectedhomeip/examples/bridge-app/esp32/build/log/idf_py_stdout_output_51215
```

Could you hep me to fix that issue?
```

wqx6, 2023-06-06:
It could be caused by that the python package `idf_component_manager` is not installed in the matter python environment. You can install it and try again.

lboue, 2023-06-07:
@wqx6 
Isn't it possible to add this dependency in the install script to avoid this error?

chshu, 2025-02-26:
`idf_component_manager` should be part of IDF env, so don't need to be added to matter script.

closed 2025-02-26 by chshu