espressif/esp-idf #11595: esp_wifi_set_protocol does not work with WIFI_IF_STA interface for WIFI_PROTOCOL_LR mode

CmosC1CA, 2023-06-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.2

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM

### Power Supply used.

USB

### What is the expected behavior?

**esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR);**

Using this line, the WIFI_IF_STA should switch to LR mode.

### What is the actual behavior?

**esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR);**
Should set the WIFI_IF_STA interface to long range mode, but it does not seem to do it. It completes without any error, but the interface is not in long range mode.

If we try to switch the rate to an LR only rate, the program will fail, and an error is logged in the terminal.
**esp_wifi_config_espnow_rate(WIFI_IF_STA, WIFI_PHY_RATE_LORA_500K);**

W (719) wifi:invalid rate, need change phy mode to LR

### Steps to reproduce.

#include <esp_wifi.h>
#include <nvs_flash.h>
#include <esp_now.h>

void app_main(void)
{
	esp_err_t status = nvs_flash_init();
	if (status == ESP_ERR_NVS_NO_FREE_PAGES || status == ESP_ERR_NVS_NEW_VERSION_FOUND)
	{
		ESP_ERROR_CHECK(nvs_flash_erase());
		ESP_ERROR_CHECK(nvs_flash_init());
	}
	ESP_ERROR_CHECK(esp_event_loop_create_default());

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

	ESP_ERROR_CHECK(esp_wifi_start());
	ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR));
	ESP_ERROR_CHECK(esp_wifi_config_espnow_rate(WIFI_IF_STA, WIFI_PHY_RATE_LORA_500K));
}

This will crash at  esp_wifi_config_espnow_rate(ESP_IF_WIFI_STA, WIFI_PHY_RATE_LORA_500K)

### Debug Logs.

```plain
I (0) cpu_start: App cpu up.
I (360) cpu_start: Pro cpu start user code
I (360) cpu_start: cpu freq: 160000000 Hz
I (360) cpu_start: Application information:
I (365) cpu_start: Project name:     hello_world
I (370) cpu_start: App version:      1
I (374) cpu_start: Compile time:     Jun  5 2023 13:34:15
I (380) cpu_start: ELF file SHA256:  223fae429c282928...
I (386) cpu_start: ESP-IDF:          v5.0.2-dirty
I (392) cpu_start: Min chip rev:     v0.0
I (397) cpu_start: Max chip rev:     v3.99 
I (401) cpu_start: Chip rev:         v1.0
I (406) heap_init: Initializing. RAM available for dynamic allocation:
I (413) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (419) heap_init: At 3FFB6590 len 00029A70 (166 KiB): DRAM
I (426) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (432) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (438) heap_init: At 40094C3C len 0000B3C4 (44 KiB): IRAM
I (446) spi_flash: detected chip: generic
I (449) spi_flash: flash io: dio
I (454) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (519) wifi:wifi driver task: 3ffbcdf0, prio:23, stack:6656, core=0
I (519) system_api: Base MAC address is not set
I (519) system_api: read default base MAC address from EFUSE
I (529) wifi:wifi firmware version: 57982fe
I (529) wifi:wifi certification version: v7.0
I (529) wifi:config NVS flash: enabled
I (529) wifi:config nano formating: disabled
I (539) wifi:Init data frame dynamic rx buffer num: 32
I (539) wifi:Init management frame dynamic rx buffer num: 32
I (549) wifi:Init management short buffer num: 32
I (549) wifi:Init dynamic tx buffer num: 32
I (559) wifi:Init static rx buffer size: 1600
I (559) wifi:Init static rx buffer num: 10
I (559) wifi:Init dynamic rx buffer num: 32
I (569) wifi_init: rx ba win: 6
I (569) wifi_init: tcpip mbox: 32
I (579) wifi_init: udp mbox: 6
I (579) wifi_init: tcp mbox: 6
I (579) wifi_init: tcp tx win: 5744
I (589) wifi_init: tcp rx win: 5744
I (589) wifi_init: tcp mss: 1440
I (599) wifi_init: WiFi IRAM OP enabled
I (599) wifi_init: WiFi RX IRAM OP enabled
I (609) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (719) wifi:mode : sta (78:e3:6d:0b:03:ec)
I (719) wifi:enable tsf
I (719) wifi:enable tsf
W (719) wifi:invalid rate, need change phy mode to LR
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400d6472
```

kriegste, 2023-06-05:
Could this somehow be related to
https://github.com/espressif/esp-idf/issues/9978
?

CmosC1CA, 2023-06-05:
I have no idea. I saw your issue, but after I tried to find the source for the wifi driver, I realized that it is not open source. So we can only guess what is the problem. Strangely enough with AP mode it just works fine.

CmosC1CA, 2023-06-28:
Found some undocumented workaround.
#11751

zhangyanjiaoesp, 2024-03-05:
@CmosC1CA the  https://github.com/espressif/esp-idf/commit/554e6880bce81817a654876ed1b09c2baa9bbdc0 commit seems to have solved this problem, please update your IDF version and  have a try.

Sherry616, 2024-04-24:
Hi @CmosC1CA, could you please share your updates on this issue? Thanks.

closed 2024-06-07 by Sherry616