espressif/esp-idf #11596: ImportError: cannot import name 'rest_of_line' from 'pyparsing'

nidefawl, 2023-06-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-547-g20e8e86c80

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

No compile error

### What is the actual behavior?

While trying to build the Sming Basic_Blink example on fresh Ubuntu 2204 installation using IDF 5.0 I get following error:
```
FAILED: esp-idf/esp_system/ld/sections.ld /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system/ld/sections.ld
cd /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system && /usr/bin/python /opt/esp-idf-5.0/tools/ldgen/ldgen.py --config /home/michael/esp-dev/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug/sdkconfig --fragments-list "/opt/esp-idf-5.0/components/xtensa/linker.lf;/opt/esp-idf-5.0/components/esp_ringbuf/linker.lf;/opt/esp-idf-5.0/components/driver/linker.lf;/opt/esp-idf-5.0/components/esp_pm/linker.lf;/opt/esp-idf-5.0/components/spi_flash/linker.lf;/opt/esp-idf-5.0/components/esp_system/linker.lf;/opt/esp-idf-5.0/components/esp_system/app.lf;/opt/esp-idf-5.0/components/esp_common/common.lf;/opt/esp-idf-5.0/components/esp_common/soc.lf;/opt/esp-idf-5.0/components/esp_rom/linker.lf;/opt/esp-idf-5.0/components/hal/linker.lf;/opt/esp-idf-5.0/components/log/linker.lf;/opt/esp-idf-5.0/components/heap/linker.lf;/opt/esp-idf-5.0/components/soc/linker.lf;/opt/esp-idf-5.0/components/esp_hw_support/linker.lf;/opt/esp-idf-5.0/components/freertos/linker.lf;/opt/esp-idf-5.0/components/newlib/newlib.lf;/opt/esp-idf-5.0/components/newlib/system_libs.lf;/opt/esp-idf-5.0/components/esp_adc/linker.lf;/opt/esp-idf-5.0/components/esp_event/linker.lf;/opt/esp-idf-5.0/components/esp_gdbstub/linker.lf;/opt/esp-idf-5.0/components/lwip/linker.lf;/opt/esp-idf-5.0/components/esp_netif/linker.lf;/opt/esp-idf-5.0/components/esp_wifi/linker.lf;/opt/esp-idf-5.0/components/esp_phy/linker.lf;/opt/esp-idf-5.0/components/espcoredump/linker.lf" --input /opt/esp-idf-5.0/components/esp_system/ld/esp32/sections.ld.in --output /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system/ld/sections.ld --kconfig /opt/esp-idf-5.0/Kconfig --env-file /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/config.env --libraries-file /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/ldgen_libraries --objdump /opt/esp32/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump
Traceback (most recent call last):
  File "/opt/esp-idf-5.0/tools/ldgen/ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "/opt/esp-idf-5.0/tools/ldgen/ldgen/entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/usr/lib/python3/dist-packages/pyparsing.py)
ninja: build stopped: subcommand failed.
make: *** [/home/michael/esp-dev/Sming/Sming/Arch/Esp32/Components/esp32/component.mk:372: checksdk] Error 1
```

I built stuff successfully before on Ubuntu 2204 using IDF version 4.4

### Steps to reproduce.

Try to build the Sming Basic_Blink example using IDF 5.0

### Build or installation Logs.

```plain
michael@vm2204:~/esp-dev/Sming/samples/Basic_Blink$ make clean

Basic_Blink: Invoking 'clean' for Esp32/esp32 (debug) architecture
Using configuration 'standard'
Cleaning application...
michael@vm2204:~/esp-dev/Sming/samples/Basic_Blink$ make

Basic_Blink: Invoking 'all' for Esp32/esp32 (debug) architecture
Using configuration 'standard'
Creating /home/michael/esp-dev/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug/sdkconfig.defaults
Executing action: reconfigure
Running cmake in directory /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/michael/esp-dev/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug"...
-- Found Git: /usr/bin/git (found version "2.34.1")
-- Initialising new submodule components/bootloader/subproject/components/micro-ecc/micro-ecc...
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Cloning into '/opt/esp-idf-5.0/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
-- Initialising new submodule components/bt/controller/lib_esp32...
[... 311 lines cut ...]
[95/101] Linking C static library esp-idf/micro-ecc/libmicro-ecc.a
[96/101] Linking C static library esp-idf/soc/libsoc.a
[97/101] Linking C static library esp-idf/xtensa/libxtensa.a
[98/101] Linking C static library esp-idf/main/libmain.a
[99/101] Linking C executable bootloader.elf
[100/101] Generating binary image from built executable
esptool.py v4.6.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/bootloader.bin
[101/101] cd /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/esp-idf/esptool_py && /usr/bin/python /opt/esp-idf-5.0/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/bootloader/bootloader.bin
Bootloader binary size 0x6710 bytes. 0x8f0 bytes (8%) free.
[668/672] Generating ld/sections.ld
FAILED: esp-idf/esp_system/ld/sections.ld /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system/ld/sections.ld
cd /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system && /usr/bin/python /opt/esp-idf-5.0/tools/ldgen/ldgen.py --config /home/michael/esp-dev/Sming/Sming/Arch/Esp32/Components/esp32/project/esp32/debug/sdkconfig --fragments-list "/opt/esp-idf-5.0/components/xtensa/linker.lf;/opt/esp-idf-5.0/components/esp_ringbuf/linker.lf;/opt/esp-idf-5.0/components/driver/linker.lf;/opt/esp-idf-5.0/components/esp_pm/linker.lf;/opt/esp-idf-5.0/components/spi_flash/linker.lf;/opt/esp-idf-5.0/components/esp_system/linker.lf;/opt/esp-idf-5.0/components/esp_system/app.lf;/opt/esp-idf-5.0/components/esp_common/common.lf;/opt/esp-idf-5.0/components/esp_common/soc.lf;/opt/esp-idf-5.0/components/esp_rom/linker.lf;/opt/esp-idf-5.0/components/hal/linker.lf;/opt/esp-idf-5.0/components/log/linker.lf;/opt/esp-idf-5.0/components/heap/linker.lf;/opt/esp-idf-5.0/components/soc/linker.lf;/opt/esp-idf-5.0/components/esp_hw_support/linker.lf;/opt/esp-idf-5.0/components/freertos/linker.lf;/opt/esp-idf-5.0/components/newlib/newlib.lf;/opt/esp-idf-5.0/components/newlib/system_libs.lf;/opt/esp-idf-5.0/components/esp_adc/linker.lf;/opt/esp-idf-5.0/components/esp_event/linker.lf;/opt/esp-idf-5.0/components/esp_gdbstub/linker.lf;/opt/esp-idf-5.0/components/lwip/linker.lf;/opt/esp-idf-5.0/components/esp_netif/linker.lf;/opt/esp-idf-5.0/components/esp_wifi/linker.lf;/opt/esp-idf-5.0/components/esp_phy/linker.lf;/opt/esp-idf-5.0/components/espcoredump/linker.lf" --input /opt/esp-idf-5.0/components/esp_system/ld/esp32/sections.ld.in --output /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/esp-idf/esp_system/ld/sections.ld --kconfig /opt/esp-idf-5.0/Kconfig --env-file /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/config.env --libraries-file /home/michael/esp-dev/Sming/Sming/out/Esp32/esp32/debug/build/esp32/sdk/ldgen_libraries --objdump /opt/esp32/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump
Traceback (most recent call last):
  File "/opt/esp-idf-5.0/tools/ldgen/ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "/opt/esp-idf-5.0/tools/ldgen/ldgen/entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/usr/lib/python3/dist-packages/pyparsing.py)
ninja: build stopped: subcommand failed.
make: *** [/home/michael/esp-dev/Sming/Sming/Arch/Esp32/Components/esp32/component.mk:372: checksdk] Error 1
```

nidefawl, 2023-06-05:
I fixed it with
```
pip install pyparsing --upgrade
```

dobairoland, 2023-06-06:
```
/usr/bin/python /opt/esp-idf-5.0/tools/ldgen/ldgen.py
```
Looking at the above I judge that you are not using the supported way of setting up ESP-IDF environments where a separate virtual environment is used for each ESP-IDF release.  In your scenario, of course, you must manage package versions by yourself.

For the supported way of setting up the environment, our scripts ensure that `pyparsing>=2.0.3,<2.4.0` is used for ESP-IDF v4.4.* and `pyparsing>=3.0.3,<3.1` for ESP-IDF v5.0.*.

closed 2023-06-06 by dobairoland