espressif/esp-idf #11597: [v5.1] linking binary files is broken aka extern const unsigned char startAsm[] asm("_binary_warn_png_start");

chipweinberger, 2023-06-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

Linking multiple binary files in the same source file should work. They should each return a different assembly address.

### What is the actual behavior?

All assembly address are the same. 

```
            extern const unsigned char startAsm[] asm("_binary_jAssets_js_start"); // 0x3c1badce
            extern const unsigned char endAsm[]   asm("_binary_jAssets_js_end"); // 0x3c1bb591

            extern const unsigned char startAsm[] asm("_binary_style_css_start"); // 0x3c1badce
            extern const unsigned char endAsm[]   asm("_binary_style_css_end"); // 0x3c1bb591
```

### Steps to reproduce.

```
void pd_api_webapp_wildcard(pd_xtrans_t* xtrans, const char* uri, pd_error_t* error)
{
    pd_err_unset(error);

    {
        const unsigned char* start = NULL;
        const unsigned char* end = NULL;

        // allow caching
        eTry(,pd_xtrans_resp_set_hdr,(xtrans, "Cache-Control", "public, max-age=31536000", error));

        if (strstr(uri, "/webapp/css/style.css") != NULL){
            extern const unsigned char startAsm[] asm("_binary_style_css_start");
            extern const unsigned char endAsm[]   asm("_binary_style_css_end");
            start = startAsm;
            end = endAsm;
            ESP_LOGI(TAG, "getting %s /webapp/css/style.css startAsm %p endAsm %p", uri, startAsm, endAsm);
        } 

        if (strstr(uri, "/webapp/js/jAssets.js") != NULL){
            extern const unsigned char startAsm[] asm("_binary_jAssets_js_start");
            extern const unsigned char endAsm[]   asm("_binary_jAssets_js_end");
            start = startAsm;
            end = endAsm;
            ESP_LOGI(TAG, "getting %s /webapp/js/jAssets.js startAsm %p endAsm %p", uri, startAsm, endAsm);
        }

       ..... about ~20 more handlers just like this .....

       
        const size_t size = end - start;

        // found the file we wanted?
        if (!start || !end) {
            eFail("could not find %s", uri);
            return;
        }

        if (xtrans->which == PD_XTRANS_HTTP){
            pd_http_set_mime_from_uri(xtrans->to.httpReq, uri, true);
        }
        
        // send
        eTry(,pd_xtrans_resp_send,(xtrans, start, size, error));

        pd_err_success(error);
    }
end:
    return;
}
```

### Debug Logs.

```plain
I (14621) pd api webapp: getting /webapp/js/jAssets.js?v=6a64ebfff1dc /webapp/js/jAssets.js startAsm 0x3c1badce endAsm 0x3c1bb591
I (14361) pd api webapp: getting /webapp/css/style.css?v=6a64ebfff1dc /webapp/css/style.css startAsm 0x3c1badce endAsm 0x3c1bb591
```

chipweinberger, 2023-06-06:
this worked on v4.4

chipweinberger, 2023-06-06:
I was able to fix it by giving them each a unique name. This was **not** required on v4.4.

            extern const unsigned char startAsm1[] asm("_binary_style_css_start");
            extern const unsigned char endAsm1[]   asm("_binary_style_css_end");

            extern const unsigned char startAsm2[] asm("_binary_style_css_start");
            extern const unsigned char endAsm2[]   asm("_binary_style_css_end");

etc

o-marshmallow, 2023-06-07:
Hello @chipweinberger ,

The compiler is explicit about what is happening, when I try on my side, I get the warning:
```
warning: 'asm' declaration ignored due to conflict with previous rename [-Wpragmas]
   29 |         extern const unsigned char startAsm[] asm("_binary_jAssets_js_start");
      |         ^~~~~~
warning: 'asm' declaration ignored due to conflict with previous rename [-Wpragmas]
   30 |         extern const unsigned char endAsm[]   asm("_binary_jAssets_js_end");
      |         ^~~~~~
```
So the compiler will simply ignore the redefinition of the global symbol `startAsm`. Moreover, this seems like an undefined behavior to me in the sense that it is highly dependent on the compiler and linker, which would explain why it may work with other compiler versions (v4.4's) and not now.

Anyway, you don't need to use `asm` here, C can already reference any assembly symbol thanks to the linker, so the proper way to access your files is:
```
        if (strstr(uri, "/webapp/css/style.css") != NULL){
            extern uint8_t _binary_style_css_start;
            extern uint8_t _binary_style_css_end;
            start = &_binary_style_css_start;
            end = &_binary_style_css_end;
            ESP_LOGI(TAG, "getting %s /webapp/css/style.css startAsm %p endAsm %p", uri, startAsm, endAsm);
        } 

        if (strstr(uri, "/webapp/js/jAssets.js") != NULL){
            extern uint8_t _binary_jAssets_js_start;
            extern uint8_t _binary_jAssets_js_end;
            start = &_binary_jAssets_js_start;
            end = &_binary_jAssets_js_end;
            ESP_LOGI(TAG, "getting %s /webapp/js/jAssets.js startAsm %p endAsm %p", uri, startAsm, endAsm);
        }
```
Make sure that in your assembly files, both `*_start` and `*_end` are global:
```
    .globl _binary_jAssets_js_start
    .globl _binary_jAssets_js_end
    .globl _binary_style_css_start
    .globl _binary_style_css_end
```

chipweinberger, 2023-06-07:
Thanks for the very detailed response @o-marshmallow !

I may have missed the warning somehow! As long as this is expected behavior, then that is totally fine. I just wanted to make sure this was not an unexpected regression. Thanks again.

closed 2023-06-07 by chipweinberger

o-marshmallow, 2023-06-07:
My pleasure, I hope the previous example works for you