espressif/esp-idf #11598: Wrong logic of checking CONFIG_SPIRAM_USE_MALLOC in i2c_cmd_allocate

kylezhou, 2023-06-06:
In i2c.c:i2c_cmd_allocate() function:
`
#if !CONFIG_SPIRAM_USE_MALLOC
        *outptr = calloc(n, size);
#else
        *outptr = heap_caps_calloc(n, size, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
#endif
`
This seems to be the other way around.

suda-morris, 2023-06-06:
I think the logic there is correct. The driver wants to always malloc the command list from internal SRAM. If `CONFIG_SPIRAM_USE_MALLOC` is on, the `malloc` function has the possibility to allocate a memory block from the PSRAM.

kylezhou, 2023-06-06:
Thanks suda-morris for prompt reply. But I have the following two questions:

- What is the reason that the driver wants to always malloc from internal SRAM?
- When CONFIG_SPIRAM_USE_MALLOC is off, why calloc() is used?

suda-morris, 2023-06-06:
> When CONFIG_SPIRAM_USE_MALLOC is off, why calloc() is used?

because in that condition, the calloc can only allocate mamroy from SRAM, which is safe.

> What is the reason that the driver wants to always malloc from internal SRAM?

Because when the flash cache is disabled, we still want the i2c driver to work. A memory allocated in the PSRAM is not accessible by the CPU when the cache is disabled.

kylezhou, 2023-06-06:
Thanks for the explanation. Then why we don't just call: heap_caps_calloc(n, size, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT); always?

closed 2023-08-15 by suda-morris