espressif/esp-idf #11601: [help] how to add a component in the current project

fatsheep2020, 2023-06-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

have read programming guide, but not clearly got it, my question:
for current project (such as BLINK), how to add a component into it (for example wi-fi)?
environment: VSCode+IDF plugin, windows
THanks!

Alvin1Zhang, 2023-06-07:
Thanks for reporting, would you please help provide more details per the issue template https://github.com/espressif/esp-idf/issues/new/choose? Thanks.

fatsheep2020, 2023-06-08:
actually this is not report a bug, just want to know how to use IDF framework. my question is how to add a component (for example WI-FI) into an existing project?  by menuconfig or manually code it, what's the process
(怎样给一个已有的项目添加组件，具体是怎样的流程，是通过手写还是配置来实现的？）

igrr, 2024-05-10:
@fatsheep2020 Most of the time, you can add additional functionality the following way:

1. Add the component where this functionality is implemented to the dependencies list (`REQUIRES` or `PRIV_REQUIRES`) in your component's CMakeLists.txt
2. Include the relevant header file
3. Call APIs to initialize the feature you need

Sometimes, you will need to enable certain features in menuconfig, such as for example PSRAM support. This will be mentioned in the relevant section of the IDF Programming Guide.

closed 2024-05-10 by igrr