espressif/esp-idf #11606: Esp32C2-26M esp_wifi_scan_get_ap_record can't get any AP when it awakened from deep-sleep.

Tsg333, 2023-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-C2 | Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I'm using esp_deep_sleep_start() and  esp_sleep_enable_timer_wakeup() to make the device enter deep-sleep.

When the device awakened from deep-sleep, it will use esp_wifi_scan_start() to scan APs. 

### What is the actual behavior?

When the device booted from POWERON, it can normally get the APs.

However, when it restarted from DSLEEP, it can't get any AP.

### Steps to reproduce.

```
void enter_sleep(uint32_t time_ms)
{
    esp_sleep_enable_timer_wakeup(time_ms * 1000);
    esp_deep_sleep_start();
}

void wifi_sta_init()
{
	ESP_ERROR_CHECK(esp_netif_init());
	ESP_ERROR_CHECK(esp_event_loop_create_default());
	wifiHandle = esp_netif_create_default_wifi_sta();

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));

	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	ESP_ERROR_CHECK(esp_wifi_start());
}

void wifi_scan()
{
       uint16_t sta_number = 0;
	wifi_ap_record_t ap_list[MAX_SCAN_LIST_SIZE];

	// 开始扫描
	esp_wifi_scan_start(NULL, true);

	esp_wifi_scan_get_ap_num(&sta_number);
	sta_number = sta_number > MAX_SCAN_LIST_SIZE ? MAX_SCAN_LIST_SIZE : sta_number;

	memset(ap_list, 0, sizeof(ap_list));
	memset(&m_scan_list, 0, sizeof(m_scan_list));
	if (esp_wifi_scan_get_ap_records(&sta_number, ap_list) == ESP_OK)
	{
		for (uint16_t i = 0; i < sta_number; i++)
		{
			
			if (strlen((char *)ap_list[i].ssid) >= MAX_SSID_LENGTH)
			{
				continue;
			}
			strcpy(m_scan_list.item[m_scan_list.num].ssid, (char *)ap_list[i].ssid);
			m_scan_list.item[m_scan_list.num].authmode = ap_list[i].authmode;
			m_scan_list.item[m_scan_list.num].rssi = ap_list[i].rssi;
			m_scan_list.item[m_scan_list.num].channel = ap_list[i].primary;
			m_scan_list.num++;
		}

		printf("sta_number:%d\r\n",m_scan_list.num);
	}
	return &m_scan_list;
}

void main()
{
     wifi_sta_init();
     wifi_scan();
     enter_sleep(5000);
}
```

Xiehanxin, 2023-07-06:
hi, @Tsg333  I can't run your code by just copying, and In your code I haven't seen the nvs init. so I write a simple example on base of the example/wifi/scan, and you can try this? I just add deepsleep and timer wake up in this example. 
[deep_sleep_scan.txt](https://github.com/espressif/esp-idf/files/11965280/deep_sleep_scan.txt)

closed 2024-02-01 by Tsg333

Sherry616, 2024-03-04:
Hi @Tsg333, do you still have this problem? Could you please share the latest update? Thanks.