espressif/esp-idf #11607: BT_BTC: btc_transfer_context No memory

Sky-Soo-Ap, 2023-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

1) Reboot firmware is unexpectable;
2) No memory leak;
3) BTC and BTU can drop packet if stack size is full;

### What is the actual behavior?

It seems memory leak and crashed then rebooting;
And, it seems BTC and BTU can NOT drop packets if stack size is full.
[BTC_no_memory.log](https://github.com/espressif/esp-idf/files/11672070/BTC_no_memory.log)
[sdkconfig.log](https://github.com/espressif/esp-idf/files/11672071/sdkconfig.log)

### Steps to reproduce.

1. ESP32 is GATT client;
2. Scanning and connected with GATT server;
3. Continuing receive notification packet and publish to mqtt broker; 

### Debug Logs.

```plain
W (29548) BT_BTC: btc_transfer_context No memory
E (29548) BT_BTC: btc_gattc_cback transfer failed
E (29628) BT_HCI: host_recv_pkt_cb couldn't aquire memory for inbound data buffer
assert failed: host_recv_pkt_cb hci_hal_h4.c:552 (0)
```

Sky-Soo-Ap, 2023-06-08:
What is the expected behavior?
1. Reboot firmware is unacceptable; // Just revised this line
2. No memory leak;
3. BTC and BTU can drop packet if stack size is full;

wmy-espressif, 2023-06-08:
HI @Sky-Soo-Ap  You have enabled Bluetooth and Wi-Fi  according to your menuconfig, the wireless protocol stack will consume much memory. I would like to ask the following questions:

1. Why do you think this issue is caused by memory leak? The issue can happen when there is really insufficient memory from heap. You can evaluate the free heap size using the following APIs in esp_system.h:
- esp_get_free_heap_size
- esp_get_minimum_free_heap_size
- esp_get_free_internal_heap_size
And have you got SPI-RAM on your hardware? If yes you can enable the psram options in menuconfig. Also, I provide some optimizations on sdkconfig options for Wi-Fi and Bluetooth protocol stack, to save some memory, you can try it out, use the options provided in the file to replace those in your sdkconfig file.
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/11687449/sdkconfig.zip)

2. As for the expected behaviour you mentioned, "BTC and BTU can drop packet if stack size is full", I am sorry that the Bluedroid protocol stack is unable to handle the exception that heap memory is exhausted, for most of the time. Bluedroid uses a lot of "osi_malloc" and "osi_free", the exeuction of it largely depend on the heap and message transfer from task to task. We don't know how one allocation failure can make some state goes wrong, so there are ususally assertions for memory acquisition in the Bluedroid stack. This is a structural and design issue for Bluedroid. So we can only ensure that the heap memory is sufficent to keep Bluedroid working fine.

Sky-Soo-Ap, 2023-06-09:
Hi @wmy-espressif,
Thank you for providing optimization configurations and detailed explanation.
I will try the optimization configurations.

I think this issue is caused by a memory leak because sometimes it can run about 30 minutes.

Sky-Soo-Ap, 2023-06-09:
Hi [wmy-espressif](https://github.com/wmy-espressif),
I encountered heap suddenly depleted while tracing the heap left as below.

I (12:40:32.422) azp_gatt_client: Received BM78xBT#6 device's notify, esp_get_free_heap_size:29728 bytes, esp_get_minimum_free_heap_size:18968 bytes, esp_get_free_internal_heap_size:29728 bytes.
E (2937870) BT_HCI: host_recv_pkt_cb couldn't aquire memory for inbound data buffer, pkt_size:172 esp_get_free_heap_size:1480 bytes, esp_get_minimum_free_heap_size:1304 bytes, esp_get_free_internal_heap_size:1480 bytes.
assert failed: host_recv_pkt_cb hci_hal_h4.c:552 (0)
[0609_trace_internal_heap.log](https://github.com/espressif/esp-idf/files/11702480/0609_trace_internal_heap.log)
[0609_sdkconfig.log](https://github.com/espressif/esp-idf/files/11702476/0609_sdkconfig.log)

Sky-Soo-Ap, 2023-06-09:
Hi @[wmy-espressif](https://github.com/wmy-espressif),
It seems BLE RX events did not parse to App side and handle one by one;
Then, heap memory is exhausted, let see log for details.
[0609_2_trace_heap.log](https://github.com/espressif/esp-idf/files/11702899/0609_2_trace_heap.log)
[0609_2_sdkconfig.log](https://github.com/espressif/esp-idf/files/11702901/0609_2_sdkconfig.log)

Sky-Soo-Ap, 2023-06-09:
Hi @[wmy-espressif](https://github.com/wmy-espressif),
Following is the log that turn all BT debug log level to verbose.
[0609_3_trace_heap_verbose.log](https://github.com/espressif/esp-idf/files/11704968/0609_3_trace_heap_verbose.log)

Sky-Soo-Ap, 2023-06-15:
Hi [wmy-espressif](https://github.com/wmy-espressif),
It seems the API “esp_mqtt_client_publish” may have a memory leak;
Because keeping calling the API will cause free heap size to decrease.
Let see the details in the attached log file.
[0614.zip](https://github.com/espressif/esp-idf/files/11752380/0614.zip)

I (18:46:03.854) azp_mqtt: parseResponsePacketToMQTT with topic:iometer/agent/devices/notification/bb:a0:50:fe:34:5a , esp_get_free_heap_size:4111472 bytes
W (18:46:13.883) transport_base: Poll timeout or error, errno=Success, fd=48, timeout_ms=10000
E (18:46:13.884) mqtt_client: Writing didn't complete in specified timeout: errno=0
I (18:46:13.891) azp_mqtt: MQTT_EVENT_ERROR
I (18:46:13.895) azp_mqtt: Last error code reported from esp-tls: 0x0
I (18:46:13.902) azp_mqtt: Last tls stack error number: 0x0
I (18:46:13.908) azp_mqtt: Last captured errno : 0 (Success)
I (18:46:13.927) azp_mqtt: MQTT_EVENT_DISCONNECTED
W (18:46:13.928) mqtt_client: Publish: Losing qos0 data when client not connected
I (18:46:13.931) azp_mqtt: parseResponsePacketToMQTT with topic:iometer/agent/devices/notification/bb:a0:50:fe:28:4d , esp_get_free_heap_size:4145808 bytes

wmy-espressif, 2023-06-15:
Hi @Sky-Soo-Ap thanks for your finding. To debug the memory leak issue, you can ues the heap debugging feature in esp-idf, please refer to the API reference: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/heap_debug.htm

I am a guy in Bluetooth and wireless coexistence. Will help another colleague of our company to help support this issue.

Sky-Soo-Ap, 2023-06-15:
Hi Mr./Miss,
I am using Standalone Mode to diagnose memory leaks.

ESP_ERROR_CHECK( heap_trace_start(HEAP_TRACE_LEAKS) );
esp_mqtt_client_publish(...)
ESP_ERROR_CHECK( heap_trace_stop() );
heap_trace_dump();

Following partial log is using [IDF Monitor](https://docs.espressif.com/projects/esp-idf/en/v5.0.2/esp32/api-guides/tools/idf-monitor.html) to automatically decode PC addresses to their source files & line number for more details.
[0615.txt](https://github.com/espressif/esp-idf/files/11754682/0615.txt)

euripedesrocha, 2023-06-19:
@Sky-Soo-Ap the decreasing in heap memory is expected if you are publishing QoS > 0 messages, internally esp-mqtt copies those messages to our outbox to be able to track their proper publishing and to retry publishing after some timeout. After the messages are fully processed, i.e., the client receives confirmation messages from the broker or the message expires, configurable expiration timeout, the message is removed from the outbox. 

What you may be experiencing is that outbox is storing messages faster than we can remove.

There are a few options:
- Reduce the timeout for the message expiration so the outbox can be cleaned faster.
- Recently we introduced a size limit to the outbox to control the memory usage, available on esp-mqtt master.
- Depending on your application needs, you can try to reduce the size of the messages or the frequency of publishing.

Sky-Soo-Ap, 2023-06-20:
Hi [euripedesrocha](https://github.com/euripedesrocha),
I’m using QoS = 0 to publish, so the reason is  that outbox is storing messages faster than removing them?

euripedesrocha, 2023-06-20:
@Sky-Soo-Ap with QoS=0 and publish messages shouldn't go to the outbox. 

I'll need your help to isolate this, I see that you are using MQTT over websocket. Could you share the config you are using? It would be great if you could share a reproducer for this.

Sky-Soo-Ap, 2023-06-20:
Hi [euripedesrocha](https://github.com/euripedesrocha),
Thank you for looking at it and the following is the sdkconfig file.
[0609_2_sdkconfig.log](https://github.com/espressif/esp-idf/files/11796401/0609_2_sdkconfig.log)

euripedesrocha, 2023-06-20:
@Sky-Soo-Ap I meant the runtime config for esp-mqtt client. If you could share a portion of the code that configures and uses the MQTT client, it will be very helpful. 

Also which version of idf and esp-mqtt are in use?

Sky-Soo-Ap, 2023-06-20:
Hi [euripedesrocha](https://github.com/euripedesrocha),
Following is the esp_mqtt_client_config_t, all the default except broker URL & PORT and IDF version is v5.0.2, about esp-mqtt is not changed.
ESP32 connected 3 BLE devices, and kept publishing notification packets with QoS=0.

// default broker & port also has this phenomenon
    const esp_mqtt_client_config_t mqtt_cfg = {
        .broker = {
            .address.uri = nvs_mqtt_broker_url,			
            .address.port = atoi(nvs_mqtt_broker_url_port)
        },
    };

euripedesrocha, 2023-06-21:
@Sky-Soo-Ap I tried to reproduce the issue, but I wasn't able to. Could you provide a test project that shows the issue?

Alvin1Zhang, 2023-08-07:
@Sky-Soo-Ap Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Sky-Soo-Ap, 2023-08-07:
Hi [Alvin1Zhang](https://github.com/Alvin1Zhang),
About continuing receive notification packet and publish to mqtt broker;
The free memory is dropping at the beginning and finally it will stay in the horizontal line.
We still monitor it.
Anyway, we closed it first.

closed 2023-08-07 by Sky-Soo-Ap

DV-Loss, 2024-04-05:
Hello, 
Sorry my misunderstanding, but I can't find how this Issue was solved. I'm getting very similar problem and don't know what is happening.

IDF version.
v5.2.1

Operating System used.
Windows 11

How did you build your project?
VS Code IDE

If you are using Windows, please specify command line type.
None

Development Kit.
ESP32-WROOM-32D Develpment kit

Power Supply used.
External 5V

What is the actual behavior?

My firmware connects to a BLE device and registers two characteristics to receive notifications. The device sends notifications at a 10Hz rate, with about 230 data bytes each time, the ESP32 central ressend this bytes to the UART2 port. If my ESP32 receiver and the device are placed over the table, the pair can work uninterrupted for days. However, if I position the device in a way that the physical radio signal becomes weak, it behaves as expected, losing data. But after some time (minutes), it begins to send the "btc_gattc_cback transfer failed" and "btc_transfer_context No memory" errors. At this state, it doesn't recover normal reception anymore, even if I place the ESP32 receiver and the device side by side. It's necessary to reset the ESP32 to make it work again. Disconnecting the device and connecting it again is not enough to recover normal behavior; the ESP32 must be reset. It seems like the BLE layer is receiving a message, but if it is not completed (due to weak signal reception), it doesn't clear the memory and will accumulate space with corrupted messages until there's not enough space for the good packets.

Steps to reproduce.
ESP32 is GATT client;
Scanning and connected with GATT server;
Continuing receive notification packet and Sending it to UART port

Debug Logs.
[0;33mW (187933) BT_BTC: btc_transfer_context No memory
[0m
[0;31mE (187933) BT_BTC: btc_gattc_cback transfer failed
[0m
[0;33mW (187983) BT_BTC: btc_transfer_context No memory
[0m
[0;31mE (187983) BT_BTC: btc_gattc_cback transfer failed
[0m
[0;33mW (187983) BT_BTC: btc_transfer_context No memory
[0m
[0;31mE (187983) BT_BTC: btc_gattc_cback transfer failed
[0m
[0;33mW (188023) BT_BTC: btc_transfer_context No memory
[0m
[0;31mE (188033) BT_BTC: btc_gattc_cback transfer failed
[0m
[0;33mW (188033) BT_BTC: btc_transfer_context No memory
[0m
[0;31mE (188033) BT_BTC: btc_gattc_cback transfer failed
[0m
[0;33mW (188073) BT_BTC: btc_transfer_context No memory
[0m
[0;31mE (188073) BT_BTC: btc_gattc_cback transfer failed
[0m
[0;33mW (188073) BT_BTC: btc_transfer_context No memory
[0m
[0;31mE (188073) BT_BTC: btc_gattc_cback transfer failed
[0m