espressif/esp-idf #11608: ESP32-S3 RISC-V I2C always failed in first read/write operation with G-sensor

FrankZyyz, 2023-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-921-gdc016f5987

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 N8R2

### Power Supply used.

USB

### What is the expected behavior?

I am using ESP32-S3 to test with I2C sensor LSM6DS3. 
SCL: GPIO_NUM_0
SDA: GPIO_NUM_1

1. I have a BMP180 sensor, it worked well with esp-idf/example/../ulp_riscv/i2c code.

2. Then I  change code of ulp_riscv/i2c to adopt LSM6DS3 sensor (i2c address and chip ID register ).

Normally, I should got chip ID (0x69) after code run.

### What is the actual behavior?

I call chip ID read function for twice. The first read operation always failed.

Log info:

> --- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1240
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa5c
load:0x403cc700,len:0x2d18
entry 0x403c9890
Not a ULP-RISC V wakeup (cause = 0)
Initializing RTC I2C ...
E (106) ulp_riscv_i2c: **Read Failed!**
E (106) ulp_riscv_i2c: RTC I2C Interrupt Raw Reg 0x12c
E (106) ulp_riscv_i2c: RTC I2C Status Reg 0x66d60000
mpu6050 sensor chip id: **0x00**
RTC_I2C_CTRL_REG: 20000004
SENS_SAR_I2C_CTRL_REG: 786B
RTC_I2C_STATUS_REG: 66D60000
RTC_I2C_INT_RAW_REG: 0000
mpu6050 sensor chip id: **0x69**
enter loop

### Steps to reproduce.

```c

void app_main(void)
{
    uint8_t data_rd = 0;
    int16_t ut_data = 0;
    int32_t up_data = 0;
    int16_t i16raw = 0;
    float temperature = 0.0;

    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();

    /* Not a wakeup from ULP
     * Initialize RTC I2C
     * Setup MPU6050
     * Store current temperature and G-sensor values
     * Load the ULP firmware
     * Go to deep sleep
     */
    if (cause != ESP_SLEEP_WAKEUP_ULP)
    {
        printf("Not a ULP-RISC V wakeup (cause = %d)\n", cause);

        /* Initialize RTC I2C */
        init_i2c();
        /* Configure I2C slave address */
        ulp_riscv_i2c_master_set_slave_addr(LSM6DS3_I2C_ADDRESS);
        /* Set LSM6DS3 */
        uint8_t data_wr = 0;

        /* Read the sensor ID*/
        ulp_riscv_i2c_master_set_slave_reg_addr(LSM6DS3_WHO_AM_I);
        ulp_riscv_i2c_master_read_from_device(&data_rd, 1);
        printf("1. mpu6050 sensor chip id: 0x%02x \n", data_rd);

        printf("RTC_I2C_CTRL_REG: %04X\n", *(unsigned int *)RTC_I2C_CTRL_REG);
        printf("SENS_SAR_I2C_CTRL_REG: %04X\n", *(unsigned int *)SENS_SAR_I2C_CTRL_REG);
        printf("RTC_I2C_STATUS_REG: %04X\n", *(unsigned int *)RTC_I2C_STATUS_REG);
        printf("RTC_I2C_INT_RAW_REG: %04X\n", *(unsigned int *)RTC_I2C_INT_RAW_REG);

        /* Confirm that the sensor is alive */
        ulp_riscv_i2c_master_set_slave_reg_addr(LSM6DS3_WHO_AM_I);
        ulp_riscv_i2c_master_read_from_device(&data_rd, 1);
        printf("2. mpu6050 sensor chip id: 0x%02x \n", data_rd);
#if 0
        /* Read temperature */
        lsm6ds3_read_temp(&i16raw, &temperature);
        printf("mpu6050 tempearture = %.2f, [%x]\n", temperature, i16raw);
#endif
        /* Load ULP firmware
         *
         * The ULP is responsible of monitoring the temperature and pressure values
         * periodically. It will wakeup the main CPU if the temperature and pressure
         * values are above a certain threshold.
         */  
        printf("enter loop\n");
        while(1)
        {
            vTaskDelay(pdMS_TO_TICKS(5000));

            // lsm6ds3_read_temp(&i16raw, &temperature);
            // printf("mpu6050 tempearture = %.2f, [%x]\n", temperature, i16raw);
        }
        // init_ulp_program();
    }

    if (cause == ESP_SLEEP_WAKEUP_ULP){
        printf("ULP RISC-V woke up the main CPU\n");

        /* Pause ULP while we are using the RTC I2C from the main CPU */
        ulp_timer_stop();
        ulp_riscv_halt();

        // ulp share value
        printf("reg val = %lx \n", ulp_reg_val);
        printf("temp = %lx = %f C\n", ulp_raw_temp, (float)(ulp_raw_temp)/16.0f+25.0f);
        printf("tempf= %f \n", (float)ulp_degree);

        /* Resume ULP and go to deep sleep again */
        ulp_timer_resume();
    }

    /* ULP RISC-V read and detected a temperature or pressure above the limit */
#if 1
    /* Add a delay for everything to the printed before heading in to deep sleep */
    vTaskDelay(100);

    /* Go back to sleep, only the ULP RISC-V will run */
    printf("Entering deep sleep\n\n");

    /* RTC peripheral power domain needs to be kept on to keep RTC I2C related configs during sleep */
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);

    ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());

    esp_deep_sleep_start();    
#endif
}
``` 

### Debug Logs.

```plain
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1240
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa5c
load:0x403cc700,len:0x2d18
entry 0x403c9890
Not a ULP-RISC V wakeup (cause = 0)
Initializing RTC I2C ...
E (106) ulp_riscv_i2c: Read Failed!
E (106) ulp_riscv_i2c: RTC I2C Interrupt Raw Reg 0x12c
E (106) ulp_riscv_i2c: RTC I2C Status Reg 0x66d60000
1. mpu6050 sensor chip id: 0x00
RTC_I2C_CTRL_REG: 20000004
SENS_SAR_I2C_CTRL_REG: 786B
RTC_I2C_STATUS_REG: 66D60000
RTC_I2C_INT_RAW_REG: 0000
2. mpu6050 sensor chip id: 0x69
enter loop
```

### More Information.

[image]
[image]
[image]

I have three i2c sensor,  bmp180 mpu6050, lsm6ds3, all this 3 sensors can work normal in Soc i2c driver.
But mpu6050 and lsm6ds3 always failed at first read function (no ACK feedback) when app_main call rtc risc-v rtc driver to access i2c.
The rtc ulp i2c driver can work good when enter deep-sleep mode which ulp running.

sudeep-mohanty, 2023-06-07:
Hi @FrankZyyz,
Thanks for reporting the issue. From the logic analyzer captures that you have shared, I can see that the first read/write operation with the LMS6DS3 looks correct but the ULP RISC-V I2C driver is timing out. It may happen that certain I2C devices could take longer to initialize and then have shorter read/write cycles subsequently. I suspect that the LMS6DS3 (and the MPU6050) behaves this way.

What I would like you to try is to increase the ULP RISC-V I2C driver read/write timeout. You can update this in the menuconfig by changing the `CONFIG_ULP_RISCV_I2C_RW_TIMEOUT` option. The default is set to 500 CPU cycles. Please try to increase this value and see if the first read goes through. Setting this to -1 will make the read/write loops wait forever.

FrankZyyz, 2023-06-07:
Hi @sudeep-mohanty  
Thanks for quick reply.
I adjust RW_TIMEOUT to -1, got same result:
```c
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1240
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa5c
load:0x403cc700,len:0x2d18
entry 0x403c9890
Not a ULP-RISC V wakeup (cause = 0)
Initializing RTC I2C ...
E (92) ulp_riscv_i2c: Read Failed!
E (92) ulp_riscv_i2c: RTC I2C Interrupt Raw Reg 0x12c
E (92) ulp_riscv_i2c: RTC I2C Status Reg 0x66d60000
1. sensor chip id: 0x00
RTC_I2C_CTRL_REG: 20000004
SENS_SAR_I2C_CTRL_REG: 786B
RTC_I2C_STATUS_REG: 66D60000
RTC_I2C_INT_RAW_REG: 0000
2. sensor chip id: 0x69
enter loop

``` 

Logic analyzer captures almost same waveform.
[ulp_100k_timeout-1.zip](https://github.com/espressif/esp-idf/files/11675386/ulp_100k_timeout-1.zip)

FrankZyyz, 2023-06-09:
@sudeep-mohanty  I apply the patch.
And I set ULP RISC-V I2C transaction timeout in ticks = 5000.
Nothing changed...
[image]
[image]

It looks like LMS6DS3 did not receive i2c master signal, then we can't got ACK at 9th clock. (But this issue only happened at first run.)
I am thinking maybe you can get some i2c sensor(mpu6050, LMS6DS3) to do the test in your local hardware. (there are something changed when I change the BMP180 to G-sensor.)
Remind again, I got BMP180 sensor do the test with example code, all the thing looks good.

sudeep-mohanty, 2023-06-09:
Thank you for trying out @FrankZyyz! I shall have to test with either or both of MPU6050 and LMS6DS3 to debug why the driver fails the read in the first attempt. May I also ask if you have any external pull-ups connected in your setup? If no, then it could be a reason for this behavior as the internal pull-up resistors are weak (30-50 kOhm). I would also like you to try this new patch which disables ACK checks during Write operations. Please let me know if it helps. Thanks!

[i2c_ack_chk_dis.patch](https://github.com/espressif/esp-idf/files/11703312/i2c_ack_chk_dis.patch)

FrankZyyz, 2023-06-13:
@sudeep-mohanty  Sorry for the late reply.
About pull-ups, my setup has 10k resistor on sensor board. For esp32s3 part, I tried both enable/disable internal pull-up configuration before create this issue. The result is the same.

For you new patch test. It's did not work.
I got following log:
[image]

And logic file FYI. (software: https://www.saleae.com/downloads/)
[ulp_i2c_ack_chk_dis.zip](https://github.com/espressif/esp-idf/files/11732762/ulp_i2c_ack_chk_dis.zip)

sudeep-mohanty, 2024-07-18:
Please check for potential fix [here](https://github.com/espressif/esp-idf/issues/14043#issuecomment-2236295533)

closed 2024-07-24 by espressif-bot