espressif/esp-idf #11610: BLE failed to establish connection

Sky-Soo-Ap, 2023-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

Can establish BLE connection normally.

### What is the actual behavior?

Failed to establish BLE connection;
Looping into connected/disconnected;
[BLE_Failed_To_Establish_Connection.log](https://github.com/espressif/esp-idf/files/11673037/BLE_Failed_To_Establish_Connection.log)
[sdkconfig.log](https://github.com/espressif/esp-idf/files/11673039/sdkconfig.log)

### Steps to reproduce.

1. ESP32 is GATT client;
2. Scanning and connect to GATT server;

### Debug Logs.

```plain
Attached log in actual behavior field.
```

zhp0406, 2023-06-10:
- Cause Analysis:

  - Error code rsn=0x3e indicates "Connection fail to establish". There are many reasons for this issue, such as peripheral not broadcasting or being too far away, interference in the air causing packet collisions, and so on.

- Solution:

  - Ensure that the peripheral is broadcasting.

  - The current log is not sufficient to determine the specific cause. Please provide us with some packet capture logs for further analysis.

Sky-Soo-Ap, 2023-08-17:
Anyway, it seems not happen again, closed this issue first.

closed 2023-08-17 by Sky-Soo-Ap