espressif/esp-idf #11611: Change IDF location after installing idf-tools

arex-ebee, 2023-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should be possible to change the location of esp-idf after idf tools have already been installed (via adapting IDF_PATH env variable accordingly).

### What is the actual behavior?

Installing IDF tools for only specific targets (esp32s3 in my case) from a copy of esp-idf at any location on the host machine succeeds with customized IDF_PATH and IDF_TOOLS_PATH environment variables pointing to the appropriate directories. Hence the "export.sh" script can be sourced without errors.
When now changing IDF_PATH to another copy (of the same IDF version) the export script fails claiming missing tools although those are not required.

### Steps to reproduce.

```
export IDF_TOOLS_PATH=/tmp/esp-idf-tools
export IDF_PATH=/tmp/esp-idf-1
git clone --recursive -b v5.0.2 https://github.com/espressif/esp-idf.git ${IDF_PATH}
${IDF_PATH}/install.sh esp32s3
...
( . ${IDF_PATH}/export.sh )
```

Output of export.sh script:
```
Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Adding ESP-IDF tools to PATH...
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Checking if Python packages are up to date...
Constraint file: /tmp/esp-idf-tools/espidf.constraints.v5.0.txt
Requirement files:
 - /tmp/esp-idf-1/tools/requirements/requirements.core.txt
Python being checked: /tmp/esp-idf-tools/python_env/idf5.0_py3.8_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  
  /tmp/esp-idf-1/components/espcoredump
  /tmp/esp-idf-1/components/partition_table
  /tmp/esp-idf-1/components/app_update
  /tmp/esp-idf-tools/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /tmp/esp-idf-tools/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin
  /tmp/esp-idf-tools/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
  /tmp/esp-idf-tools/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /tmp/esp-idf-tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
  /tmp/esp-idf-tools/python_env/idf5.0_py3.8_env/bin
  /tmp/esp-idf-1/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```
All fine up to here. Now change location of esp-idf (or clone to a new one)

```
export IDF_PATH=/tmp/esp-idf-2
git clone --recursive -b v5.0.2 https://github.com/espressif/esp-idf.git ${IDF_PATH}
...
( . ${IDF_PATH}/export.sh )
```

**Error message:**
```
Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
ERROR: tool riscv32-esp-elf-gdb has no installed versions. Please run '/usr/bin/python3 /tmp/esp-idf-2/tools/idf_tools.py install' to install it.
ERROR: tool xtensa-esp32-elf has no installed versions. Please run '/usr/bin/python3 /tmp/esp-idf-2/tools/idf_tools.py install' to install it.
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/usr/bin/python3 /tmp/esp-idf-2/tools/idf_tools.py install' to install it.
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
```

### More Information.

Same approach is working with IDF v4.4.4

igrr, 2023-06-07:
Aren't you missing `${IDF_PATH}/install.sh esp32s3` after the second `git clone`? It should work fine if you do run the install script, as mentioned in the getting started guide.

arex-ebee, 2023-06-08:
> Aren't you missing `${IDF_PATH}/install.sh esp32s3` after the second `git clone`? It should work fine if you do run the install script, as mentioned in the getting started guide.

This is intentionally not done as both IDF copies are expected to be identical.

Background is that the project's compile run is done inside a self-built docker container whose image already has the idf-tools installed (read-only) but **not** the IDF sources. The latter are part of the project and hence are run-time mounted into the container. But the actual mount path is considered unknown (it may change e.g. due to different users, Jenkins instances, etc.) so the IDF source path at container image creation time differs to the one at container run time. An additional requirement is to have a fixed build setup without any external dependencies e.g. to github (which probably would be introduced with a call to install.sh at the second run).

igrr, 2023-06-08:
> An additional requirement is to have a fixed build setup without any external dependencies e.g. to github

I understand, this is the requirement we don't support, then. (It just so happened to work in v4.x, but wasn't by design.) I would relabel this as 'feature request' rather than a 'bug', and look for a way to offer you a procedure which would meet your requirements.

(Thanks for clearly explaining your use case, by the way!)

I would suggest trying to capture the environment variables at installation time. Then at container run time you only need to set IDF_PATH and then source the environment variables. To get the environment after running install.sh, call `idf_tools.py export` and store stdout in a file. Then make sure your container entry point sources this file.

One thing i would suggest taking care of in this approach is not getting the tools and python packages outdated relative to IDF requirements. I.e. make sure you have different versions of the image for different versions of IDF.

arex-ebee, 2023-06-08:
> I understand, this is the requirement we don't support, then. (It just so happened to work in v4.x, but wasn't by design.) I would relabel this as 'feature request' rather than a 'bug', and look for a way to offer you a procedure which would meet your requirements.

I'm fine with it, thanks for attending to the issue.

> I would suggest trying to capture the environment variables at installation time. Then at container run time you only need to set IDF_PATH and then source the environment variables. To get the [...]

Just for my own clarification: Your proposal means not to source export.sh within the entrypoint but the previously stored environment instead, right? I will try...

> One thing i would suggest taking care of in this approach is not getting the tools and python packages outdated relative to IDF requirements. I.e. make sure you have different versions of the image for different versions of IDF.

I always kept that in mind. The image version is coupled to the IDF version ;)