espressif/esp-idf #11614: NVS partition lookup fails after OTA

MartyMcFlyInTheSky, 2023-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4665-gef64e4e5b3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32

### Power Supply used.

External 3.3V

### What is the expected behavior?

Our main method of upgrading our esp32-powered devices is through OTA. However just now an upgrade from a fairly old version to our newest version failed, while the same upgrade from a recent version to the newest version passed flawlessly.

### What is the actual behavior?

The OTA update itself completes successfully but the following error occurs on reboot:

```
I (13) boot: ESP-IDF v5.1-dev-327-g7c2b137ffe 2nd stage bootloader
I (13) boot: compile time 11:45:37
I (13) boot: chip revision: V300
I (17) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (24) boot.esp32: SPI Speed      : 40MHz
I (28) boot.esp32: SPI Mode       : DIO
I (33) boot.esp32: SPI Flash Size : 4MB
I (37) boot: Enabling RNG early entropy source...
I (43) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 0000a000 00004000
I (61) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (69) boot:  2 phy_init         RF data          01 01 00010000 00001000
I (76) boot:  3 emul_efuse       efuse            01 05 00011000 00002000
I (84) boot:  4 ota_0            OTA app          00 10 00020000 001e0000
I (91) boot:  5 ota_1            OTA app          00 11 00200000 001e0000
I (99) boot: End of partition table
I (140) boot_comm: chip revision: 3, min. application chip revision: 0
I (140) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=3c4fch (247036) map
I (234) esp_image: segment 1: paddr=0023c524 vaddr=3ffb0000 size=038d0h ( 14544) load
I (240) esp_image: segment 2: paddr=0023fdfc vaddr=40080000 size=0021ch (   540) load
I (240) esp_image: segment 3: paddr=00240020 vaddr=400d0020 size=c9d80h (826752) map
I (546) esp_image: segment 4: paddr=00309da8 vaddr=4008021c size=17208h ( 94728) load
I (597) boot: Loaded app from partition at offset 0x200000
I (597) boot: Disabling RNG early entropy source...
I (609) cpu_start: Multicore app
I (609) cpu_start: Pro cpu up.
I (610) cpu_start: Starting app cpu, entry point is 0x400814cc
0x400814cc: call_start_cpu1 at C:/dev/esp/esp-idf/components/esp_system/port/cpu_start.c:159

I (608) cpu_start: App cpu up.
[... 131 lines cut ...]
    assert(it != NULL); /* has to be at least one app partition */

    while (it != NULL) {
        const esp_partition_t *p = esp_partition_get(it);
        if (p->address <= phys_offs && p->address + p->size > phys_offs) {
            esp_partition_iterator_release(it);
            curr_partition = p;
            return p;
        }
        it = esp_partition_next(it);
    }

    abort(); /* Partition table is invalid or corrupt */
```

So it appears to me that the partition can't be found for some reason. This would make sense to me if the partition table had changed, but it hasn't! I've attached our current partition table for reference, but we haven't changed it since the erroneous release.
[anti_rollback_partition.csv](https://github.com/espressif/esp-idf/files/11675253/anti_rollback_partition.csv)

We really need OTA to _always_ work or we have to unscrew our casings of about 1000 devices in the future, which is very worrysome!

Additional note:

### Steps to reproduce.

Try OTA with a release a few minor version numbers back.

mahavirj, 2023-06-12:
@MartyMcFlyInTheSky 

> Try OTA with a release a few minor version numbers back.

Would it be possible for you to share more detailed information and steps on recreating this failure? Additionally, do you see similar behavior with any of the IDF provided OTA update examples, e.g., `simple_ota`?

MartyMcFlyInTheSky, 2023-06-13:
@mahavirj It's kinda hard to answer this as it only ever appeared once the version of the software was way older. I would have to retest with an old software version and even then it might just work. Idk, maybe some nonvolatile thingy got polluted over time. I will retest with the old software version and report back.

mahavirj, 2023-07-03:
@MartyMcFlyInTheSky Any further update on this?

closed 2023-07-07 by Alvin1Zhang