espressif/esp-idf #11615: ESP32: Wifi/Network stack broken after beacon timeout

FnxQT, 2023-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board with ESP32DOWDQ6

### Power Supply used.

External 3.3V

### What is the expected behavior?

The ESP32 connects to wifi and stays connected normally until it is powered off or a call to esp_wifi_disconnect() is made.

### What is the actual behavior?

Randomly our ESP32 lose Wifi connection, it happens in the filed regularly. The ESP32 operates correctly then suddenly two error messages appears :

```
wifi:m f null
wifi:bcn_timout,ap_probe_send_start
```

The first message is quite not comprehensible but the second suggests a timeout happened and that the AP did not send a beacon in the correct interval. Normally, the would actively probe the AP five times to "re-enable" the connection. However in this, it is not done and the ESP stays in this state until a reboot is performed. 

All network services stop working, no more web server, mdns, udp sockets and aws connection. The network is completely done. At first I thought it could be because of a memory leak but when it happens memory is high (60K left).

The problem is that the issue is rather random and not reproducible. It seems that using the "WIFI_PS_MIN_MODEM" power saving mode and connecting to Ubiquiti appliances trigger the problem more often.

 

### Steps to reproduce.

That's the problem, we cannot reproduce it. We see it happening a lot and it happened yersterday on an ESP32 we were monitoring and we just got the two messages above. It seems to be related to power saving mode and to ubiquiti appliances.

idea--list, 2023-06-08:
Same here on an ESP32-S3-MINI-! based board and IDF 5.0.2. Connecting to AWS.
Minor differences: i am using WIFI_PS_MAX_MODEM and auto light sleep enabled.
On my side it occurs ca 10 times during 24 hours irregularly, at least i can not recognize any pattern.

Xiehanxin, 2023-06-08:
hi, could you please provide a description of the application scenarios? Are there any uses of Bluetooth or other applications besides WiFi connectivity? Insufficient memory can sometimes result in this issue.

FnxQT, 2023-06-08:
In my case Bluetooth is not used, the application does not write to the flash a lot (I read somewhere that this maybe a problem because the chip needs to constantly read the flash to operate).

At the application level the ESP32 is used for multiple things: he has an HTTPD web server, makes broadcast requests on UDP, reads sensors from I2C, controls LEDs, communicates to another appliance through UART, communicates its data to AWS for the main points.

All other functions continue to work properly (except the one based on the network obviously). 

As for the memory, I'm tracking free heap. Both with esp_get_free_heap_size() to get the total free heap and with heap_caps_get_per_task_info() to get indication per tasks.

However, memory does not seem to be an issue since it is above 50-60K free in all scenarios.

I'm available to provide more information as needed.

idea--list, 2023-06-08:
In my case i integrated [iot reference repo](https://github.com/FreeRTOS/iot-reference-esp32c3) into my project. That means WiFi provisioning is done via Bluetooth, but i opted to turn Bluetooth off once provisioning is done. Beyond that the board sends MQTT messages, listens to OTA (over MQTT) to AWS, and periodically sends files over FTP to another server. All WiFi relevant task is pinned to core 0 in my case and i can confirm that all tasks/function that are not relevant to networking continue to run.

My ESP32S3 logs sensor data over SPI and I2C into ringbuffers and stores data to SD over SPI.

I am also monitoring memory: the issue occurs even when i have more than 100K free heap. Regarding task stack sizes, all my tasks have more than 1300 bytes free when the issue occurs, so at first glance available memory should be enough.

On top of that i also set stack smashing protection level to strong inside menuconfig, but that does not log anything.
(However when setting that to overall level, then the code does not even compile just like the original iot reference repo i integrated into my code). Not sure however if this might be relevant to the issue.

Xiehanxin, 2023-06-08:
hi @FnxQT, The current version is a bit outdated. We have made many fixes on a more recent version. Please try updating to version 4.3.3 or higher version.
hi @idea--list,  you have used the BLE, which seems to be another issue. Please open a new issue for that

FnxQT, 2023-06-09:
Hi @Xiehanxin, 

I can try a lucky punch by updating to a newer version but since it seems to still be a problem in 5.0.2, I'm not sure about it. 

However, I have news. We were able to monitor an ESP again when the problem happened. It seems the problem is not related to the beacon timeout since this time we only received the error log:

`wifi: m f null`

Could you please inform us on what this message is about ? Since the wifi lib is precompiled it is a bit difficult to debug. The heap was still big (47K available, biggest free block 15K). 

Another is that the ESP32 still thinks it is connected to the access point. esp_wifi_sta_get_ap_info(&ap_info) still returns all the information about the connection. However when checking on the Ubiquiti/Unifi web GUI, the AP thinks the ESP is disconnected.

Xiehanxin, 2023-06-09:
hi, @FnxQT,  could you please begin a periodic task to print  _wifi_statis_dump(0xffff)_ and upload your log? without being too frequent, just print every minutes

FnxQT, 2023-06-12:
Hello @Xiehanxin,

Sorry for the late reply. I started the system with wifi_statis_dump as you said. Will provide you with logs once the issue is triggered again.

FnxQT, 2023-06-19:
Hello @Xiehanxin,

Do you have any news about the issues ? Were the logs helpful, do you need more ?

Xiehanxin, 2023-06-19:
hi @FnxQT We are analyzing, and if there is any progress, we will inform you as soon as possible.

Espressif-liuuuu, 2023-06-19:
Hi @FnxQT , Sorry for late reply! This one is a little bit difficult to debug, so lets make sure two things at first on v4.4.4:
1) the `m f null` happened 
2) there was no log shown like `beacon timeout` or `wifi disconnect`, but it failed to see STA in AP`s list

And from the log you provided, it showed that the hardware was still working, meanwhile, only a little traffic seen. We may need a bit more information for it:
1) A piece of Wi-Fi packet captured by sniffer would be great, please set your AP to OPEN so we could easy to debug
2) A quick way for workaround on this issue is turning off the Wi-Fi power-save, you could try `esp_set_wifi_ps(0)`
3) Is it easy to reproduce the issue? It would be appreciated to provide us with a reproducible project

FnxQT, 2023-06-20:
Hello @Espressif-liuuuu, thank you for analysing the problem. Here are my responses:

1. Yes the `m f null` happened on v4.4.4
2. There was no log shown such as `beacon timeout` or `wifi disconnect`, no indication whatsoever that the wifi is malfunctioning.

I let the system under surveillance run since the 14th of June. I monitored what happened and noticed the following. The issue repeats itself but not in a periodical manner. Since the 14th of June it happened:

- 14-06 at 4:40pm
- 15-06 at 11:25am
- 15-06 at 1:07pm
- 15-06 at 2:02pm
- 15-06 at 2:57pm
- 16-06 at 9:19am
- 16-06 at 12am
- 17-06 at 6:01am
- 19-06 at 9:21am
- 19-06 at 5:34pm

I found a way to "detect" and restart the wifi stack entirely when the problem happens. I simply open a UDP socket and send broadcast packets on the network. If I cannot send a packet for more than 15min I stop and restart the wifi interface :

```
int result = sendto(sock, message, strlen(message), 0, (struct sockaddr*)&broadcast_addr, sizeof(broadcast_addr));
            if (result < 0) {
                int error = 0;
                socklen_t len = sizeof (error);
                int retval = getsockopt (sock, SOL_SOCKET, SO_ERROR, &error, &len);

                if (retval != 0) {
                    // there was a problem getting the error code
                    ESP_LOGE(TAG, "error getting socket error code: %s", strerror(retval));
                }

                if (error != 0) {
                    // socket has a non-zero error status
                    ESP_LOGE(TAG, "socket error: %s", strerror(error));
                }

                ESP_LOGE(TAG, "Failed to send message. Error: %d", errno);
                ESP_LOGE(TAG, "Error message: %s", strerror(errno));
                udp_error_cnt++;
if(udp_error_cnt >= 90){
                  ESP_LOGE(TAG, "UDP socket failed for more than 15 minutes, resetting wifi");
                  wl_on_disconnect(WL_EVENT_STA_DISCONN);
                  esp_wifi_disconnect();
                  esp_wifi_stop();
                  esp_wifi_start();
                  vTaskDelay(20000 / portTICK_PERIOD_MS);
                  esp_wifi_connect();
                  udp_error_cnt = 0;
              }
```

Interestingly, the behaviour after the `wifi:m f null` error is always the same. For 3 to 4 attempts, UDP packets are sent correctly but then they all fail with : 

`Errno 12 : Not enough space `

Until the wifi interface is completely stopped and restarted. It seems the TX buffer is completely full because the packets cannot be sent due to the wifi interface being in a bad condition.

As for the other question:

1. I'll try to use a sniffer to give what you want. However, what are you calling open mode ? Do you want that I remove the security of my network or that I monitor it all?
2. We tried this workaround already with a few clients, we're still waiting for feedback but it seems to improve the situation. However, we need the power-saving option to reduce the heat of the ESP32 at it is needed by our product.
3. Unfortunately the issue is really not easy to reproduce. We currently have one device in the office that we monitor and that produces the issue more or less frequently. Since we are making a commercial product I unfortunately cannot pass along all the codebase. However, I'll try to create a reduced version with just the wifi stack we use and try to reproduce the problem. This I could give to you.

Let me know if you need anything else in the interval. Thanks.

Xiehanxin, 2023-06-20:
hi @FnxQT  ,  as for open mode, yes, it means you set the router open, without any password, so that we can analyze the Wi-Fi packet more easily.

FnxQT, 2023-06-21:
Hi @Espressif-liuuuu, @Xiehanxin 

I ran Wireshark to check on the monitored system. Unfortunately, there is basically nothing to show as when the error "wifi:m f null" happens then the ESP32 sends nothing more on the network and remains quiet until the entire wifi stack is restarted. After that, the network operations are normal again.

I'm trying to trim down our production code base so that you have an example to try. However since our code is pretty large it takes me a bit of time. In the mean time I can already send you our wifi stack code.

Let me know if I can do anything else to help you. It seems to be related to wifi power saving since it's the only significant change we made in our wifi stack.

[wl.zip](https://github.com/espressif/esp-idf/files/11814784/wl.zip)

FnxQT, 2023-06-22:
Hi @Espressif-liuuuu, 

I attached a packet capture to this message. The Wifi stopped working on the 21th of June at 09:05am until 09:21am. After 15 minutes of not being able to send UDP packets, wifi interface was reset. I also join the corresponding log from the ESP directly. The IP of the system under surveillance is 192.168.0.202. The ESP log begins a bit earlier with the previous wifi:m f null error so that you see the evolution.

No we do not use AP + STA mode simultaneously. We use AP mode only during provisioning. However, in the case of the system under surveillance, AP mode is never enabled. It's only in STA mode.
[wifi_cap_and_logs.zip](https://github.com/espressif/esp-idf/files/11829547/wifi_cap_and_logs.zip)

I updated the system under surveillance with your libs thanks. However, I do not seem to notice a different output from wifi_statis_dump, is it normal ? I will send the associated logs and packet capture once the problem has been triggered again.

[wifi_cap_and_logs.zip](https://github.com/espressif/esp-idf/files/11829550/wifi_cap_and_logs.zip)

FnxQT, 2023-06-22:
Hi @Espressif-liuuuu, @Xiehanxin,

The issue was triggered using the debug libs you provided and new debug information was out.

The `wifi:m f null` happened two times, first on 22-06 at 12:03PM and second at 2:17PM.

Thank you for your help.

[Archive.zip](https://github.com/espressif/esp-idf/files/11835205/Archive.zip)

Espressif-liuuuu, 2023-06-25:
hi @FnxQT , there was a festival days before so sorry for the late reply  
I saw captured packet and it seems to be an ethernet one, and the wireless one would be helpful  :)  

And I think I have found the reason for this issue, which is a mistake for Wi-Fi power-save state machine when handling the failure of allocating nulldata. I have attacthed another lib below and this one shall be able to workaround this issue.   

The `m f null` may still happen, which was beacause of the shortage of memory at that moment, while the blocked packets shall be relieved later since the memory would be recoverd later. If you still find that there is no traffic after `m f null` for a long time, please feel free to provide us with the packet and log. Thanks!
[esp32_76263271.zip](https://github.com/espressif/esp-idf/files/11859592/esp32_76263271.zip)

FnxQT, 2023-06-25:
Hello @Espressif-liuuuu, no problem thank you for analysing the issue. 

It's strange, I ran Wireshark on my Mac on the wireless interface, not on the ethernet one. I will check if I can do something different.

Ok. I will update the running system with your library tomorrow and check for errors. However there is something strange. The system have about 50K of ram left before the `m f null` error and about the same amount after. Why is it like that ? Why does the wifi stack sometimes need to allocate a lot more memory ?

Thank you.

Espressif-liuuuu, 2023-06-25:
> Hello @Espressif-liuuuu, no problem thank you for analysing the issue.
>
> It's strange, I ran Wireshark on my Mac on the wireless interface, not on the ethernet one. I will check if I can do something [...]

I guess thats because of the shortage of internal memory. Memory with DMA capability is needed for Wi-Fi TX/RX, you could try `esp_get_free_internal_heap_size` instead to query left internal RAM.

FnxQT, 2023-06-29:
Hello @Espressif-liuuuu,

Since Monday, the `wifi:m f null` appeared only once and it has not caused any problem ! 💯 Wifi was still working after and our web interface too. However, I logged the internal DMA capable RAM as you said, and both the amount before and after the `wifi:m f null` were above 35K free. 

Is there something else that could cause the error ?

I will continue monitoring the system and keep you updated. I'm out of office for a week though.

Espressif-liuuuu, 2023-06-29:
> Hello @Espressif-liuuuu,
> 
> Since Monday, the `wifi:m f null` appeared only once and it has not caused any problem ! 100 Wifi was still working after and our web interface too. However, I logged the internal DMA capable RAM as you said, and both the amount before and after the `wifi:m f null` were above 35K free.
> 
> Is there something else that could cause the error ?
> 
> I will continue monitoring the system and keep you updated. I'm out of office for a week though.

Yes thats do weird. Could please add a check at the line below when the `heap_caps_malloc` return null, and log the internal memory by `esp_get_free_internal_heap_size`? Lets make sure if the memory is sufficient at the moment.

https://github.com/espressif/esp-idf/blob/8b94183c9cb47ede8f02df5598d8b9d68c754860/components/esp_wifi/esp32/esp_adapter.c#L544C4-L544C4

FnxQT, 2023-07-06:
Hey @Espressif-liuuuu, I just returned from days off. Before going, I added the esp_get_free_internal_heap_size print where you told me to.

However, during the previous week, the problem did not trigger again. I will keep you updated.

FnxQT, 2023-07-13:
Hi @Espressif-liuuuu,

The `wifi:m f null` error triggered again but as expected it did not cause the wifi stack to crash, so it's all okay. I saw in the log that at the moment the wifi tries to allocate internal memory the heap is nearly full, about 564 bytes left. Here is the log :

```
13-07 06:18:47 E (222535226) WIFI_MD: internal DMA capable free heap: 564
13-07 06:18:47 W (222535226) wifi:m f null
13-07 06:18:47
13-07 06:18:47 time: 3490810494, pm: 0, cca; 51db4, qblock: 0, txst1: 0, txst2: 0
13-07 06:18:47 [ac 0: sw:0, hw: 16f6d44]
13-07 06:18:47 [ac 1: sw:0, hw: 0]
13-07 06:18:47 [ac 2: sw:0, hw: 22e602c]
13-07 06:18:47 [ac 3: sw:0, hw: 0]
13-07 06:18:47 [ac 4: sw:0, hw: 0]
13-07 06:18:47 === tx st info ===
13-07 06:18:47 ac: 0
13-07 06:18:47 [i: 3, t: 3448288321, e: 0, s: 2667, b: 3] [i: 4, t: 3448305647, e: 0, s: 2668, b: 1]
13-07 06:18:47 [i: 5, t: 3448306045, e: 0, s: 2668, b: 7] [i: 6, t: 3448306604, e: 0, s: 2527, b: 3] [i: 7, t: 3448306908, e: 0, s: 2527, b: 3] [i: 8, t: 3448392768, e: 0, s: 2669, b: 2] [i: 9, t: 3451702135, e: 0, s: 2670, b: 1]
13-07 06:18:47 [i: 10, t: 3451702730, e: 0, s: 2528, b: 1] [i: 11, t: 3451762473, e: 0, s: 2671, b: 3] [i: 12, t: 3461708143, e: 0, s: 2672, b: 3] [i: 13, t: 3461708774, e: 0, s: 2529, b: 1] [i: 14, t: 3461759734, e: 0, s: 2673, b: 0]
13-07 06:18:47 [i: 15, t: 3471712034, e: 0, s: 2674, b: 2] [i: 16, t: 3471715013, e: 0, s: 2530, b: 0] [i: 17, t: 3471767138, e: 0, s: 2675, b: 2] [i: 18, t: 3481716583, e: 0, s: 2676, b: 0] [i: 19, t: 3481717174, e: 0, s: 2531, b: 2]
13-07 06:18:47 [i: 20, t: 3481767992, e: 0, s: 2677, b: 3] [i: 21, t: 3487205539, e: 0, s: 2678, b: 2] [i: 22, t: 3487206174, e: 0, s: 2532, b: 0] [i: 23, t: 3487206527, e: 0, s: 2532, b: 7] [i: 24, t: 3487206770, e: 0, s: 2532, b: 7]
13-07 06:18:47 [i: 25, t: 3487206975, e: 0, s: 2532, b: 6] [i: 26, t: 3487207165, e: 0, s: 2532, b: 4] [i: 27, t: 3487207395, e: 0, s: 2532, b: 0] [i: 28, t: 3487207559, e: 0, s: 2532, b: 4] [i: 29, t: 3487207841, e: 0, s: 2532, b: 4]
13-07 06:18:47 [i: 30, t: 3487208076, e: 0, s: 2532, b: 3] [i: 31, t: 3487208937, e: 0, s: 2533, b: 0] [i: 32, t: 3487270897, e: 0, s: 2679, b: 3] [i: 33, t: 3488307721, e: 0, s: 2680, b: 0] [i: 34, t: 3488308293, e: 0, s: 2534, b: 2]
13-07 06:18:47 [i: 35, t: 3488311629, e: 0, s: 2535, b: 1] [i: 36, t: 3488362685, e: 0, s: 2681, b: 2] [i: 37, t: 3489408773, e: 0, s: 2682, b: 2] [i: 38, t: 3489412125, e: 0, s: 2536, b: 1] [i: 39, t: 3489412769, e: 0, s: 2537, b: 1]
13-07 06:18:47 [i: 40, t: 3489463701, e: 0, s: 2683, b: 3] [i: 41, t: 3490506666, e: 0, s: 2684, b: 1] [i: 42, t: 3490507288, e: 0, s: 2538, b: 0] [i: 43, t: 3490509066, e: 0, s: 2538, b: 0] [i: 44, t: 3490509228, e: 0, s: 2538, b: 5]
13-07 06:18:47 [i: 45, t: 3490509553, e: 0, s: 2538, b: 5] [i: 46, t: 3490509849, e: 0, s: 2538, b: 0] [i: 47, t: 3490513254, e: 0, s: 2539, b: 0] [i: 48, t: 3490564259, e: 0, s: 2685, b: 2] [i: 49, t: 3490706422, e: 0, s: 2686, b: 3]
13-07 06:18:47
13-07 06:18:47 [i: 0, t: 3490707022, e: 0, s: 2540, b: 1] [i: 1, t: 3490757896, e: 0, s: 2687, b: 2] [i: 2, t: 3490758299, e: 0, s: 2687, b: 0]
13-07 06:18:47 ac: 1
13-07 06:18:47
13-07 06:18:47
13-07 06:18:47 ac: 2
13-07 06:18:47 [i: 7, t: 3741193037, e: 0, s: 513, b: 8] [i: 8, t: 3741194026, e: 0, s: 0, b: 20] [i: 9, t: 3741194791, e: 0, s: 0, b: 24]
13-07 06:18:47 [i: 10, t: 3741196183, e: 0, s: 513, b: 14] [i: 11, t: 3741196597, e: 0, s: 0, b: 15] [i: 12, t: 3746205640, e: 0, s: 514, b: 11] [i: 13, t: 46409463, e: 0, s: 515, b: 0] [i: 14, t: 46510074, e: 0, s: 516, b: 11]
13-07 06:18:47 [i: 15, t: 46510556, e: 0, s: 0, b: 10] [i: 16, t: 51521867, e: 0, s: 517, b: 5] [i: 17, t: 51522275, e: 0, s: 0, b: 15] [i: 18, t: 51522984, e: 0, s: 0, b: 27] [i: 19, t: 51523804, e: 0, s: 0, b: 54]
13-07 06:18:47 [i: 20, t: 51525475, e: 0, s: 517, b: 7] [i: 21, t: 51525827, e: 0, s: 0, b: 15] [i: 22, t: 51526964, e: 0, s: 517, b: 12] [i: 23, t: 564431845, e: 0, s: 518, b: 12] [i: 24, t: 564432422, e: 0, s: 518, b: 0]
13-07 06:18:47 [i: 25, t: 564435526, e: 0, s: 519, b: 1] [i: 26, t: 564435975, e: 0, s: 0, b: 13] [i: 27, t: 577567321, e: 0, s: 520, b: 6] [i: 28, t: 577606773, e: 0, s: 521, b: 14] [i: 29, t: 577619009, e: 0, s: 522, b: 9]
13-07 06:18:47 [i: 30, t: 577622885, e: 0, s: 523, b: 7] [i: 31, t: 577623443, e: 0, s: 0, b: 27] [i: 32, t: 577624739, e: 0, s: 523, b: 13] [i: 33, t: 646482827, e: 0, s: 524, b: 2] [i: 34, t: 646540708, e: 0, s: 525, b: 13]
13-07 06:18:47 [i: 35, t: 646541274, e: 0, s: 0, b: 12] [i: 36, t: 651596540, e: 0, s: 526, b: 9] [i: 37, t: 651597110, e: 0, s: 0, b: 24] [i: 38, t: 651598509, e: 0, s: 526, b: 10] [i: 39, t: 1246766873, e: 0, s: 527, b: 9]
13-07 06:18:47 [i: 40, t: 1246817872, e: 0, s: 528, b: 8] [i: 41, t: 1246818422, e: 0, s: 0, b: 14] [i: 42, t: 1846907154, e: 0, s: 529, b: 0] [i: 43, t: 1847105710, e: 0, s: 530, b: 4] [i: 44, t: 1847106171, e: 0, s: 0, b: 5]
13-07 06:18:47 [i: 45, t: 1851958648, e: 0, s: 531, b: 9] [i: 46, t: 2377627450, e: 0, s: 532, b: 3] [i: 47, t: 2377666296, e: 0, s: 533, b: 5] [i: 48, t: 2377666872, e: 0, s: 0, b: 6] [i: 49, t: 2377673745, e: 0, s: 534, b: 13]
13-07 06:18:47
13-07 06:18:47 [i: 0, t: 2447218203, e: 0, s: 535, b: 4] [i: 1, t: 2447397808, e: 0, s: 536, b: 15] [i: 2, t: 2447398377, e: 0, s: 0, b: 15] [i: 3, t: 3047500318, e: 0, s: 537, b: 7] [i: 4, t: 3047684782, e: 0, s: 538, b: 5]
13-07 06:18:47 [i: 5, t: 3047687559, e: 0, s: 0, b: 3] [i: 6, t: 3052617846, e: 0, s: 539, b: 13]
13-07 06:18:47 ac: 3
13-07 06:18:47
[... cut ...]
13-07 06:22:12 I (222740581) wifi: ifx auth     assoc    deauth action amsdu
13-07 06:22:12 I (222740581) wifi: sta 1        1        0      40     0
13-07 06:22:12 I (222740582) wifi: ap  0        0        0      0      0
13-07 06:22:12 I (222740582) wifi:
13-07 06:22:12 I (222740583) wifi: HMAC TX:
13-07 06:22:12 I (222740583) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
13-07 06:22:12 I (222740584) wifi: 89193    0        0      0      0      0       1
13-07 06:22:17 D (222745519) MDS: Query success
```

FnxQT, 2023-07-18:
Hi @Espressif-liuuuu,

In the past few days the issue triggered like 3 times (less than before) and the wifi lib stayed stable. The problem seems to be fixed.

It's only strange to me that the heap goes from 55K free to ~500 bytes free but one assumption, as you explained, is that it is the amount of memory mbedtls needs to create SSL/TLS connections.

Do you need something more ?

Espressif-liuuuu, 2023-07-19:
Hi @FnxQT , thanks for your information  
yes from your log it seems that the memory was little at `m f null` while it came back quickly later, we would discuss today and keep you updated if there is anything else needed
13-07 06:18:47 E (222535226) WIFI_MD: internal DMA capable free heap: 564
13-07 06:19:12 E (222559793) HEAP_DEBUG: internal DMA capable free heap: 34692

nonoo, 2023-09-15:
Any updates on a fix?

Espressif-liuuuu, 2023-09-18:
> Any updates on a fix?

hi, do you mean you also face with the same issue? A patch could be provided to fix that issue, please provide us with your sdk version and we could provide you the patch.

AxelLin, 2023-09-18:
> > Any updates on a fix?
> 
> hi, do you mean you also face with the same issue? A patch could be provided to fix that issue, please provide us with your sdk version and we could provide you the patch.

I don't get it.
If you already have a fix, why not fix it in github and close the issue?

AxelLin, 2023-09-18:
@Espressif-liuuuu 
Maybe you want to take a look at this "beacon timeout" issue as well: https://github.com/espressif/esp-idf/issues/11124#issuecomment-1720411364

nonoo, 2023-09-18:
> > Any updates on a fix?
> 
> hi, do you mean you also face with the same issue? A patch could be provided to fix that issue, please provide us with your sdk version and we could provide you the patch.

Yes. Also with an Ubiquity AP. Both with SDK version 4.4 and 5.0, with an ESP32S3

Espressif-liuuuu, 2023-09-18:
> > > Any updates on a fix?
> > 
> > 
> > hi, do you mean you also face with the same issue? A patch could be provided to fix that issue, please provide us with your sdk version and we could provide you the patch.
> 
> Yes. Also with an Ubiquity AP. Both with SDK version 4.4 and 5.0, with an ESP32S3

this issue happened with output `m f null`, just make sure that there is same situation on your side.  
And pls provide with the exact commit number or tag, e.g v5.0.2

Espressif-liuuuu, 2023-09-18:
> > > Any updates on a fix?
> > 
> > 
> > hi, do you mean you also face with the same issue? A patch could be provided to fix that issue, please provide us with your sdk version and we could provide you the patch.
> 
> I don't get it. If you already have a fix, why not fix it in github and close the issue?

this issue resulted from nulldata, which we want to make big change later, and it happened not that frequently, so we`d like to provide a patch instead of workaround to fix it now:)

Espressif-liuuuu, 2023-09-18:
> > > > > Any updates on a fix?
> > > > 
> > > > 
> > > > hi, do you mean you also face with the same issue? A patch could be provided to fix that issue, please provide us with your sdk version and we could provide you the patch.
> > > 
> > > 
> > > I don't get it. If you already have a fix, why not fix it in github and close the issue?
> > 
> > 
> > this issue resulted from nulldata, which we want to make big change later, and it happened not that frequently, so we`d like to provide a patch instead of workaround to fix it now:)
> 
> @Espressif-liuuuu
> 
> 1. "it happened not that frequently" is a big problem if you have thousands of devices on the market.
>    Because some devices just stop working randomly.
> 2. If you don't fix it in esp-idf and provide fix privately instead, you are asking people stick on a certain version with your fix.
>    Then there is no way to get other bug fix by sdk update.
> 3. People want to get fix for current issue, not really care about the big change later.
>    Don't mix bugfix with your refactor in the future.
>    You need to fix it first before further refactor.

yes ur right, we will think about it, thanks for ur suggestion:)

AxelLin, 2023-10-12:
Hi @Espressif-liuuuu

Is [c8a6ac54c577b479ab10f69341f98a3ac1d345a3](https://github.com/espressif/esp32-wifi-lib/commit/c8a6ac54c577b479ab10f69341f98a3ac1d345a3)  related to this issue?

Espressif-liuuuu, 2023-10-12:
> Hi @Espressif-liuuuu
> 
> Is [c8a6ac54c577b479ab10f69341f98a3ac1d345a3](https://github.com/espressif/esp32-wifi-lib/commit/c8a6ac54c577b479ab10f69341f98a3ac1d345a3) related to this issue?

Hi, it didnt. It fixes another issue introduced recently on master, beacon timeout happend immedieatly after connection established.

plehmkuhl-ampere, 2023-10-12:
> Hi, it didnt. It fixes another issue introduced recently on master, beacon timeout happend immedieatly after connection established.

Can you give an ETA on a fix for this issue?

Espressif-liuuuu, 2023-10-12:
> Can you give an ETA on a fix for this issue?

Its under review and would be merged recently. Maybe on v5.2 or the next master sync.  

I have to clarify that the title of this ticket maybe misunderstanding. The issue we resolved is the station could not reply and receive data from AP after log shown as `m f null`.  

The original issue of `beacon timeout` has gone after upgrading to v4.4.4. Please consider to open another ticket if you still have some trouble related with `beaocn timeout`.

Espressif-liuuuu, 2023-10-25:
This issue, the no response of station after `m f null` happened, has been fixed on lastest master and release/v5.1.
cc @FnxQT

AxelLin, 2023-10-25:
@Espressif-liuuuu 
Could you share the commit for the fix? (It's strange the espressif-bot does not push the fix to this issue.)
Since this issue is still OPEN, so which branches still need fix?

Espressif-liuuuu, 2023-10-25:
> @Espressif-liuuuu Could you share the commit for the fix? (It's strange the espressif-bot does not push the fix to this issue.) Since this issue is still OPEN, so which branches still need fix?

Maybe I missed the link of this ticket in MR.
The fixed version of v5.1: 21360b9ef3a2c5747b2effdb3213182db47dd2a7
And correct for master: it would be in the next sync

Alvin1Zhang, 2023-10-25:
Thanks for tracking, the fix on master branch is available at https://github.com/espressif/esp-idf/commit/db4598e305aa0eb2f140c7cef6df922ea80b4235. 

And we plan to back port to release/4.3 and release/4.4, will update once fixes on release/4.3 and release/4.4 are available. Thanks.

FnxQT, 2023-10-25:
Hi @Espressif-liuuuu, 

thank you for the followup. However as suggested by @Alvin1Zhang, it would be great to see the fix back ported to v4.4 and v4.3 since they are still under maintenance. Our products are currently using 4.4.4 and we can't easily make a jump to v5.1...

What do you think about that ?

FnxQT, 2023-11-21:
Hi @Espressif-liuuuu,

Maybe this information is still important for you. I was playing with the updated library you gave me on the 25th of June in the file : [esp32_76263271.zip](https://github.com/espressif/esp-idf/files/11859592/esp32_76263271.zip)

And today out of nowhere my esp32 crashed after the wifi:m f null error happened in infinite loop. Here is the log if it's of some importance to you. Unfortunately I did not have the "esp_wifi_statis_dump(0xffff);" method in my code.

[output_logs.txt](https://github.com/espressif/esp-idf/files/13426195/output_logs.txt)

AxelLin, 2023-11-22:
> Thanks for tracking, the fix on master branch is available at [db4598e](https://github.com/espressif/esp-idf/commit/db4598e305aa0eb2f140c7cef6df922ea80b4235).
> 
> And we plan to back port to release/4.3, release/4.4 and release/5.0, will update once fixes on those branches are available. Thanks.

@Alvin1Zhang 
How is the status of the backport? I still cannot find the fix in 4.3, 4.4, and 5.0.

Espressif-liuuuu, 2023-11-22:
backport information for old branches, anyway I suggest to update to lastest version to avoid the bug in middle version
4.3: 365b867
4.4: d665b47
5.0: e9ee047
And thanks for your report @FnxQT , I saw the log and thought that in your scenary the memory is indeed to the limit. The consecutive debug print in high priority task (Wi-Fi task) may leave no time for your application to free memory, which resulting in more `m f null` and more debug print. I suggest that:
1. update the branch to lastest v4.3 at your convenience

AxelLin, 2023-11-26:
> backport information for old branches, anyway I suggest to update to lastest version to avoid the bug in middle version 4.3: 5.0: [e9ee047](https://github.com/espressif/esp-idf/commit/e9ee0479086a63fda0cc7f28b17b069a7ec5503a) 

@Espressif-liuuuu Are you sure above commit is the fix fo this issue in 5.0 branch? (The commit looks different from other branches).

AxelLin, 2023-11-27:
> > > backport information for old branches, anyway I suggest to update to lastest version to avoid the bug in middle version 4.3: 5.0: [e9ee047](https://github.com/espressif/esp-idf/commit/e9ee0479086a63fda0cc7f28b17b069a7ec5503a)
> > 
> > 
> > @Espressif-liuuuu Are you sure above commit is the fix fo this issue in 5.0 branch? (The commit looks different from other branches).
> 
> yes it is. Its brought in the MR.

I check the v5.0 branch of esp32-wifi-lib, but I still cannot find the fix.
Could you please point out which commit fix this issue?

Espressif-liuuuu, 2023-11-29:
> I check the v5.0 branch of esp32-wifi-lib, but I still cannot find the fix. Could you please point out which commit fix this issue?

Hi, it did in the C3 eco7 MR https://github.com/espressif/esp-idf/commit/e9ee0479086a63fda0cc7f28b17b069a7ec5503a while its not named in the commit information, you may see it in the changelog of the release/v5.0 later, which shall be v5.0.5

AxelLin, 2023-11-29:
> Hi, it did in the C3 eco7 MR [e9ee047](https://github.com/espressif/esp-idf/commit/e9ee0479086a63fda0cc7f28b17b069a7ec5503a) while its not named in the commit information, you may see it in the changelog of the release/v5.0 later, which shall be v5.0.5

The bug report was against ESP32DOWDQ6, but you said the fix is in the C3 eco7 MR? confused.

Espressif-liuuuu, 2023-11-29:
The series of chips share most of the Wi-Fi code, so the issue is common. Sorry for the ambiguity.

Moldisocks, 2023-12-12:
Not sure if this helps, but for me I had a memory leak. Make sure any heap allocated memory gets freed

closed 2024-02-27 by Alvin1Zhang