espressif/esp-idf #11617: Incompatible settings for CONFIG_FREERTOS_UNICORE and CONFIG_SPIRAM

derrick-senva, 2023-06-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

A 2nd stage bootloader built with single-core mode and PSRAM disabled should be able to boot into application firmware built with dual-core mode and PSRAM enabled.

### What is the actual behavior?

The application firmware crashes on an IllegalInstruction.

### Steps to reproduce.

1. Build a bootloader and partition table with ESP-IDF v5.0.2 on an ESP32 with the attached sdkconfig.txt file. The main configuration settings that are in question are` CONFIG_SPIRAM` is not set and `CONFIG_FREERTOS_UNICORE=y`.

[sdkconfig-original.txt](https://github.com/espressif/esp-idf/files/11682496/sdkconfig-original.txt)

2. Build application firmware with the following changes to the sdkconfig file. The settings that were manually changed in menuconfig are `CONFIG_SPIRAM=y` and `CONFIG_FREERTOS_UNICORE is not set`. The rest seem to be automatically updated based on those two.

`diff --git a/sdkconfig b/sdkconfig
index 4150717..d87e4f3 100644
--- a/sdkconfig-debug
+++ b/sdkconfig-debug
@@ -648,10 +648,11 @@ CONFIG_ESP32_UNIVERSAL_MAC_ADDRESSES=4
 #
 # Sleep Config
 #

-CONFIG_ESP_SLEEP_POWER_DOWN_FLASH=y
 CONFIG_ESP_SLEEP_RTC_BUS_ISO_WORKAROUND=y
 # CONFIG_ESP_SLEEP_GPIO_RESET_WORKAROUND is not set
+CONFIG_ESP_SLEEP_PSRAM_LEAKAGE_WORKAROUND=y
 # CONFIG_ESP_SLEEP_FLASH_LEAKAGE_WORKAROUND is not set

+# CONFIG_ESP_SLEEP_MSPI_NEED_ALL_IO_PU is not set
 CONFIG_ESP_SLEEP_DEEP_SLEEP_WAKEUP_DELAY=2000
 # end of Sleep Config
 
@@ -726,7 +727,63 @@ CONFIG_ESP_PHY_REDUCE_TX_POWER=y
 #
 # ESP PSRAM
 #
-# CONFIG_SPIRAM is not set
[... 284 lines cut ...]
I (753) cpu_start: Pro cpu up.
I (756) cpu_start: Starting app cpu, entry point is 0x400818c4
0x400818c4: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401022b4: bad00bad bad00bad bad00bad
0x401022b4: bootloader_flash_read at C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader_support/bootloader_flash/src/bootloader_flash.c:92

Core  1 register dump:
PC      : 0x401022ba  PS      : 0x00060030  A0      : 0x80007c18  A1      : 0x3ffe7d90
0x401022ba: bootloader_init_mem at C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader_support/src/bootloader_mem.c:16

A2      : 0x00000000  A3      : 0x40006840  A4      : 0x00000044  A5      : 0x3ffe7e04
A6      : 0x3ff9eae0  A7      : 0x3ff9eac9  A8      : 0x800818e9  A9      : 0x3ffe7d70
A10     : 0x40000de8  A11     : 0x40000de8  A12     : 0x3ffe4358  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к� [... cut ...] �к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к�к��
```

### More Information.

Our project is using multiple components: Secure Boot v2, encrypted flash, encrypted NVS partition, FAT partition, PSRAM, Ethernet, WiFi, lwIP, MQTT, Modbus, and OTA firmware updates. The crash shown above seems to happen very early before it reaches any of our actual application code in app_main().

If the bootloader is rebuilt

derrick-senva, 2023-06-27:
Hi @KonstantinKondrashov,

Just following up on this issue to see if it's still worth pursing a solution for. Would you agree with the expected behavior described in my original post? Or is there something about the design/architecture of the ESP32/ESP-IDF that would make this an invalid configuration in the first place?

> A 2nd stage bootloader built with single-core mode and PSRAM disabled should be able to boot into application firmware built with dual-core mode and PSRAM enabled.

KonstantinKondrashov, 2023-06-28:
Hi @derrick-senva!
There was a similar issue linked with  "Unicore bootloader + multi core app" -  https://github.com/espressif/esp-idf/issues/10714.
Please try this fix https://github.com/espressif/esp-idf/commit/861a5fb86301bc49f272f9b1e9a82c49ba897168, it should help for your case.

KonstantinKondrashov, 2023-07-03:
Hi @derrick-senva!
Could you confirm that the fix works for your case as well?

KonstantinKondrashov, 2023-07-04:
I close this issue for now. Feel free to re-open if the fix did not help your case.

closed 2023-07-04 by KonstantinKondrashov