espressif/esp-idf #11618: Undefined Reference to "..." - collect2: error: ld returned 1 exit status

parksj10, 2023-06-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect the project to build

### What is the actual behavior?

I get ld link errors

### Steps to reproduce.

```cmake
#  ./CMakeLists.txt
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

set(EXTRA_COMPONENT_DIRS 
    ${CMAKE_CURRENT_LIST_DIR}/hw/main/ 
    ${CMAKE_CURRENT_LIST_DIR}/hw/components/ 
    ${CMAKE_CURRENT_LIST_DIR}/hw/libs/esp32-arduino-lib-builder/components/arduino_tinyusb
    ${CMAKE_CURRENT_LIST_DIR}/hw/libs/Adafruit_NeoPixel
)
project(vsl)
```
------------
```cmake
# ./hw/main/CMakeLists.txt
set(requires arduino arduino_tinyusb Adafruit_NeoPixel blemesh_console fatfs)

set(CMAKE_EXE_LINKER_FLAGS "-Wl -t")

idf_component_register(
    SRCS "main.cpp"
    INCLUDE_DIRS "."
    REQUIRES "${requires}"
)
```
------------
```cmake
# ./hw/components/blemesh_console/CMakeLists.txt
set(srcs "ble_mesh_adapter.c"
        "ble_mesh_model.c"
        "ble_mesh_console_lib.c"
        # "ble_mesh_console_main.c"
        "ble_mesh_console_system.c"
        "ble_mesh_register_cmd.c"
[... 192 lines cut ...]
[4/6] Linking CXX executable vsl.elf
FAILED: vsl.elf 
: && /Users/josh/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address  -g  CMakeFiles/vsl.elf.dir/project_elf_src_esp32.c.obj -o vsl.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/arduino/libarduino.a  esp-idf/Adafruit_NeoPixel/libAdafruit_NeoPixel.a  esp-idf/ble_mesh_init/libble_mesh_init.a  esp-idf/blemesh_console/libblemesh_console.a  esp-idf/main/libmain.a  -Wl,--cref  -Wl,--Map="/Users/josh/Code/hw/la/build/vsl.map"  -Wl,--gc-sections  -fno-rtti  -fno-lto  esp-idf/unity/libunity.a  esp-idf/Adafruit_NeoPixel/libAdafruit_NeoPixel.a  esp-idf/arduino/libarduino.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/json/libjson.a  esp-idf/blemesh_console/libblemesh_console.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/ble_mesh_init/libble_mesh_init.a  esp-idf/bt/libbt.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/ulp/libulp.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /Users/josh/esp/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a  /Users/josh/esp/esp-idf/components/esp_wifi/lib/esp32/libcore.a  /Users/josh/esp/esp-idf/components/esp_wifi/lib/esp32/libespnow.a  /Users/josh/esp/esp-idf/components/esp_wifi/lib/esp32/libmesh.a  /Users/josh/esp/esp-idf/components/esp_wifi/lib/esp32/libnet80211.a  /Users/josh/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a  /Users/josh/esp/esp-idf/components/esp_wifi/lib/esp32/libsmartconfig.a  /Users/josh/esp/esp-idf/components/esp_wifi/lib/esp32/libwapi.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_ [... cut ...]
/Users/josh/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/josh/Code/hw/la/hw/main/main.cpp:162: undefined reference to `ble_mesh_register_gen_onoff_client()'
/Users/josh/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/josh/Code/hw/la/hw/main/main.cpp:162: undefined reference to `ble_mesh_register_configuration_client_model()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
```

Sherry616, 2023-06-08:
Hi @parksj10,
"ld returned 1 exit status" indicates a compilation error. It seems that the code has not been successfully compiled yet.
Typically, our development process, including the release and GitHub versions, ensures that the code is thoroughly tested and compiles without any issues. 
However, considering the recent change in the CMakeLists.txt file, it might be necessary for you to investigate the root cause on your own first.

igrr, 2023-06-08:
The issue likely occurs because your main file is a C++ source file, while the implementation of these functions is done in C source files. In this case, you have to inform the C++ compiler that the functions are declared with C linkage.
Please try adding `#ifdef __cplusplus`/`extern "C" {`/`#endif` guards in your header file. Otherwise the C++ compiler will consider that the functions `register_system` and others are declared with C++ linkage.

For more information about combining C and C++ code, please see the docs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/cplusplus.html#combining-c-and-c-code.

parksj10, 2023-06-08:
@igrr you tha man! thanks! for reference for anyone else with this issue, this is the implementation of @igrr 's fix

```c
// ./hw/components/blemesh_console/ble_mesh_console_decl.h
/* Console example — declarations of command registration functions.

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#pragma once

#ifdef __cplusplus
extern "C"
{
#endif

#include "esp_ble_mesh_defs.h"

   // Register system functions
   void register_system(void);

   // Register blutooth
   void register_bluetooth(void);

   // Register mesh node cmd
   void ble_mesh_register_mesh_node(void);

   // Register Test Perf client cmd
   void ble_mesh_register_mesh_test_performance_client(void);

#if (CONFIG_BLE_MESH_CFG_CLI)
   // Register mesh config client operation cmd
   void ble_mesh_register_configuration_client_model(void);
#endif

   // Register mesh config server and generic server operation cmd
   void ble_mesh_register_server(void);

#if (CONFIG_BLE_MESH_GENERIC_ONOFF_CLI)
   // Register mesh client operation cmd
   void ble_mesh_register_gen_onoff_client(void);
#endif

#if (CONFIG_BLE_MESH_CFG_CLI)
   // Register mesh config client operation cmd
   void ble_mesh_register_configuration_client_model(void);
#endif

#ifdef __cplusplus
}
#endif
```

parksj10, 2023-06-08:
@igrr out of curiosity, it's clear you have a ton of experience/expertise with this stuff, but how would someone such as myself know/figure out that this lack of C++ guard was causing the issue?

closed 2023-06-08 by parksj10

igrr, 2023-06-08:
It's not a guaranteed way (there's always a chance of some odd bug haha!) but trying to follow the [linker logic](https://eli.thegreenplace.net/2013/07/09/library-order-in-static-linking#the-linking-process) helps fairly often. Ask yourself questions like, "should the linker be able to find the library?", "is the function actually in the library?", "is the name of the function defined in the library same as the function being referenced?" and so on:

1. Determine the library ("component" in IDF) where the caller is. In this case, this is `main` (a.k.a. `hw/main/`)
2. Determine the library where the "missing" functions are supposed to be located. In this case, that is `blemesh_console`.
3. Is there a dependency between the two at CMake level? In this case — yes, there is: `blemesh_console` is in the `requires` list in `hw/main/CMakeLists.txt`. Also, most of the time if the dependency at CMake level is missing, you will get a compile time error that the header file is not found.
4. If there is a dependency at CMake level, then the library linking order is correct. What else can be wrong? At this point I would find the static library where the missing functions are supposed to be (`libblemesh_console.a`), and use objdump (`xtensa-esp32-elf-objdump`) to check whether the functions are actually there.
   - this might show that the whole object file is missing, if for instance you forgot to compile the .c file where the functions are
   - or the object file is there, but functions which are supposed to be inside are not there — could be because they were surrounded by some `#ifdef` which got preprocessed out
   - or the functions are there, but the names are mangled — something like `_Z15register_systemv` instead of `register_system` — this is the case you had
   - or the function is there, but there is a typo in the name of the function defined, compared to the function declared
   - possibly some other options, can't think of anything right now.

Hope this helps!

parksj10, 2023-06-09:
That's super helpful! Thanks so much for all the details and help! I had no idea about the `objdump` that's definitely gonna come in handy. I actually checked the build folder and saw the `_Z15Register_...` thing but I didn't realize it wasn't supposed to be like that

clovisf, 2024-07-22:
I was having the same issue, using Arduino IDE 2.3.2 and trying to compile an example with the BluetoothSerial library from here: https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial , for the DFRobot Beetle ESP32-C3 and also SeeedStudio Xiao ESP32-C6.

I added 
`#ifdef __cplusplus
extern "C"
{
#endif` 
to the main code (seen below) and now am getting the 
`exit status 1

Compilation error: conflicting declaration of 'void setup()' with 'C' linkage`
 error. 
Any help?
Complete code I am trying to compile on Arduino IDE:

`#include "BluetoothSerial.h"

#ifdef __cplusplus
extern "C"
{
#endif

BluetoothSerial SerialBT;
String MAC_Address;
 
void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32 Bluetooth");
}
 
void loop() {
  MAC_Address = SerialBT.getBtAddressString();
  Serial.println(MAC_Address.c_str());
  delay(500);
}`