espressif/esp-idf #11620: Cannot get the RS485 communication on ESP32C3's UART0 working using auto switching RS485 transmitter/receiver.

DicoderZ, 2023-06-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a custom board to read an RS485 device using UART0 of the ESP32C3, GPIO 20 for RX, 21 for TX. The board has auto switching RS485 transmitter/receiver built using SP3485. I have now spent days trying to troublshoot it with no success. So any hints will be really really helpful.

[Here is the schematic.](https://imgur.com/Ij1L2jr)

[this is adapted from the Espressif's documentation, ](https://imgur.com/RrP3sg8)
https://docs.espressif.com/projects/esp-idf/en/v3.3.6/api-reference/peripherals/uart.html

I couldn't manage to get any data. Same schematic with SP3485 works fine with RPI PICO w based board. I tested them side by side.

Here is the code:

#include "uart_handler.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_log.h"
#include "driver/uart.h"
#include "string.h"
#include "driver/gpio.h"

const uart_port_t uart_num = UART_NUM_0;
uart_config_t uart_config = {
.baud_rate = 38400,
.data_bits = UART_DATA_8_BITS,
.parity = UART_PARITY_DISABLE,
.stop_bits = UART_STOP_BITS_1,
.flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
//.rx_flow_ctrl_thresh = 122,
};
// Configure UART parameters
ESP_ERROR_CHECK(uart_param_config(uart_num, &uart_config));
// Set UART pins(TX: IO4, RX: IO5, RTS: IO18, CTS: IO19)
ESP_ERROR_CHECK(uart_set_pin(UART_NUM_0, 20, 21, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
// Setup UART in rs485 mode
ESP_ERROR_CHECK(uart_set_mode(uart_num, UART_MODE_RS485_APP_CTRL));
// ESP_ERROR_CHECK(uart_set_mode(uart_num, UART_RS485_CONF_REG.UART_RS485RXBY_TX_EN = 1));
// ESP_ERROR_CHECK(uart_set_mode(uart_num, , UART_RS485_CONF_REG.UART_RS485TX_RX_EN = 0));

// Setup UART buffered IO with event queue
const int uart_buffer_size = (1024 * 2);
QueueHandle_t uart_queue;
// Install UART driver using an event queue here
ESP_ERROR_CHECK(uart_driver_install(UART_NUM_0, uart_buffer_size, \
uart_buffer_size, 10, &uart_queue, 0));
void tx_task() {
// Write data to UART.
uart_flush(uart_num);
// uart_flush_input(uart_num);
char* test_str = "getData\n\r";
uart_write_bytes(uart_num, (const char*)test_str, strlen(test_str));
// Write data to UART, end with a break signal.
// uart_write_bytes_with_break(uart_num, const char*)test_str,strlen(test_str), 100);
vTaskDelay(pdMS_TO_TICKS(500));
}
void rx_task(char rxdata[RX_BUF_SIZE]) {
tx_task();
// Read data from UART.
const uart_port_t uart_num = UART_NUM_0;
uint8_t data[128];
int length = 0;
ESP_ERROR_CHECK(uart_get_buffered_data_len(uart_num, (size_t*)&length));
length = uart_read_bytes(uart_num, data, length, 100);
ESP_LOGI("UART HANDLER", "Read %d bytes: '%s'", length, (char*) data);
uart_flush(uart_num);

DicoderZ, 2023-06-12:
Hi,
I am tried all three now.
1. Changed UART0 to UART to ESP_CONSOLE_UART_NUM=1
2. yes
3. Tested with the example. 
It works now. Communicates with the RS485 device. 

Now, there is a weird problem. It works as long as the device is connected to PC with USB in monitor mode. As soon as I take the cable out. The RS485 reading part does not read. ESP32C3 device continues to work, and reads other sensors such as board temperature.

alisitsyn, 2023-06-13:
Hi,

I can recommend you to use the [galvanic isolation](https://www.ti.com/lit/an/slla416c/slla416c.pdf?ts=1686642124077&ref_url=https%253A%252F%252Fgithub.com%252Fespressif%252Fesp-idf%252Fissues%252F10781). The Schematic variant C supports the auto switching of direction and galvanic isolation as well it can solve many problems in your project.

alisitsyn, 2023-06-26:
@diwakrZha ,

Do you have any update on this issue? Thanks.

diwakrZha, 2023-06-27:
No, the problem still exists while using UART0. Reads actual values from RS485 as long the board is powered by USB data cable. Returns empty values, when restarting the board with USB power only cable. Checked the character lengths, for TX and RX, Board is sending correctly to the RS485 device, but the RX is empty. Cable's are thicker, and the supply is 5V 2.4A capable.

Thinking of ditching the UART0 completely and move to using UART1.

alisitsyn, 2023-08-21:
@diwakrZha ,

It is hard to check your issue with the same environment on my side. However, my board with different driver, suggested schematic in similar conditions works just fine. Please check and share the oscillogram for the communication and check the logic levels at the output of driver. Also check the configuration to use separate UARTs for console and for Modbus communication.

alisitsyn, 2023-09-14:
Hi @diwakrZha ,
Do you have any update on this issue? Thanks.

closed 2023-11-10 by dobairoland