espressif/esp-idf #11622: Pairing / bond client to reconnect automatically when it is advertising

TheodoreBardy, 2023-06-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm looking for a way to persistently save a peer's address.

The behavior I've had so far is as follows:
1. A BLE connection is established between the ESP32 and a server (remote control, for example).
2. Data can be exchanged. When I press a button on the remote control, the ESP32 prints a log.

This is fine, but if the ESP32 or the remote control is powered off, the connection step has to be repeated.

What I want is:
1. A BLE connection is made between the ESP32 and a server (remote control, for example).
2. Data can be exchanged. When I press a button on the remote control, the ESP32 prints a log.
3. Switch off remote control
4. Switch the remote on again and the ESP32 reconnects automatically.

Is there a way to do this?
Basically what can be done by the peer manager on the nRF SDK.

zhp0406, 2023-06-11:
1. As a client, ESP32 calls ”esp_ble_gap_start_scan“ to scan for the target device. After finding the device, it calls ”esp_ble_gap_stop_scan“ to stop scanning, then calls ”esp_ble_gap_set_security_param“ to enable bonding function, ”esp_ble_gattc_open“ to connect to the server, and ”esp_ble_set_encryption“ to pair and bond. Once bonded, the other device's MAC address will be automatically saved.

2. After connecting to the server, ESP32 calls ”esp_ble_gattc_register_for_notify“ to register for notifications and receive notifications from the device.

3. In the gattc_profile_event_handler callback function, when the ESP_GATTC_DISCONNECT_EVT event is triggered, the device is reconnected by:

   3.1 Calling esp_ble_gap_start_scanning to start scanning again.

   3.2 In the ESP_GAP_BLE_SCAN_RESULT_EVT event of esp_gap_cb, calling esp_ble_get_bond_device_list to obtain the list of bonded devices. If the scanned device is in the bond device list, call esp_ble_gattc_open to connect to it.
   
---
Thank you for raising the issue. You can use the above method for now, and we will provide simpler and more efficient APIs or methods to achieve this functionality in the future.

TheodoreBardy, 2023-06-16:
Hello, thank you for your reply @zhp0406.

I applied the sequence you described but it doesn't seem to work.
The peer is not saved after calling the `esp_ble_set_encryption` function. The return value of the function is `ESP_OK` but right after that, I checked the number of bonded devices by calling `esp_ble_get_bond_device_num` which is equal to 0.
Is there any way to force a MAC peer address to be saved?