espressif/esp-idf #11624: Bluetooth controller crash with IllegalInstruction from btdm_sleep_check_duration with BT modem sleep disabled

rockys-spindance, 2023-06-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2-279-ga6a962d5f8

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

DevkitC with ESP32-WROVER-E module

### Power Supply used.

USB

### What is the expected behavior?

I should be able to write advertising data to the nimBLE stack via a call to `ble_gap_adv_set_data()` while periodic BLE advertising is running without crashing the Bluetooth controller (libbtdm_app.c and/or bt.c).

### What is the actual behavior?

In our product application, we write advertising data relatively frequently while advertising is running.
This usually works, but infrequently we would see crashes with the following backtrace signature:

```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 369
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40091df8: f01d020c 00004136 f01d0000
0x40091df8: btdm_sleep_check_duration at C:/Espressif/frameworks/esp-idf-v5.0/components/bt/controller/esp32/bt.c:950
```
This line number is on exit from btdm_sleep_check_duration() which is apparently getting called by `libbtdm_app.a` code even though we have modem sleep disabled (`CONFIG_BTDM_CTRL_MODEM_SLEEP=n`).

The crashes would sometimes appear during connection / disconnection, or sometimes just spontaneously while connected for a while with ongoing advertising.

Neither the nimBLE nor Espressif documentation specifies limitations on when `ble_gap_adv_set_data()` can be called (except obviously after ble sync) and we never get an error code returned from the call.

### Steps to reproduce.

We have attached a simple example project (similar to `examples/bluetooth/nimble/bleprph`) that **reliably** reproduces the crash (or at least generates exactly the same proximate cause in the logs).

To reproduce

1. Copy the attached `ble_panic_demo.zip` file to the `examples/bluetooth/nimble` directory in your esp-idf release/v5.0 toolchain
2. Extract the contents there (you should end up with a directory  `examples/bluetooth/nimble/ble_panic_demo`
3. Navigate to the new directory
4. Build, flash, and run the project using the `idf.py` tools
5. Observe the console output.  You'll see logging every 100 milliseconds when the code writes advertising data.  Approximately 21 seconds after boot, you should see code reliably crash and show the backtrace mentioned above.

[ble_panic_demo.zip](https://github.com/espressif/esp-idf/files/11691612/ble_panic_demo.zip)

### Debug Logs.

```plain
Here is a log from the `ble_panic_demo` project showing the reproducible crash.
==================
[... 301 lines cut ...]

0x40086271: r_assert_param at ??:?

Backtrace: 0x40091dfc:0x3ffd0210 0x4008635e:0x3ffd0230 0x400e9a5d:0x3ffd0250 0x40019fb5:0x3ffd0270 0x4001a1f2:0x3ffd02a0 0x40019471:0x3ffd02c0 0x400175a7:0x3ffd0300 0x40019d11:0x3ffd0320 0x40055b4d:0x3ffd0340 0x400e9e27:0x3ffd0360 0x400ea465:0x3ffd0380 0x400969d1:0x3ffd03b0
0x40091dfc: r_assert at C:/Espressif/frameworks/esp-idf-v5.0/components/bt/controller/esp32/bt.c:1889

0x4008635e: r_assert_param at ??:?

0x400e9a5d: r_platform_reset at ??:?

0x400e9e27: r_rw_schedule at ??:?

0x400ea465: btdm_controller_task at ??:?

0x400969d1: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

### More Information.

The original intermittent crash we observed in our product looks very similar to the issue reported here
https://github.com/espressif/esp-idf/issues/10555.  But the most recent `libbtdm_app.a` in the comment thread there does not fix our issue.

Hopefully, the reproducible version from the `ble_panic_demo` project has the same root cause.

Per comments we included in `ble_panic_demo/main/app_main.c`, we _don't_ see the panic occur if the periodic call to `ble_gap_adv_set_data` is made directly from a timer context.  Only when it is called from a task context.

esp-cjh, 2023-06-14:
Hi, @rockys-spindance 
I have reproduced the problem using your example. We will fix it ASAP.

esp-cjh, 2023-06-15:
Hi, @rockys-spindance 
I have found out the reason. The adv interval is 625ms and you set adv data per 100ms. The adv data you set will take effect on the next adv event. So that you set adv data 6 times during an adv interval, but only the last time will take effect. The memory leak is caused by controller. I will fix it.

I suggest that the interval to set adv data should be less than adv interval, otherwise the adv data will be overwritten and this is useless. That would solve the problem too. You can set adv data per 1000ms.

esp-cjh, 2023-06-15:
Hi, @rockys-spindance 
You can use my lib to fix this problem. Unzip libbtdm_app.zip and replace it with esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a.

[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/11756782/libbtdm_app.zip)

rockys-spindance, 2023-06-19:
@esp-cjh Thanks for the quick response.  I have confirmed that the new library you provided fixes the issue in our example app when we run it here.  Still working to confirm if the fix also solves the intermittent panic we were observing in our actual application.  That will take us a while since that panic was relatively infrequent.

rockys-spindance, 2023-06-20:
@esp-cjh Here's an update from our side

### The good news
When built with the v5.0.2-279-ga6a962d5f8 esp-idf tools plus [ the new library you provided above](https://github.com/espressif/esp-idf/issues/11624#issuecomment-1592848541), our production code no longer shows the original panic.

### The not-so-good news
With the new library we now infrequently encounter two new panics in our production code. They occur anywhere between several minutes and several hours of operation.

1. The first shows the following backtrace:

> 	assert failed: ble_hs_hci_cmd_tx ble_hs_hci.c:324 (ble_hs_hci_ack == ((void *)0))
> 
> 
> 	Backtrace: 0x400823da:0x3ffbd170 0x40093699:0x3ffbd190 0x40098f99:0x3ffbd1b0 0x400e1c16:0x3ffbd2d0 0x400deaa5:0x3ffbd300 0x400d89f5:0x3ffbd340 0x400d8557:0x3ffbd3d0 0x40096775:0x3ffbd400 0x400967ad:0x3ffbd420 0x40096934:0x3ffbd450 0x40096a1d:0x3ffbd480
> 	0x400823da: panic_abort at C:/Espressif/frameworks/esp-idf-v5.0-2/components/esp_system/panic.c:424
> 
> 	0x40093699: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.0-2/components/esp_system/esp_system.c:153
> 
> 	0x40098f99: __assert_func at C:/Espressif/frameworks/esp-idf-v5.0-2/components/newlib/assert.c:78
> 
> 	0x400e1c16: ble_hs_hci_cmd_tx at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:324 (discriminator 1)
> 
> 	0x400deaa5: ble_gap_adv_set_data at C:/Espressif/frameworks/esp-idf-v5.0-2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:2827
> 
> 	0x400d89f5: ble_advertise at C:/ego/iot-infra8/code/projects/wifi_connector/main/app_main.c:248
> 
> 	0x400d8557: on_ble_adv_restart_timeout at C:/ego/iot-infra8/code/projects/wifi_connector/main/app_main.c:82
> 
> 	0x40096775: prvProcessExpiredTimer at C:/Espressif/frameworks/esp-idf-v5.0-2/components/freertos/FreeRTOS-Kernel/timers.c:572
> 
> 	0x400967ad: prvProcessTimerOrBlockTask at C:/Espressif/frameworks/esp-idf-v5.0-2/components/freertos/FreeRTOS-Kernel/timers.c:641
> 
> 	0x40096934: prvTimerTask at C:/Espressif/frameworks/esp-idf-v5.0-2/components/freertos/FreeRTOS-Kernel/timers.c:604 (discriminator 1)
> 
> 	0x40096a1d: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0-2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

		
That ble_gap.c line 2827 is inside ble_gap_adv_set_data().

Additional experimentation revealed that this panic occurs infrequently when a call to `ble_gap_adv_set_data` occurs between 
`ble_gap_adv_stop` and `ble_gap_adv_start` calls but only for very specific timing of the calls. There is no documented restriction on such a call, so we would definitely count this as a controller bug (or at least a documentation bug).  But it's an easy one to dodge by modifying our application code to disable `ble_gap_adv_set_data` calls between any `ble_gap_adv_stop` and `ble_gap_adv_start` calls. 

If it is of interest, we have another example project that reproduces this error on a DevkitC-WROVER-E and can share it.

**The second panic** shows a different backtrace, although one still related to setting the advertising data:

> 
> 			Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
> 
> 			Core  0 register dump:
> 			PC      : 0x4000c2e4  PS      : 0x00060c30  A0      : 0x8010d9a0  A1      : 0x3ffdba90
> 			A2      : 0x00000000  A3      : 0x3ffb30d4  A4      : 0x0000002c  A5      : 0x00000000
> 			A6      : 0xffffffff  A7      : 0x00050805  A8      : 0x00000000  A9      : 0x00000000
> 			A10     : 0x3ffb53c8  A11     : 0x00060323  A12     : 0x80083ff7  A13     : 0x3ffc0b00
> 			A14     : 0x00060323  A15     : 0x3ffb290c  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d
> 			EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000001
> 
> 
> 			Backtrace: 0x4000c2e1:0x3ffdba90 0x4010d99d:0x3ffdbaa0 0x4004c94d:0x3ffdbac0 0x400873fc:0x3ffdbae0 0x40019d11:0x3ffdbb10 0x40055b4d:0x3ffdbb30 0x400fffcb:0x3ffdbb50 0x40100629:0x3ffdbb70 0x4009876d:0x3ffdbba0
> 			0x4010d99d: hci_le_set_adv_data_cmd_handler at llm_hci.c:?
> 
> 			0x400873fc: ke_task_schedule at ke_task.c:?
> 
> 			0x400fffcb: r_rw_schedule at ??:?
> 
> 			0x40100629: btdm_controller_task at ??:?
> 
> 			0x4009876d: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0-2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

This panic occurs during steady-state advertising (no stop/start in the vicinity) independent of whether BLE is connected or not.  
But other than that, we have not been able to narrow down conditions that trigger it or find ways to reproduce it.

@esp-cjh We have two questions

1. Does the backtrace for the second panic above suggest anything to you?  Any pointers from that call tree inside the controller might suggest things we can try to identify the trigger condition or reproduce the panic more reliably.
2. Are these controller issues inside nimBLE code or only inside theESP32 port?  We are looking for ways we can help debug.

rockys-spindance, 2023-06-22:
@esp-cjh 
We are attaching a new example app here 
[ble_panic_demo-2.zip](https://github.com/espressif/esp-idf/files/11840591/ble_panic_demo-2.zip)
that reproduces the second ble controller panic (inside `hci_le_set_adv_data_cmd_handler` ) described in the [previous comment](https://github.com/espressif/esp-idf/issues/11624#issuecomment-1599569222).

Unfortunately, the reproducibility rate is not 100%.  Typically if we run multiple ~10 minute long runs, we see the crash in about 25% of the runs. That's not great for debugging, but it's a lot better than the one such crash every hour or two that we see in our production code.

The example app just turns on BLE advertising with a 10 ms interval and then writes advertising data to the stack every 10 ms with a little random delay to alter the timing of the write relative to the ad interval.

We built it with the v5.0.2-279-ga6a962d5f8 esp-idf tools plus [your library from above](https://github.com/espressif/esp-idf/issues/11624#issuecomment-1592848541) .

To reproduce the crash on a DevkitC-WROVER-E:

1. Copy the attached ble_panic_demo-2.zip file to the examples/bluetooth/nimble directory in your esp-idf release/v5.0 toolchain

1. Extract the contents there (you should end up with a directory examples/bluetooth/nimble/ble_panic_demo-2)

1. Navigate to the new directory

1. Build, flash, and run the project using the idf.py tools

1. if no crash is observed in the first 10-12 minutes then power cycle the devkit and try again.  We're seeing a crash in about 25% of such runs.

esp-cjh, 2023-06-25:
@rockys-spindance
OK, I will reproduce the problem and find out the cause.

esp-cjh, 2023-06-28:
Hi, @rockys-spindance 
Please use this lib to test. I have test and didn't reproduce the crash.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/11889625/libbtdm_app.zip)

rockys-spindance, 2023-06-28:
Thanks @esp-cjh !  We will test this right away and let you know how it goes.

rockys-spindance, 2023-06-28:
@esp-cjh 
Alas, we started testing with the latest library you provided in [your comment above](https://github.com/espressif/esp-idf/issues/11624#issuecomment-1610654709) by running the ble_panic_demo-2 example project.  We're still using esp-idf from the release/v5.0 branch (v5.0.2-285-g7493b907ef) plus your new library.

Almost every run hits a spontaneous reset at some point typically within a few minutes.  No panic or error logging of any kind, just a reboot.  Here's a log snippet of one such reset:
```
I (101807) app_main: write data delay     2495  phase    6
I (101817) app_main: write data delay     2496  phase    7
I (101827) app_main: write data delay     2497  phase    6
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
...
```
Note that because there's no panic, there's no print-and-halt.  So it's easy to miss these resets.

Are you able to see these spontaneous resets there?  For us, four runs of ble_panic_demo-2 out of 5 attempts hit the reset, while one run lasted more than 10 minutes with no reset.

esp-cjh, 2023-06-28:
@rockys-spindance 
Yes, I have noticed the restart. This is because you invoke `esp_restart()` in your code.
[image]

rockys-spindance, 2023-06-28:
@esp-cjh Apologies.  I forgot that was in there.  The intent of the original code was to vary timings, so we were periodically resetting in case some of the hardware bring-up was adding variable timings.
I'll continue testing.

rockys-spindance, 2023-06-29:
@esp-cjh  Our testing with the latest `libbtdm_app.a` from https://github.com/espressif/esp-idf/issues/11624#issuecomment-1610654709 is looking very promising.

With the new library, we've run the `ble_panic_demo_2` project for more than 7 hours across 17 runs with no panics at all.
And we've run our production code for more than 17 hours with no panics at all.

We'll continue testing today as well (given the intermittent nature for some of the original panic observations), but at this point it looks like your latest library fixes all the previous panics.

Alvin1Zhang, 2023-07-17:
Thanks for reporting, will close due to short of feedback, feel free to reopen. Thanks.

closed 2023-07-17 by Alvin1Zhang

rockys-spindance, 2023-07-17:
@Alvin1Zhang Is there a plan to merge @esp-cjh's fixes for the `libbtdm_app.a` library into the release/v5.0 branch at some point?  And if so, is there a way we can track when that happens?

Alvin1Zhang, 2023-07-18:
@rockys-spindance Sorry that I didn't share the details earlier, the fix is on release/5.0 available at https://github.com/espressif/esp-idf/commit/d3031ddddca242ee36d621a450a69203296b9e56. Thanks.