espressif/esp-idf #11627: System crashes when we try to enter deep sleep after terminating the BLE( Properly Terminating the BLE as per docs )

AbIoTSolutionsPrivateLimited, 2023-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.0

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

The system must enter the deep sleep without any crash.

1. We first terminate the ble using the following commands as mentioned in the docs:
2. `esp_bluedroid_disable();
        esp_bluedroid_deinit();
        esp_bt_controller_disable();
        esp_bt_controller_deinit();`
3. Then after a small delay of few seconds enter in deep sleep mode.
4. Now the system must disable ble and enter the deep sleep.

### What is the actual behavior?

The system crashes at an random place:
0x40375aca: panic_abort at C:/Users/suhas/esp/esp-idf/components/esp_system/panic.c:452

0x40381a75: esp_system_abort at C:/Users/suhas/esp/esp-idf/components/esp_system/port/esp_system_chip.c:90

0x403876ea: abort at C:/Users/suhas/esp/esp-idf/components/newlib/abort.c:38

0x4037705b: lock_acquire_generic at C:/Users/suhas/esp/esp-idf/components/newlib/locks.c:130

0x40377171: _lock_acquire_recursive at C:/Users/suhas/esp/esp-idf/components/newlib/locks.c:158

0x40377240: __retarget_lock_acquire_recursive at C:/Users/suhas/esp/esp-idf/components/newlib/locks.c:314 (discriminator 3)

0x4208dde1: __env_lock at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdlib/envlock.c:43

0x4208e871: _findenv_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdlib/getenv_r.c:84

0x4208e8f5: _getenv_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdlib/getenv_r.c:125

0x4208927f: _tzset_unlocked_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/time/tzset_r.c:28

0x4208924d: _tzset_unlocked at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/time/tzset.c:115

0x4207d90c: localtime_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/time/lcltime_r.c:34 (discriminator 8)

0x42097908: esp_log_system_timestamp at C:/Users/suhas/esp/esp-idf/components/log/log_freertos.c:84
[... 263 lines cut ...]
I (314) cpu_start: ESP-IDF:          v5.2-dev-974-g9f9bd8c033
I (321) cpu_start: Min chip rev:     v0.0
I (325) cpu_start: Max chip rev:     v0.99 
I (330) cpu_start: Chip rev:         v0.1
I (335) heap_init: Initializing. RAM available for dynamic allocation:
I (342) heap_init: At 3FCA2760 len 00046FB0 (283 KiB): DRAM
I (348) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (355) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (361) heap_init: At 600FE070 len 00001F90 (7 KiB): RTCRAM
I (368) spi_flash: detected chip: mxic (opi)
I (372) spi_flash: flash io: opi_str
I (377) sleep: Configure to isolate all GPIO pins in sleep state
I (383) sleep: Enable automatic switching of GPIO sleep configuration
I (390) coexist: coex firmware version: ebddf30
I (396) coexist: coexist rom version e7ae62f
I (401) app_start: Starting scheduler on CPU0
I (0) app_start: Starting scheduler on CPU1
I (14:00:36.314) main_task: Started on CPU0
I (14:00:36.324) main_task: Calling app_main()
W (14:00:36.328) NM52_WAKE: Reset Cause: Exception/Panic
W (14:00:36.334) NM52_WAKE: Wakeup Cause: Undefined
W (14:00:36.340) MAIN: Unknown Wakep Source!
I (13:59:30.011) NM52_SETTINGS: nm52_settings_nvs_init() OK
I (13:59:30.012) NM52_SETTINGS_NVS: Stats for partition:
```

patrickmagee, 2023-06-09:
I am also working on this project. To add some info:
- Crash does not happen on light sleep, or without sleep.
- If we do not call esp_bluedroid_disable(); esp_bluedroid_deinit(); esp_bt_controller_disable(); esp_bt_controller_deinit(); then we can enter and wake deep sleep without any issue. However the device will crash if we attempt to enable BT again, even after deep sleep.

KaeLL, 2023-06-09:
I think #10649 was supposed to have fixed this error.

AbIoTSolutionsPrivateLimited, 2023-06-09:
Hi KaeLL, i tried this with master branch and with latest commit. The issue still exists.

igrr, 2023-06-09:
> I think https://github.com/espressif/esp-idf/pull/10649 was supposed to have fixed this error.

While the backtrace does look similar, I think the root cause is different in this case. This is because an RTOS mutex is being acquired inside a critical section, due to an ESP_LOGE call, which in turn is due to a recent regression which we will soon fix!

AbIoTSolutionsPrivateLimited, 2023-06-09:
Thank you @igrr.

patrickmagee, 2023-06-10:
Is it known in which commit the regression happened? 
Is there a 5.2 commit that does not have this issue?

ESP-Marius, 2023-06-10:
> Is it known in which commit the regression happened?

Looks like it might have been e304db558b8d20bc20150dede2f035d12e39e2dd

closed 2023-06-14 by espressif-bot