espressif/esp-idf #11628: About esp32-s3 sdmmc write speed

Wanderer-web, 2023-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I am testing my own esp32-s3 pcb , it has a 4-line sd card slot. The esp32-s3 datasheet indicates that esp32-s3 sdmmc frequency is up to 40MHz, so the theoretical max sdmmc write speed for 4-line mode might be 160Mbps? But I wrote a program to test the writing speed and was frustrated to find that the write speed was 4Mbps, which is much slower. So I want to know the benchmark of esp32-s3 sdmmc write speed.

**Here is my environment:**

- esp-idf version : 5.0.2
- chip: esp32-s3r8
- flash: 16MB

**Here is the code:**

```c
#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>
#include <inttypes.h>
#include "soc/soc_caps.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"

#include "esp_log.h"
#include "esp_attr.h"
#include "esp_timer.h"

#include "driver/sdmmc_host.h"

#include "sdkconfig.h"
#include "sdmmc_cmd.h"

#define PIN_NUM_D0 (48)
#define PIN_NUM_D1 (47)
#define PIN_NUM_D2 (41)
#define PIN_NUM_D3 (38)
#define PIN_NUM_CMD (40)
#define PIN_NUM_CLK (39)

static const char *TAG = "app_main";

void app_main()
{
    esp_err_t err = ESP_OK;
    /* Probe */
    sdmmc_host_t config = SDMMC_HOST_DEFAULT();
    ESP_LOGI(TAG, "Probe using SD 4-bit...\n");
    config.flags = SDMMC_HOST_FLAG_4BIT;

    config.max_freq_khz = SDMMC_FREQ_DEFAULT;

    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();

    slot_config.width = 4;
    slot_config.clk = PIN_NUM_CLK;
    slot_config.cmd = PIN_NUM_CMD;
    slot_config.d0 = PIN_NUM_D0;
    slot_config.d1 = PIN_NUM_D1;
    slot_config.d2 = PIN_NUM_D2;
    slot_config.d3 = PIN_NUM_D3;

    err = sdmmc_host_init();
    ESP_ERROR_CHECK(err);

    err = sdmmc_host_init_slot(SDMMC_HOST_SLOT_1, &slot_config);
    ESP_ERROR_CHECK(err);

    sdmmc_card_t *card = (sdmmc_card_t *)malloc(sizeof(sdmmc_card_t));
    for (;;)
    {
        if (sdmmc_card_init(&config, card) == ESP_OK)
        {
            break;
        }
        ESP_LOGW(TAG, "slave init failed, retry...");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    sdmmc_card_print_info(stdout, card);

    gpio_pullup_en(PIN_NUM_CMD);
    gpio_pulldown_dis(PIN_NUM_CMD);
    gpio_pullup_en(PIN_NUM_CLK);
    gpio_pulldown_dis(PIN_NUM_CLK);
    gpio_pullup_en(PIN_NUM_D0);
    gpio_pulldown_dis(PIN_NUM_D0);
    gpio_pullup_en(PIN_NUM_D1);
    gpio_pulldown_dis(PIN_NUM_D1);
    gpio_pullup_en(PIN_NUM_D2);
    gpio_pulldown_dis(PIN_NUM_D2);
    gpio_pullup_en(PIN_NUM_D3);
    gpio_pulldown_dis(PIN_NUM_D3);

    size_t sector_count = 40;
    size_t sector_size = 512;
    char *sdio_write_buffer_str = (char *)malloc(sector_count * sector_size);
    assert(sdio_write_buffer_str != NULL);
    memset(sdio_write_buffer_str, 0, (sector_count * sector_size));
    int64_t start_time = esp_timer_get_time();
    ESP_ERROR_CHECK(sdmmc_write_sectors(card, sdio_write_buffer_str, 11, sector_count));
    ESP_LOGI(TAG, "sdio write %dB take %ldus, speed: %ldMbps", (sector_count * sector_size), (int32_t)(esp_timer_get_time() - start_time), (int32_t)(sector_count * sector_size * 8 / (int32_t)(esp_timer_get_time() - start_time)));
    free(sdio_write_buffer_str);
}
```

**Here is my run log:**

```
[0;32mI (715) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (722) heap_init: At 3FC95998 len 00053D78 (335 KiB): DRAM[0m
[0;32mI (729) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM[0m
[0;32mI (735) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (741) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM[0m
[0;32mI (748) esp_psram: Adding pool of 8000K of PSRAM memory to heap allocator[0m
[0;32mI (755) spi_flash: detected chip: winbond[0m
[0;32mI (760) spi_flash: flash io: qio[0m
[0;32mI (764) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (784) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (784) app_main: Probe using SD 4-bit...
[0m
[0;32mI (794) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (804) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (814) gpio: GPIO[48]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (824) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (834) gpio: GPIO[41]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (844) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (904) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
Name: SD32G
Type: SDHC/SDXC
Speed: 20 MHz
Size: 30436MB
CSD: ver=2, sector_size=512, capacity=62333952 read_bl_len=9
SSR: bus_width=4
[0;32mI (954) app_main: sdio write 20480B take 39902us, speed: 4Mbps[0m
```

Here is my sd slot schematic:
[image]

The sd connection has been processed with equal length

adokitkat, 2023-06-12:
Hello. I am getting similar speeds (5Mbps) as you when using the default 20MHz SDMMC frequency. However when using 40MHz (`SDMMC_FREQ_HIGHSPEED`), the speed increases greatly (using your code example). Try to change `config.max_freq_khz ` like this:

```c
config.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
```

My output:

```log
Name: EB1QT
Type: SDHC/SDXC
Speed: 40.00 MHz (limit: 40.00 MHz)
Size: 30528MB
CSD: ver=2, sector_size=512, capacity=62521344 read_bl_len=9
SSR: bus_width=4
I (442) app_main: sdio write 20480B take 2134us, speed: 76Mbps
I (452) app_main: sdio write 20480B take 2057us, speed: 79Mbps
I (452) app_main: sdio write 20480B take 2043us, speed: 80Mbps
I (462) main_task: Returned from app_main()
```

adokitkat, 2023-06-12:
Okay so I found out something interesting and now I am not sure how to handle the situation, this will need a further exploration. For some reason as I was switching branches in ESP-IDF the issue of slow writes disappeared even on default 20MHz freq. I literally cannot reproduce the 5Mbps I was getting at first, now I get around 50Mbps on default and 80Mbps on highspeed...

At first I was using `master` branch version of ESP-IDF on ESP32-S3-USB-OTG devkit board.

Can you try to go inside your esp-idf folder, git checkout the branch to `release/v5.0` and/or `release/v5.1`, run `./install.sh` and `. ./export.sh`, then in the project folder run `idf.py erase-flash`, `idf.py fullclean` and `idf.py build flash monitor`? You can also try to set `config.max_freq_khz = SDMMC_FREQ_HIGHSPEED;` in the code as I told you before.

Wanderer-web, 2023-06-13:
> Okay so I found out something interesting and now I am not sure how to handle the situation, this will need a further exploration. For some reason as I was switching branches in ESP-IDF the issue of [...]

Hello, @adokitkat. I seem to have found the reason. When I forbid esp32-s3 to use external spi ram, my program test write speed can reach 40Mbps under 40Mhz clk. Maybe there is bandwidth crowding between spi ram and sdmmc? Can this problem be solved in esp-idf from the software level? Or is this just a hardware limitation?

adokitkat, 2023-06-13:
Could you please provide the whole project folder as a zip file here?

Wanderer-web, 2023-06-13:
sure, `idf.py --version` output `ESP-IDF v5.0.2-285-g7493b907ef`.

The compressed package contains two projects, one of which enables the external spi ram, but the programs of the two are the same, and the running results are different. The sdmmc write speed of the project with external spi ram is much slower than that of the project without external spi ram
[sdmmc_test.zip](https://github.com/espressif/esp-idf/files/11735310/sdmmc_test.zip)

adokitkat, 2023-06-15:
Unfortunately right now I do not have ESP32-S3 hardware with both 4-bit SDMMC and PSRAM connected so I cannot progress much further. I am not sure why at first I got similar result and after that until now I am getting correct speeds. So do you have PSRAM on your custom board as well? What pins does it use? Does is share some pins with SDMMC?

adokitkat, 2023-06-15:
Maybe this applies in your case: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/external-ram.html#restrictions

As I said, I do not have access to a hardware similar to yours right now, so I am only guessing but you can try to:

- Change SPI access method: `Component config → ESP PSRAM → Support for external, SPI-connected RAM → SPI RAM config → SPI RAM access method` to `Make RAM allocatable using heap_caps_malloc(..., MALLOC_CAP_SPIRAM)` so it won't be accessible by regular `malloc()`

- Enable experimental features in `idf.py menuconfig` and set PSRAM clock speed to 120Hz: `Component config → ESP PSRAM → Support for external, SPI-connected RAM → SPI RAM config → Set RAM clock speed`. This have it's own gotchas: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#flash-psram-combination

Wanderer-web, 2023-06-15:
> Maybe this applies in your case: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/external-ram.html#restrictions
> 
> You are using GPIO pins 47 and 48 which are SPICLK_P_DIFF and SPICLK_N_DIFF pins, `Serial peripheral interface differential clock negative/positive for SPI bus`. https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf
> 
> As I said, I do not have access to a hardware similar to yours right now, so I am only guessing but you can try to:
> 
> * Change PSRAM access method: `Component config → ESP PSRAM → Support for external, SPI-connected RAM → SPI RAM config → SPI RAM access method` to `Make RAM allocatable using heap_caps_malloc(..., MALLOC_CAP_SPIRAM)` so it won't be accessible by regular `malloc()`
> * Enable experimental features in `idf.py menuconfig` and set PSRAM clock speed to 120Hz: `Component config → ESP PSRAM → Support for external, SPI-connected RAM → SPI RAM config → Set RAM clock speed`. This have it's own gotchas: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#flash-psram-combination

Thanks, `Make RAM allocatable using heap_caps_malloc(..., MALLOC_CAP_SPIRAM)` works, I think after that I have to put all sd write buffers in internal ram instead of external ram.

closed 2023-06-15 by Wanderer-web

adokitkat, 2023-06-18:
Great, nice to see it helped.