espressif/esp-idf #11629: std::thread stack size configuration

Zhu-jiatong, 2023-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

According to the documentation, ESP32 supports `std::thread` in C++. However, in many cases, the stack size of a newly created thread in not enough for use. I do know that using the RTOS functions, the stack size could be configured when creating a new thread. 
I have also heard of the use of `pthreads` functions, but that does not seem to be compatible with std::thread, and that using `esp_pthread_set_cfg` will overwrite the global rule of new thread creation, whereas I am only looking for a solution to configure stack size for individual thread.
Please do let me know if I am wrong in the aforementioned statements.
So, is there a way to set stack size of a thread created using `std::thread`?

Thank you very much.

igrr, 2023-06-09:
esp_pthread_set_cfg sets a thread-local value which will be inherited by pthreads and std::threads created from the current thread/task. It shouldn't affect creation of threads by other tasks running at the same time.

Zhu-jiatong, 2023-06-10:
Thank you. However, what I meant was that I was trying to create multiple threads from the same main thread. By using esp_pthread_set_cfg​ would overwrite the thread creation configuration in the main (originating) thread. So, is there a way to create threads with custom stack sizes without overwriting any default rules? Thanks.

igrr, 2023-06-12:
I think you could do something like (pseudocode)
```
// save the old config
esp_pthread_cfg_t orig_cfg;
esp_pthread_get_cfg(&orig_cfg);

// set up the new config
esp_pthread_cfg_t new_cfg = { ... };
esp_pthread_set_cfg(&new_cfg);

// create a thread
auto t = std::thread(...);

// restore the old config
esp_pthread_set_cfg(&orig_cfg);
```

0xjakob, 2023-06-25:
@Zhu-jiatong Does @igrr's suggestion work for you?

Zhu-jiatong, 2023-06-26:
Yes. Thanks.

0xjakob, 2023-06-26:
@Zhu-jiatong Great to hear. We are closing this issue then. Feel free to re-open it if you encounter any problems directly related to the `std::thread` stack size configuration.

closed 2023-06-26 by 0xjakob