espressif/esp-idf #11630: ESP_GATTS_RESPONSE_EVT status always 0

fzboffice, 2023-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

if congested event coming，write response will send failed，I depend ESP_GATTS_RESPONSE_EVT status to resend rsp

### What is the actual behavior?

ESP_GATTS_RESPONSE_EVT status always equie 0

### Steps to reproduce.

1. Step
2. Step
3. Step
...

zhp0406, 2023-06-16:
@fzboffice Which chip are you using, and can you provide some logs? thanks.

fzboffice, 2023-06-16:
@zhp0406 Actually, this has nothing to do with any chips, please take a look esp-idf\components\bt\host\bluedroid\btc\profile\std\gatt\btc_gatts.c:689 
param.rsp.status = 0;