espressif/esp-idf #11631: [ESPC2][NimBLE]Keep the MTU value agreed with the client after the connection parameters are updated

BurnerK, 2023-06-09:
### Is your feature request related to a problem?

The ESPC2 module acts as the BLE server and the APP as the client. Currently, after the APP establishes a connection with the ESPC2 module, the `BLE_GAP_EVENT_CONN_UPDATE` event is triggered 2 or 3 times, resulting in the MTU agreement request sent by the APP being reset again after the connection is established. I now need a way to not reset the MTU value agreed with the client after the connection parameters are updated.

### Describe the solution you'd like.

Keep the MTU value agreed with the client after the connection parameters are updated

rahult-github, 2023-06-09:
MTU exchange / update and BLE_GAP_EVENT_CONN_UPDATE event are two separate events.  

CONN_UPDATE event is sent when connection parameters are updated. In your use case if this event is seen multiple times , it implies that connection parameters have been updated multiple times. Can you share a debug log and we can check if remote is performing multiple connection updates. 

Also, can you point out why MTU is getting updated in your case when connection update event is received ? Ideally, MTU value should be set when the first channel is created OR an explicit MTU exchange handshake happens. So, it would be good to know where Connection Update is changing MTU value.

BurnerK, 2023-06-14:
> MTU exchange / update and BLE_GAP_EVENT_CONN_UPDATE event are two separate events.
>
> CONN_UPDATE event is sent when connection parameters are updated. In your use case if this event is seen multiple [...]

Hi, this is my debug log, Figure 1 shows that after `connection establiished`, three `connection updated` events were triggered, and Figure 2 shows that the client sent an MTU request before the last `connection updated` event was triggered. In the ios App test, only one `connection updated` event was triggered after the connection was established, and then the MTU was exchanged, and the data sending and receiving test was normal.

BurnerK, 2023-06-14:
[image]
[image]

rahult-github, 2023-06-16:
Hi @BurnerK ,

 LE Connection update event is posted either when LE connection update command was issued by host OR if the connection parameters are updated following a request from peer device.  
 
 In the first snapshot , we do see 2 connection update events showing change in connection intervals, but that log is not sufficient to gauge if the request was initiated by host or remote.  The second snapshot does show MTU update event , but the following update event too is not associated by any command from host ( atleast log doesn't show host sending it). 
 
 Please let us know if you can get one debug enabled log . 

1. If you are NOT on master branch, please apply attached patch. This patch gives extra information.   This patch has to be applied in the $IDF_PATH/components/bt/host/nimble/nimble path. 
[0001-Add-debug-code-to-print-HCI-events-information.txt](https://github.com/espressif/esp-idf/files/11765931/0001-Add-debug-code-to-print-HCI-events-information.txt)

2. Once patch is applied, please enable debug logs as below and share the console log to debug this more. 

To enable debug logs:

idf.py menuconfig
Component config → Bluetooth → NimBLE Options → NimBLE Host log verbosity → Debug logs
Component config → Log output → Default log verbosity → Debug 

Thanks,
 Rahul

rahult-github, 2023-06-19:
Hi @BurnerK ,

Can you please double confirm that the patch provided in comment [here](https://github.com/espressif/esp-idf/issues/11631#issuecomment-1594078845)  has been applied.  Please share a log with the patch applied to get better information. 

With the patch applied and debug logs enabled,  we should see prints like below  to indicate events .

**D (12309) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e 
D (12309) NimBLE: subevent: 0x3
D (12309) NimBLE:** 

I (12309) NimBLE: connection updated; status=0 
I (12309) NimBLE: handle=1 our_ota_addr_type=0 our_ota_addr=

I checked the logs you shared. From logs, can see that we have NOT sent any connection update command . This points that remote is issuing all those connection update commands.

BurnerK, 2023-06-19:
> Hi @BurnerK ,
>
> Can you please double confirm that the patch provided in comment [here](https://github.com/espressif/esp-idf/issues/11631#issuecomment-1594078845) has been applied. Please share a log [...]

1. ok. I have put the patch file in [$IDF_PATH/components/bt/host/nimble/nimble path], but still no message like `NimBLE: subevent: 0x3` appears. Am I using it in the wrong way?
2. I now know it is the connection parameter update command sent by the remote, but I want the MTU value not to be reset when the parameter is updated, can I achieve that? Or I can only solve this problem by modifying the MTU sending timing of the remote client?

rahult-github, 2023-06-19:
Hi @BurnerK 

> ok. I have put the patch file in [$IDF_PATH/components/bt/host/nimble/nimble path], but still no message like NimBLE: subevent: 0x3 appears. Am I using it in the wrong way?

Looks like the change is not gone in. May be you can open the patch file and copy those few lines manually in your codebase. It just prints the events information when packet is received. 

> I now know it is the connection parameter update command sent by the remote, but I want the MTU value not to be reset when the parameter is updated, can I achieve that? Or I can only solve this problem by modifying the MTU sending timing of the remote client ?

Connection update and MTU exchange are mutually exclusive.  MTU updation has no dependancy on connection parameters getting updated. So, again, MTU exchange can be done by client ( in this case the remote ) . 

The print in your log shows that remote has modified the MTU and set to 247. 

> eyb_ble: mtu update event; conn_handle=1 cid=4 mtu=247

If you don't want remote to re-negotiate the MTU, may be you can set the initial mtu itself to 247 by calling the ble_att_set_preferred_mtu in your code during init process. Please try if this helps.

rahult-github, 2023-08-09:
Hi @BurnerK  , any updates ?

rahult-github, 2023-09-04:
Closing issue. Please feel free to reopen in case of further queries.

closed 2023-09-04 by rahult-github