espressif/esp-idf #11632: add regex function next to httpd_uri_match_wildcard

dannybackx, 2023-06-09:
### Is your feature request related to a problem?

The httpd_uri_match_wildcard is rather limited (and its limitations are poorly documented).
Suggestion so supplement or replace it with a slower but more functional function.

### Describe the solution you'd like.

/*
 * esp-idf function is very limited, call regex in this one
 */
#include <regex.h>
bool match_wildcard(const char *tmpl, const char *uri, size_t len) {
  regex_t r;

  ESP_LOGD(ws_tag, "%s(%s, %s, %d)", __FUNCTION__, tmpl, uri, len);
  if (regcomp(&r, tmpl, 0)) {
    ESP_LOGE(ws_tag, "%s: could not compile regex %s", __FUNCTION__, tmpl);
    return false;
  }
  if (regexec(&r, uri, 0, 0, 0) == 0) {
    regfree(&r);
    ESP_LOGD(ws_tag, "%s: match %s %s", __FUNCTION__, tmpl, uri);
    return true;
  }
  regfree(&r);
  ESP_LOGD(ws_tag, "%s: fail %s %s", __FUNCTION__, tmpl, uri);
  return false;
}

void StartWebServer(void) {
  esp_err_t     err;

  ESP_LOGI(ws_tag, "Starting web server ...");

  httpd_handle_t        webserver = NULL;
  httpd_config_t        wsconfig = HTTPD_DEFAULT_CONFIG();
  // wsconfig.uri_match_fn = httpd_uri_match_wildcard;
  wsconfig.uri_match_fn = match_wildcard;

### Describe alternatives you've considered.

You could move the regex_t members, to avoid spurious regcomp() calls. I've not done that in this version.

chipweinberger, 2023-06-09:
Imo i would just match on everything,

and then do your own regex and dispatch in your own handler.

you do not need to use httpd for this. do it yourself :)

dannybackx, 2023-06-09:
You can do everything yourself but the original httpd_uri_match_wildcard supports a surprisingly limited functionality. I found that replacing it with a more functional version was sooo simple that I'm suggesting this.

igrr, 2023-06-09:
It could be implemented as a new helper (`httpd_uri_match_regex`), however changing the behavior of the existing one could cause issues for existing projects.

dannybackx, 2023-06-09:
httpd_uri_match_regex looks perfect to me

hrushikesh430, 2025-02-27:
Hi @dannybackx, I just checked you alternative with all test cases, and it seems that it do not passes all the test cases. Could you please me your fully implemented solution, so that I can check it with all test cases.

dannybackx, 2025-02-27:
I gave you a sample implementation one and a half years ago, I expect you're better at professionalizing it than I am, so please do.