espressif/esp-idf #11633: ESP32 - Windows Setup Issue

ShizaShakeel-eaton, 2023-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi there,

      I am trying to setup the Espressif on Windows laptop. I am getting following error when I run the install.bat command:
[image]

I have made the following edit in the requirements.txt file:
[image]

After following these steps, I am still getting the same error. Can anyone please assist with this issue?

dobairoland, 2023-06-09:
Probably this is what you are looking for: https://github.com/espressif/esp-idf/blob/release/v4.4/requirements.txt

But be aware, v4.3 is in maintenance period, that means it is not recommended for new designs. If you can then consider using a newer version which is still in service period.

dobairoland, 2023-06-13:
I have double-checked that the fix has been backported to v4.3 as well: https://github.com/espressif/esp-idf/blob/release/v4.3/requirements.txt, here: https://github.com/espressif/esp-idf/commit/f9e52727279180f5f5ec4acb7996f01e92b40e06.

closed 2023-06-13 by dobairoland