espressif/esp-idf #11635: The ESP32 crashed with an assertion error (ASSERT_PARAM(0 0)) in lld_pdu.c at line 889 immediately after the completion of the pairing process.

spanDN, 2023-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The ESP32 BLE stack should be able to handle situations where the signal strength is low without experiencing crashes.

### What is the actual behavior?

The ESP32 crashed immediately after the pairing process was completed. We are utilizing both WIFI and BLE simultaneously. Our commercial device receives advertisements from different devices at a rate of two times per second. However, after the pairing with a device that had a weak signal, the ESP32 crashed.

### Steps to reproduce.

While the issue is not easy to reproduce, you would need to have a BLE device that advertises, and the ESP32 would need to complete the pairing process with a signal strength of -98 dBm.

### Debug Logs.

```plain
esptool.py --port /dev/ttyUSB1 chip_id

esptool.py v4.5.1
Serial port /dev/ttyUSB1
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: fc:f5:c4:6a:d8:d4
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: fc:f5:c4:6a:d8:d4
Hard resetting via RTS pin...

[0;36m93718126  [DBG ] [BTC_TASK] [004872] [CONN_HAN] [0;35mCreated event: PAIRING_COMPLETED, conn: FFFF, len:0, data:, 1[0m
[0;36m93718236  [DBG ] [BTC_TASK] [004872] [BLE_IMPL] update connection params status = 0, min_int = 0, max_int = 0,conn_int = 40,latency = 0, timeout = 300[0m
ASSERT_PARAM(0 0), in lld_pdu.c at line 889
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008615c: 920020c0 00000361 0020c000
Core  0 register dump:
PC      : 0x40086162  PS      : 0x00060334  A0      : 0x800865a1  A1      : 0x3ffc351c  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f41e71e  
A6      : 0x00000379  A7      : 0xfffffffc  A8      : 0x3ffc4a50  A9      : 0xace1e6f7  
A10     : 0x00000000  A11     : 0x3ffc34bf  A12     : 0x3ffc346b  A13     : 0x00000039  
A14     : 0x00000000  A15     : 0x3ffc3470  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400864a9  LEND    : 0x400864b1  LCOUNT  : 0x00000000  

Backtrace: 0x4008615f:0x3ffc351c |<-CORRUPTED
```

### More Information.

We are using ESP32 - WROVER  B module.

zhp0406, 2023-08-11:
@spanDN 
Has the issue been resolved?

spanDN, 2023-08-11:
@zhp0406 
Hi, Have there been any changes in the code? Which version should be installed to check? We found a workaround so as not to even attempt a connection if the signal level is low.

zhp0406, 2023-08-11:
@spanDN 
I am still trying to reproduce the issue. The log you provided is not very comprehensive, and I cannot find information such as the version of IDF and the version of the bt library. If you are still able to reproduce the issue, please provide me with the complete log, and it would be best if you could also give me the ELF file. I will do my best to help you resolve the problem as soon as possible.and thanks you for reporting this problem.

closed 2025-03-06 by Alvin1Zhang