espressif/esp-idf #11636: [BUG][PATCH]Delay in esp_timer task callback prevents dispatch of ESP_TIMER_ISR callbacks

acf-bwl, 2023-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Lolin D32, in-house board

### Power Supply used.

USB

### What is the expected behavior?

I would expect that callbacks dispatched as ESP_TIMER_ISR would run asynchronously to those dispatched as ESP_TIMER_TASK, so that even if a long delay were to occur in a callback in the timer task, the callbacks run in the ISR would continue to run unimpeded during this delay. Indeed, in esp_timer code, the esp_timer_impl abstraction layer appears to provide two separate alarms: one for the time task and one for the isr. One would expect that even when the timer task is busy, the separate ISR alarm would continue unaffected.

### What is the actual behavior?

Introducing a one second delay into a callback in the timer task causes all ISR callbacks to stop being serviced until the aforementioned timer task callback returns. This means no ISR-dispatched esp_timer s are serviced for around a second. This seems like it somewhat defeats the purpose of the ESP_TIMER_ISR dispatch method.

### Steps to reproduce.

Please build and run the attached code on ESP32. A copy of the main file is provided below for convenience. Note that the CONFIG_ESP_TIMER_SUPPORTS_ISR_DISPATCH_METHOD option must be enabled.

Note that following a print of "task callback", which prints with a period of 2 seconds, "isr callback" stops printing for some time prior to resuming its regular printing with a period of 200ms.

```C
#include "esp_log.h"
#include "esp_timer.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#define TAG "timer_issue"

void task_timer_cb(void *arg);
void isr_timer_cb(void *arg);

void app_main(void)
{
    esp_timer_handle_t task_timer_handle;
    esp_timer_handle_t isr_timer_handle;
    
    const esp_timer_create_args_t task_timer_args = {
        .callback = &task_timer_cb,
        .arg = NULL,
        .dispatch_method = ESP_TIMER_TASK,
        .name = "task_timer",
        .skip_unhandled_events = true,
    };
    
    const esp_timer_create_args_t isr_timer_args = {
        .callback = &isr_timer_cb,
[... 136 lines cut ...]
I (10519) timer_issue: isr callback
I (10519) timer_issue: task callback
I (11519) timer_issue: isr callback
I (11719) timer_issue: isr callback
I (11919) timer_issue: isr callback
I (12119) timer_issue: isr callback
I (12319) timer_issue: isr callback
I (12519) timer_issue: isr callback
I (12519) timer_issue: task callback
I (13519) timer_issue: isr callback

Done
```
```

### More Information.

This occurs because the LAC timer has only one compare register to use for both the timer task dispatch alarm and the ISR dispatch alarm. Both methods use the
```C
void IRAM_ATTR esp_timer_impl_set_alarm_id(uint64_t timestamp, unsigned alarm_id)```
function to set their timers, with alarm_id distinguishing between the two. The timestamps of both alarms are stored, and esp_timer_impl_set_alarm_id compares them and places the lesser of the two in the actual hardware timer compare register.

During the interrupt, however, the hardware timer compare register is never updated. The next update to the compare register comes *after* the next callback is finished running, whether in the ISR or in the timer task. This means that the timer compare register will continue to hold an outdated compare value, rather than the compare value for the upcoming ISR alarm, until the callback in the timer task is complete and the time task once again calls esp_timer_impl_set_alarm_id(). This scenario only occurs if this timer expiry was dispatched to the timer task and the next is destined to be dispatched to be run in the ISR, or perhaps theoretically the opposite.

This appears to affect the LAC esp_timer backend; I have not checked whether any of the others are affected. I have a patch for this bug for both esp-idf master and release/v4.4 which I will attach to this bug report as pull requests.

acf-bwl, 2023-06-09:
Pull requests are #11637 and #11638

KonstantinKondrashov, 2023-07-05:
Hi @acf-bwl!
Thanks for your attention to this case. I agree that if such a delay happends then following timers will starving. According to [the esp_timer doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html) **it is recommended to only do the minimal possible amount of work from the callback itself**. 
> ESP_TIMER_TASK. Timer callbacks are dispatched from a high-priority esp_timer task. Because all the callbacks are dispatched from the same task, it is recommended to only do the minimal possible amount of work from the callback itself, posting an event to a lower-priority task using a queue instead.

Yes, I confirm that your fix is able to fix a ceratin scenario. I will applay [your changes](https://github.com/espressif/esp-idf/pull/11637) with some addoptions.

closed 2023-07-05 by KonstantinKondrashov

acf-bwl, 2023-07-05:
great, thanks :)