espressif/esp-idf #11641: Ledc randomly invert PWM signal output

API-rep, 2023-06-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.4

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 dev board (WROOM-)

### Power Supply used.

USB

### What is the expected behavior?

I'm working on a small Arduino library to drive DC and servo motor via ledc PWM output. To go deeper in timer management, I write it with ESP32 IDF regular syntax and structure. I've noticed that randomly, PWM output is inverted, regardless of the port use. I supposed a hardware trouble, but this problem happened with other ESP32 dev kit module.

### What is the actual behavior?

During my research to fix this problem, I noticed that the Espressif ESP32 IDF have a **output_invert flag**_ on its ledc_channel_config_t struct. I tried to implement it on my code, but compilator return me an error at build. I'm not a c expert programmer, but I suspect a non initialization of this variable  who can cause this random problem ?

Trouble was fixed by adding a _**GPIO.func_out_sel_cfg[SERVO_PIN].inv_sel = 0**_ on my main code after PWM setup. 

### Steps to reproduce.

PWM setup code (cleaned for readability)
```

    // create timer config structure
_timers_config[_pwmTimer] = new ledc_timer_config_t;
			
    // seeding timer config parameters 
_timers_config[_pwmTimer]->speed_mode		= LEDC_HIGH_SPEED_MODE;		// set timer mode
_timers_config[_pwmTimer]->freq_hz		= pwmFreq;			// set frequency of PWM signal
_timers_config[_pwmTimer]->timer_num		= (ledc_timer_t)_pwmTimer;	// set timer index
_timers_config[_pwmTimer]->clk_cfg		= LEDC_AUTO_CLK;		// set LEDC source clock
_timers_config[_pwmTimer]->duty_resolution	= (ledc_timer_bit_t)resBit;	// set duty resolution

    // create PWM channel config structure
ledc_channel_config = new ledc_channel_config_t;

    // seeding PWM channel config parameters 
_ledc_channel_config->channel		= (ledc_channel_t)_pwmChannel;
_ledc_channel_config->duty		= 0;
_ledc_channel_config->hpoint		= 0;
_ledc_channel_config->gpio_num		= pwmPin;
_ledc_channel_config->speed_mode	= LEDC_HIGH_SPEED_MODE;
_ledc_channel_config->timer_sel		= (ledc_timer_t)_pwmTimer;
_ledc_channel_config->intr_type		= LEDC_INTR_DISABLE;
// _ledc_channel_config->output_invert	= 0;	// don't work on Arduino ESP32 IDE

    // start PWM signal on pwmPin.
ledc_channel_config(_ledc_channel_config);	

    // set ledc fade service on for ledc_set_duty_and_update/ledc_set_fade_time_and_start functions
ledc_fade_func_install(0);

...
    // output duty to port
ledc_set_duty_and_update(LEDC_HIGH_SPEED_MODE, (ledc_channel_t)_pwmChannel, dutyToWrite, 0);
```

### Debug Logs.

```plain
Error returned by compilator while output_invert struct flag is provided

error: 'struct ledc_channel_config_t' has no member named 'output_invert'
  _ledc_channel_config->output_invert = 0;
                        ^~~~~~~~~~~~~
exit status 1
```

### More Information.

Build via Arduino IDE

suda-morris, 2023-06-12:
@API-rep refer to this https://github.com/espressif/esp-idf/commit/48c848a895d0b9a5ffcaa3e0cd066407c0760da3

You may need to use `_ledc_channel_config->flags.output_invert = true;` to invert the output level.

API-rep, 2023-06-13:
Hello, thank you for your reply.

I've noticed this flag into the LEDC reference but, unfortunately, arduino ESP32 API don't support this flag and return an error at compilation. I first contact Aduino ESP team via the project github page to report this problem, but they  redirect me to ESP-IDF , assuming a bug on native IDF. Check [this post](https://github.com/espressif/arduino-esp32/issues/8306) for more information.

suda-morris, 2023-06-14:
can you check the `esp_idf_version.h` and tell me what's the version of your IDF? `ESP_IDF_VERSION_MAJOR`, `ESP_IDF_VERSION_MINOR` and `ESP_IDF_VERSION_PATCH`

closed 2023-06-16 by suda-morris