espressif/esp-idf #11642: a2DP sink : BT_AV re-start not supported after end.

0x0fe, 2023-06-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32D0WDR2

### Power Supply used.

USB

### What is the expected behavior?

The expected behaviour is that we can stop A2DP sink, to free ressource, and restart it, as many times as needed.
A ticket as already open here, where the author of this wrapper confirms it is an issue with the IDF API itself;
https://github.com/pschatzmann/ESP32-A2DP/issues/342

### What is the actual behavior?

Once the A2DP is ended, i cannot be restarted. The "advice" is to restart the ESP32 however this is not possible in my case, there are startup sound and UI effect plus a bunch of peripherals to setup, and the device often has to switch between A2DP sink and other source (SD_MMC, wifi stream) during normal operation, restarting all over is not an option.

### Steps to reproduce.

start A2DP sink
Stop A2DP sink
restart A2DP
(A2DP sink is dead until next reboot)

### Debug Logs.

```plain
none
```

BetterJincheng, 2023-06-12:
@0x0fe Do you want this?
init A2DP -> ... -> deinit A2DP -> ... -> init A2DP -> ... -> deinit A2DP ......

0x0fe, 2023-06-12:
@BetterJincheng  Yes exactly, freeing the ressources used by A2DP  each time it is deinit.
My problem is that i need to run the follow ing things : 
2 HTTPS clients for mp3 streaming and MP3 file downloading

0x0fe, 2023-06-12:
and there is something wrong with DMA too, here i tried to stop the A2DP sink without releasing the memory (which memory exacty, i dont know).

`a2dp_sink.end(false);
`cpp

```cpp
void BluetoothA2DPSink::end(bool release_memory) {
    // reconnect should not work after end
    is_autoreconnect_allowed = false;
    BluetoothA2DPCommon::end(release_memory);
    app_task_shut_down();

    // stop I2S
    if (is_i2s_output){
        ESP_LOGI(BT_AV_TAG,"uninstall i2s");
        if (i2s_driver_uninstall(i2s_port) != ESP_OK){
            ESP_LOGE(BT_AV_TAG,"Failed to uninstall i2s");
        }
        else {
            player_init = false;
            delay(15);
        }
    }
    log_free_heap();
}
```

and when i try to use the sd_mmc player, i got a DMA allocation error, it cannot allocate the DMA, leading to a crash, even though the I2S driver was uninstalled, as can be seen in the log.

```
system init
Encoder ready
init audio
2C ready
SDA:4 SCL:22 Freq:100KHz
NAU88C22 [0x1A]
HP vol:90
Spk vol:95
init player
init_sd
Type: SDHC
Total: 7678MB
Used: 128MB
start player
volume:8
Firmware Version: 0x92 = v2.0
player end
start a2dp
init a2dp
[ 21068][D][BluetoothA2DPSink.cpp:142] start(): [BT_AV] start
[ 21070][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 1896083
[ 21078][I][BluetoothA2DPSink.cpp:154] start(): [BT_AV] Device name will be set to 'FABA'
[ 21086][D][BluetoothA2DPSink.cpp:1079] init_nvs(): [BT_AV] init_nvs
[ 21092][D][BluetoothA2DPCommon.cpp:149] get_last_connection(): [BT_AV] get_last_connection
[ 21100][I][BluetoothA2DPCommon.cpp:163] get_last_connection(): [BT_AV] nvs_blob does not exist
[ 21108][D][BluetoothA2DPCommon.cpp:172] get_last_connection(): [BT_AV] => 00:00:00:00:00:00
[ 22117][I][BluetoothA2DPSink.cpp:214] init_i2s(): [BT_AV] init_i2s
[ 22117][I][BluetoothA2DPSink.cpp:216] init_i2s(): [BT_AV] init_i2s is_i2s_output
[ 22120][D][BluetoothA2DPSink.cpp:288] init_bluetooth(): [BT_AV] init_bluetooth
[ 22129][I][BluetoothA2DPSink.cpp:293] init_bluetooth(): [BT_AV] controller initialized
[ 22476][I][BluetoothA2DPSink.cpp:302] init_bluetooth(): [BT_AV] bluedroid initialized
[ 22544][I][BluetoothA2DPSink.cpp:311] init_bluetooth(): [BT_AV] bluedroid enabled
[ 22544][D][BluetoothA2DPSink.cpp:424] app_task_start_up(): [BT_AV] app_task_start_up
[ 22548][D][BluetoothA2DPSink.cpp:1198] ccall_app_task_handler(): [BT_AV] ccall_app_task_handler
[ 22556][D][BluetoothA2DPSink.cpp:380] app_task_handler(): [BT_AV] app_task_handler
[ 22564][D][BluetoothA2DPSink.cpp:332] app_work_dispatch(): [BT_API] app_work_dispatch event 0x0, param len 0
[ 22573][D][BluetoothA2DPSink.cpp:364] app_send_msg(): [BT_AV] app_send_msg
[ 22580][D][BluetoothA2DPSink.cpp:387] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x0
[ 22589][I][BluetoothA2DPSink.cpp:390] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 22599][D][BluetoothA2DPSink.cpp:355] app_work_dispatched(): [BT_AV] app_work_dispatched
[... 36 lines cut ...]
[ 57558][I][BluetoothA2DPSink.cpp:714] handle_connection_state(): [BT_AV] A2DP connection state: Disconnecting, [a4:c6:f0:85:a9:9a]
[ 57569][I][BluetoothA2DPSink.cpp:718] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_DISCONNECTING
[ 57579][I][BluetoothA2DPCommon.cpp:89] end(): [BT_AV] deinit avrc
[ 57585][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 1776159
[ 57593][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 1776183
[ 57601][D][BluetoothA2DPSink.cpp:437] app_task_shut_down(): [BT_AV] app_task_shut_down
[ 57609][I][BluetoothA2DPSink.cpp:84] end(): [BT_AV] uninstall i2s
[ 57615][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 1782895
start player
E (58625) I2S: i2s_alloc_dma_buffer(740): Error malloc dma buffer
E (58630) I2S: i2s_realloc_dma_buffer(787): Failed to allocate dma buffer
E (58636) I2S: i2s_set_clk(1728): I2S0 tx DMA buffer malloc failed
E (58642) I2S: i2s_driver_install(2026): I2S set clock failed
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4010a81e  PS      : 0x00060d30  A0      : 0x800e1ace  A1      : 0x3ffd0da0  
A2      : 0x00000000  A3      : 0x3ffc64bc  A4      : 0x3ffc64bc  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8010aee4  A9      : 0x3ffd0d70  
A10     : 0x00000000  A11     : 0x3ff4f000  A12     : 0x00040127  A13     : 0xffefffff  
A14     : 0x3ffbd274  A15     : 0x3f41e353  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x400946e5  LEND    : 0x400946f5  LCOUNT  : 0xfffffffb  

Backtrace: 0x4010a81b:0x3ffd0da0 0x400e1acb:0x3ffd0de0 0x400d888b:0x3ffd0e10 0x400d8a0f:0x3ffd0e80 0x40106039:0x3ffd0eb0
```

BetterJincheng, 2023-06-15:
@0x0fe  What is the key problem of restarting A2DP SINK? By the way, you may need to open a new issue for DMA support.

0x0fe, 2023-07-23:
@BetterJincheng  So, A2DP SINK cannot be restarted because once the ram allocated for A2DP sink is freed, it cannot be reallocated. That is unfortunately a documented limitation of the bluetooth stack, i have seen comments from espressif engineers here and there confirming this behaviour.
So, sadly, i dont think any fix for this will come to allow restarting A2DP after freeing it completely.

closed 2023-09-11 by Alvin1Zhang