espressif/esp-idf #11645: MQTT client keepalive interval

sforconi, 2023-06-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4, v5.0.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 DevKitC

### Power Supply used.

USB

### What is the expected behavior?

The expected behaviour should be that if I set the keepalive interval to 30sec. the mqtt client will send a ping to the broker every 30s (if no traffic is present).

### What is the actual behavior?

The mqtt client sends ping to the broker with a timing of keepalive/2 seconds (instead of keepalive seconds).

### Steps to reproduce.

Setup a simple mqtt client app on the ESP32 side that connects to a broker.
On the broker side enable logging (on mosquitto set "log_type all" in mosquitto.conf)

### Debug Logs.

```plain
Logs of the Mosquitto MQTT broker when I set a keepalive of 30s (PING_REQ are sent from the mqtt client every 15s/16s)

1686308238: Received PINGREQ from CLI-0000f412fad7a3e0
1686308238: Sending PINGRESP to CLI-0000f412fad7a3e0

1686308254: Received PINGREQ from CLI-0000f412fad7a3e0
1686308254: Sending PINGRESP to CLI-0000f412fad7a3e0

1686308270: Received PINGREQ from CLI-0000f412fad7a3e0
1686308270: Sending PINGRESP to CLI-0000f412fad7a3e0
```

### More Information.

I have doubts on the implementation of the `process_keepalive()` function in `mqtt_client.c`.

```
static inline bool has_timed_out(uint64_t last_tick, uint64_t timeout) {
  uint64_t next = last_tick + timeout;
  return (int64_t)(next - platform_tick_get_ms()) <= 0;
}

static esp_err_t process_keepalive(esp_mqtt_client_handle_t client)
{
    if (client->connect_info.keepalive > 0) {
        const uint64_t keepalive_ms = client->connect_info.keepalive * 1000;

        if (client->wait_for_ping_resp == true ) {
            if (has_timed_out(client->keepalive_tick, keepalive_ms)) {
                ESP_LOGE(TAG, "No PING_RESP, disconnected");
                esp_mqtt_abort_connection(client);
                client->wait_for_ping_resp = false;
                return ESP_FAIL;
            }
            return ESP_OK;
        }

        if (has_timed_out(client->keepalive_tick, keepalive_ms/2)) {
            if (esp_mqtt_client_ping(client) == ESP_FAIL) {
                ESP_LOGE(TAG, "Can't send ping, disconnected");
                esp_mqtt_abort_connection(client);
                return ESP_FAIL;
            }
            client->wait_for_ping_resp = true;
            return ESP_OK;
        }
    }
    return ESP_OK;
}
```

Looking at this code, it seems to me that the client sends ping requests to the broker with an interval of keepalive/2 secs. and waits for ping responses from the broker for at most keepalive secs.
I think that this behaviour should be inverted, the client should send the ping request to the broker with an interval of keepalive secs. and wait for the ping response from the broker for at most keepalive/2 secs.

AxelLin, 2023-06-11:
> ### What is the expected behavior?
> The expected behaviour should be that if I set the keepalive interval to 30sec. the mqtt client will send a ping to the broker every 30s (if no traffic is present).

Why? Did you check mqtt protocol?

sforconi, 2023-06-11:
From what I know the keepalive is an integer from 0 to 65535, representing the maximum time in seconds allowed to elapse between MQTT packets sent by the mqtt client.
For this reason why choose half of the keepalive interval to send the ping request to the Broker?

KaeLL, 2023-06-12:
I have asked myself the same question at some point, but didn't really bother looking for the answer because I don't mind the extra data consumption and I do enjoy the earlier error detection. But it's nonetheless a valid question.
@euripedesrocha care to weigh in?

sforconi, 2023-06-13:
I would like to remark that the ESP mqtt client works correctly with the current keepalive settings, however one of our customers make me notice that when he configured a keepalive of i.e. 60s, the PINGREQ from the mqtt client were sent every 30s and asked me why such behaviour. 
So I had to investigate why and I digged into the implementation of the ESP mqtt client.
Reading the MQTT spec., the ESP mqtt client implementation does not violates any specification. 
However I don't want to teach my customers all the details of the MQTT protocol and justify this behaviour everytime, so, I think that I'll introduce in my product configuration UI a workaround that doubles the configured keepalive value before sending to the ESP (i.e. if the user choose 60s, I will configure 120s on ESP).

euripedesrocha, 2023-06-27:
@sforconi and @KaeLL,
We just choose to be conservative and have more tries inside the allowed interval.
From MQTT specification section 3.2.1.10:

If the Keep Alive value is non-zero and the Server does not receive a Control Packet from the Client within one and a half times the Keep Alive time period, it MUST disconnect the Network Connection to the Client as if the network had failed [MQTT-3.1.2-24].

I agree that it's a bit surprising at first, but you also can easily take this into account when configuring the client. I'll add this information to the configuration field docs, so it's clear that this is the behavior at the moment.

KaeLL, 2023-06-27:
> We just choose to be conservative and have more tries inside the allowed interval.

That's understandable. But the way I see, the counter-option here seems more coherent with its meaning and intent, given that by specifying a 30s keep-alive, the window for sending a control packet will actually be 30s, not 15. There's also 2 weaker, though still valid points of familiarity/consistency, since [that's how mosquitto does it](https://github.com/eclipse/mosquitto/blob/cc1aac336ecbbc251ed4a5126257f6f4fe3ed27c/lib/util_mosq.c#L103), and economics by means of data saving, which although very small, it adds up over time and I'm sure it's very important to some.

> I agree that it's a bit surprising at first, but you also can easily take this into account when configuring the client. I'll add this information to the configuration field docs, so it's clear that this is the behavior at the moment.

I like to think the argument here is still about whether the current behavior should change or not rather than over a simple documentation change.

> If esp-mqtt's implementation does not has such issue

I don't think that's even an issue, considering it's a matter of resetting a timer every time a control packet is sent, and that [just about every packet is a control packet](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Table_2.1_-).

AxelLin, 2023-07-11:
@euripedesrocha Any feedback about above comments?

euripedesrocha, 2023-07-11:
@KaeLL and @AxelLin, sorry I've missed your comments.

> BTW, did you check the keepalive behavior in other mqtt client implementation?

As far as I know, there were no research on other MQTT client implementations behavior. 

> If esp-mqtt's implementation does not has such issue, there is no reason the send ping in 1/2 keepalive setting.

The current behavior was introduced once we changed when the timeout was restarted. Previously, we considered that every control packet sent was a point of restart. With the change, the current behavior was adopted for the reasons pointed above. 

> I like to think the argument here is still about whether the current behavior should change or not rather than over a simple documentation change.

At the moment, there are no plans to change the behavior. The documentation change is to clarify the behavior. 

To be clear about our reasoning, the current behavior makes the client consistent about when the control packet is sent, previously the timeout was cleared on each packet transmitted which could lead to a higher delay when detecting the disconnection. Making it consistent allows easier control to users. That's one point. Once we reduced the number of times we reset the timer, we decided to be conservative about when to send by default, giving users also an easy way to reduce the number of packets through configuration.