espressif/esp-idf #11646: How to detect log-level-none at compile time?

JB-DX, 2023-06-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to set a #define when log-level is set to none.

something like this:
```
#ifdef  ESP_LOG_NONE
    #define RELEASED_VERSION      
#endif
```

I tried several things but all failed, 
the docu doesn't help either.

What is the proper way to achieve that at compile-time?

ESP-Marius, 2023-06-12:
Not sure exactly what you are trying to achieve here. But if you are trying to set a define when log level is configured to none in menuconfig then it would be:

`#if CONFIG_LOG_DEFAULT_LEVEL_NONE`

JB-DX, 2023-06-12:
That has no effect. 
I tried also: 
`#ifdef CONFIG_LOG_DEFAULT_LEVEL_NONE`
but without success either.

ESP-Marius, 2023-06-12:
Worked fine when I tried it, remember that you need to include `sdkconfig.h`

JB-DX, 2023-06-12:
Ahh ! That was the problem, I had not included sdkconfig.h

Now it works !
Thank you very much !!!

closed 2023-06-13 by ESP-Marius