espressif/esp-idf #11649: With SPIRAM_SPEED_120M, combination od v4.4.4 bootloader and v5.0.2 app fail esp_restart()

aiot-embedded, 2023-06-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Combination of IDFv5.0.2 application that build with the following sdkconfig and IDFv4.4.4 bootloader fails esp_restart().
- CONFIG_SPIRAM =y
- CONFIG_ESPTOOLPY_FLASHFREQ_120M=y
- CONFIG_SPIRAM_SPEED_120M=y

## Target board

ESP32-S3-WROOM-1-N8R2

## Procedure

1. Copy hello_world in v5.0.2 and customize sdkconfig as above and build.
    - Its generate Bootloader_v5.0.2 and Application_v5.0.2.
2. Copy hello_world in v4.4.4 and customize sdkconfig as above and build.
    - Its generate Bootloader_v4.4.4 and Application_v4.4.4.
3. Flash the Bootloader_v4.4.4 and the Application_v5.0.2 and run.

## Issue

esp_restart() hang up and never restart application. But bootloader says `ets_loader.c 78` repeatedly.

Following is the log of failure.
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1440
load:0x403c9700,len:0xa48
load:0x403cc700,len:0x30d8
entry 0x403c98ec
I (79) flash HPM: Enabling flash high speed mode by dummy
W (80) FLASH/PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters may be updated in future IDF version.
W (85) FLASH/PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters may be updated in future IDF version.
I (95) esp_psram: Found 2MB PSRAM device
I (100) esp_psram: Speed: 120MHz
I (104) cpu_start: Pro cpu up.
I (107) cpu_start: Starting app cpu, entry point is 0x40375304
I (0) cpu_start: App cpu up.
I (336) esp_psram: SPI SRAM memory test OK
I (345) cpu_start: Pro cpu start user code
I (345) cpu_start: cpu freq: 160000000 Hz
I (346) cpu_start: Application information:
I (348) cpu_start: Project name:     hello_world
I (354) cpu_start: App version:      1
I (358) cpu_start: Compile time:     Jun 12 2023 10:23:47
I (364) cpu_start: ELF file SHA256:  db892894eac5fe95...
I (370) cpu_start: ESP-IDF:          5181de8
I (375) cpu_start: Min chip rev:     v0.0
I (380) cpu_start: Max chip rev:     v0.99
I (385) cpu_start: Chip rev:         v0.0
I (389) heap_init: Initializing. RAM available for dynamic allocation:
I (397) heap_init: At 3FC95A50 len 00053CC0 (335 KiB): DRAM
I (403) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (410) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (416) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (422) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (430) spi_flash: detected chip: generic
I (434) spi_flash: flash io: dio
I (439) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (459) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
Hello world!
This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision v0.0, 8MB external flash
Minimum free heap size: 2446776 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375805
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1440
ets_loader.c 78
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1440
ets_loader.c 78
```

In combination of the Bootloader_v5.0.2 and the Application_v5.0.2, esp_restart() works fine.
Currently I use ESP-IDF v4.4.4 and plan to OTA the firmware. 
The new firmware will build with ESP-IDF v5.0.
So I need this combination.

thank you

idea--list, 2023-06-12:
Does this occur also when applying a lower or default FLASHFREQ in 4.4.4 -> 5.0.2 the update?
If not, it might be a pathway to first update the devices to 5.0.2 to a lower freq that works, and shortly after that do a second OTA that raises that back to 120 MHz.

aiot-embedded, 2023-06-12:
Hi Peter B, thank you comment.
I tried it. But it was not works.

At first, I build hello_world in v5.0.2 with default FLASHFREQ & SPIRAMSPEED. (Application_v5.0.2_Freq80MHz)
And I flash Bootloader_v4.4.4 and Application_v5.0.2_Freq80MHz.
It works fine as below.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1440
load:0x403c9700,len:0xa48
load:0x403cc700,len:0x30d8
entry 0x403c98ec
I (79) esp_psram: Found 2MB PSRAM device
I (79) esp_psram: Speed: 80MHz
I (79) cpu_start: Pro cpu up.
I (79) cpu_start: Starting app cpu, entry point is 0x40375294
I (0) cpu_start: App cpu up.
I (368) esp_psram: SPI SRAM memory test OK
I (377) cpu_start: Pro cpu start user code
I (377) cpu_start: cpu freq: 160000000 Hz
I (377) cpu_start: Application information:
I (380) cpu_start: Project name:     hello_world
I (386) cpu_start: App version:      1
I (390) cpu_start: Compile time:     Jun 12 2023 19:19:56
I (396) cpu_start: ELF file SHA256:  c0c613a4a0036c8f...
I (402) cpu_start: ESP-IDF:          5181de8
I (407) cpu_start: Min chip rev:     v0.0
I (412) cpu_start: Max chip rev:     v0.99
I (417) cpu_start: Chip rev:         v0.0
I (421) heap_init: Initializing. RAM available for dynamic allocation:
I (429) heap_init: At 3FC95020 len 000546F0 (337 KiB): DRAM
I (435) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (442) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (448) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (454) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (462) spi_flash: detected chip: generic
I (466) spi_flash: flash io: dio
I (471) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (491) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
Hello world!
This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision v0.0, 8MB external flash
Minimum free heap size: 2449384 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375795
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1440
load:0x403c9700,len:0xa48
load:0x403cc700,len:0x30d8
entry 0x403c98ec
I (81) esp_psram: Found 2MB PSRAM device
I (81) esp_psram: Speed: 80MHz
I (81) cpu_start: Pro cpu up.
I (81) cpu_start: Starting app cpu, entry point is 0x40375294
I (0) cpu_start: App cpu up.
```

After that, I flash only Application_v5.0.2. (FLASHFREQ & SPIRAMSPEED is 120MHz)
Bootloader is not changed. it keeps Bootloader_v4.4.4.
Then, esp_restart() failure is happen.

idea--list, 2023-06-12:
Not sure if it is possible to change the bootloader via OTA in IDF. So can not help further, let's wait for someone else, who is more involved in that.

KaeLL, 2023-06-12:
https://github.com/espressif/esp-idf/issues/9824#issuecomment-1506218376

aiot-embedded, 2023-06-13:
Thank you for your information.

Bootloader_v4.4.4 and Application_v4.4.4 with 120MHz freq can restart successfully.
So I think this issue is application cause and can resolve by fixing only application part.

Alvin1Zhang, 2025-01-16:
Thanks for reporting, feel free to reopen with more updates.

closed 2025-01-16 by Alvin1Zhang