espressif/esp-idf #11650: When will the lvgl component be added to ESP-IDF git?

strawberry3papa, 2023-06-12:
### Is your feature request related to a problem?

It is inconvenient to patch each time.
When will the lvgl component be added to ESP-IDF git?

### Describe the solution you'd like.

Please add the lvgl component to ESP-IDF git.

suda-morris, 2023-06-12:
We won't add LVGL into esp-idf repo, but LVGL has been in our component registry for a long time: https://components.espressif.com/components/lvgl/lvgl

For how to use it, please refer to esp-idf examples, there're many. e.g. https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/spi_lcd_touch/main/idf_component.yml#L3

Before that, you may want to read the document about how to use the Espressif component manager: https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html

strawberry3papa, 2023-06-12:
thank yout about your help.
I need to do the patch below after cloning the esp-idf git.
=========================================
$ git clone https://github.com/espressif/esp-dev-kits
$ cd esp-dev-kits
$ git submodule update --init --recursive
$ cd esp-idf
$ git apply --whitespace=fix ../esp-dev-kits/esp32-s3-lcd-ev-board/factory/patch/release5.0_psram_octal_120m.patch
$ git status      # Check whether the operation is successful, the output should look like below:
=========================================
Perhaps if I do the example below, lvgl is automatically received?
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/spi_lcd_touch/main/idf_component.yml#L3

suda-morris, 2023-06-12:
The patch you posted actually is not for the LVGL library, but for the PSRAM support. A 120M DDR psram feature is not available on esp-idf v5.0. To make it work, user has to applied the patch.

Since esp-idf 5.1, the patch is not necessary, because we're adding it as an experimental feature in the esp-idf. See https://github.com/espressif/esp-idf/commit/739b3f03fbada583fe6a07ca771754505ef5b3fd

Also the doc here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#all-supported-modes-and-speeds

strawberry3papa, 2023-06-15:
I used esp-idf v5.1
[image]

An error occurred as below.
============================================================
ort.dir/port/esp32s3/mspi_timing_config.c.obj -c /mnt/sda5/espressif/esp-idf_v5.1/components/esp_hw_support/port/esp32s3/mspi_timing_config.c
In file included from /mnt/sda5/espressif/esp-idf_v5.1/components/newlib/platform_include/assert.h:16,
                 from /home/sangjina/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf/sys-include/sys/reent.h:503,
                 from /mnt/sda5/espressif/esp-idf_v5.1/components/newlib/platform_include/sys/reent.h:9,
                 from /home/sangjina/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf/sys-include/string.h:11,
                 from /mnt/sda5/espressif/esp-idf_v5.1/components/esp_hw_support/port/esp32s3/mspi_timing_config.c:9:
/mnt/sda5/espressif/esp-idf_v5.1/components/esp_common/include/esp_assert.h:12:27: error: static assertion failed: "FLASH and PSRAM Mode configuration are not supported"
   12 | #define ESP_STATIC_ASSERT static_assert
      |                           ^~~~~~~~~~~~~
/mnt/sda5/espressif/esp-idf_v5.1/components/esp_hw_support/port/esp32s3/./mspi_timing_tuning_configs.h:172:1: note: in expansion of macro 'ESP_STATIC_ASSERT'
  172 | ESP_STATIC_ASSERT(CHECK_POWER_OF_2(MSPI_TIMING_CORE_CLOCK_MHZ / MSPI_TIMING_PSRAM_MODULE_CLOCK), "FLASH and PSRAM Mode configuration are not supported");
      | ^~~~~~~~~~~~~~~~~
[331/1222] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/mspi_timing_tuning.c.objFAILED: esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/mspi_timing_tuning.c.obj
============================================================

suda-morris, 2023-06-20:
@strawberry3papa 

The following Kconfig combination work for me:

```
CONFIG_IDF_TARGET="esp32s3"
CONFIG_ESPTOOLPY_FLASHFREQ_120M=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y
CONFIG_SPIRAM_SPEED_120M=y
CONFIG_LV_MEM_CUSTOM=y
CONFIG_LV_MEMCPY_MEMSET_STD=y
CONFIG_LV_USE_PERF_MONITOR=y
CONFIG_IDF_EXPERIMENTAL_FEATURES=y
```

closed 2024-09-27 by Alvin1Zhang