espressif/esp-idf #11651: 使用ESP32的RMT控制4路WS2812灯带

MrYanLK, 2023-06-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E-N4

### Power Supply used.

External 3.3V

### What is the expected behavior?

希望能在ESP32上使用RMT控制4路WS2812灯带（现在调试时每条灯带上有40个LED，将来实际使用时可能会有几百个LED），具体步骤为蓝牙收到手机端的设置指令，然后通过led_strip.h里面的库函数设置灯带，目前IDF中的esp-idf/examples/peripherals/rmt/led_strip例程只能控制一条ws2812灯带，我尝试在代码中增加到4路RMT控制4路WS2812灯带，但是效果不理想，似乎会有时序问题，就会出现灯的颜色变了，我不确定是不是因为有蓝牙功能在使用，因为是freeRTOS系统，会不会是有更高级的任务，抢占了时间？怎么实现原子执行？（屏蔽掉其它任务），我现在不确定问题出在哪里。

### What is the actual behavior?

1

### Steps to reproduce.

1. Step
2. Step
3. Step
...
1

### Debug Logs.

```plain
1
```

### More Information.

1

suda-morris, 2023-06-12:
@MrYanLK 
This is a "well-known" issue. To drive an LED, usually it needs to transfer 24 words.  So 40LEDs means 24*40=960words. 
By default, one RMT TX channel can only support 64words in FIFO. To transmits all the 40LEDs, the driver will go into interrupt by around 960/64=15 times. The number will increase along with the LED numbers. The driver will prepare data in the interrupt, which also costs some time. 

If the RMT interrupt doesn't respond in time, the data transmission will be wrong, thus the LED displaying unwanted colors. To mitigate the issue, you can increase the FIFO size, as you want to use 4 RMT channels, then you can set [mem_block_num](https://github.com/espressif/esp-idf/blob/release/v4.4/components/driver/include/driver/rmt.h#L93) to **2**. But to be fair, this cannot help you too much because the interruption is still too frequent.

This is a hardware limitation, not a software bug.

Alternatively, to drive the LED strip, you can try with the SPI peripheral (although it's an expensive solution because you sacrifice an SPI controller for one LED strip). FYI: https://github.com/espressif/esp-iot-solution/blob/master/components/led/lightbulb_driver/drivers/ws2812/ws2812.c

MrYanLK, 2023-06-12:
@suda-morris

MrYanLK, 2023-06-16:
@suda-morris 我之前做过测试把bt设为core1没什么效果(还不知道怎么把rmt配置到core1)

suda-morris, 2023-06-19:
> @suda-morris 我之前做过测试把bt设为core1没什么效果(还不知道怎么把rmt配置到core1)

1. use xTaskCreatePinnedToCore to create a task
2. in that task, do RMT and led strip initialization

with the above steps, the RMT interrupt will be installed on the core that running by the task. If BT and RMT interrupt are running on different cores, RMT is less likey to be disturbed by the BT.

caidamao, 2023-08-01:
你可以尝试更新IDF到5.0，驱动有修改过。增加缓冲块的大小，相当大程度的减轻了这种现象。但主频必须要在240Mhz以上。把所有的函数全部塞进IRAM里

closed 2024-03-16 by espressif-bot