espressif/esp-idf #11654: i2s_channel_read doesnt work for 16 bit, 32 bits arrays. Only works for 8 bits array

chucrut, 2023-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32 s3 wroom 1

### Power Supply used.

USB

### What is the expected behavior?

i2s_channel_read should fill an array with 16 bit, 32 bits data

### What is the actual behavior?

i2s_channel_read fills the array using always the same numbers. The array is always the same.

### Steps to reproduce.

The i2s_channel_read internally convert the destination array to a uint8_t array.
It doesnt use the original destination size.
Please, could you fix this function, so it can work for 16 bit, 24 bit and 32 bit destination array.

Many thanks

chucrut, 2023-06-13:
You can see the buffer is cast to uint8_t in i2s_channel_read() function

L-KAYA, 2023-06-14:
Actually you can pass a `uint16_t *` or `uint32_t *` pointer as well.

The driver cast it to `uint8_t *` because we have to read the data from DMA byte by byte, but don't worry casting it to `uint8_t *` will disturb the endian, you can still cast it back to `uint16_t *` or `uint32_t *` in your application. The endian only relates to how you configure the I2S peripheral.

L-KAYA, 2023-06-30:
This issue is closed as it's not an issue. Please feel free to re-open if there is still problems

closed 2023-06-30 by L-KAYA