espressif/esp-idf #11656: 使用Vscode安装esp-idf时在最后一步报错

woshiwstt50123, 2023-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-rc1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

正常安装完esp-idf,同时可以使用基本的调试和烧录功能.

### What is the actual behavior?

从vscode的esp-idf插件安装esp-idf到最后一步时报错,无法安装完成.

### Steps to reproduce.

1. 打开vscode
2. 安装Espressif IDF插件
3. 点击EXPRESS按钮
4. 选择esp-idf版本为v5.1-rc1
5. 选择esp-idf目录为"C:\Users\TangTao\esp\esp-idf"
6. 选择IDF_TOOLS_PATH为"C:\Users\TangTao\esp\Espressif"
7. 点击"Install"按钮

### Build or installation Logs.

```plain
Upgrading pip and setuptools...
Skipping the download of C:\Users\TangTao\esp\Espressif\espidf.constraints.v5.1.txt because it was downloaded recently.
Installing Python packages
 Constraint file: C:\Users\TangTao\esp\Espressif\espidf.constraints.v5.1.txt
 Requirement files:
  - c:\Users\TangTao\esp\esp-idf\tools\requirements\requirements.core.txt
WARNING: Skipping page https://dl.espressif.com/pypi/setuptools/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/click/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/pyserial/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/cryptography/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/pyparsing/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/pyelftools/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/idf-component-manager/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/esp-coredump/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/esptool/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/esp-idf-kconfig/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/esp-idf-monitor/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
WARNING: Skipping page https://dl.espressif.com/pypi/freertos-gdb/ because the GET request got Content-Type: Unknown. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html

Installing ESP-IDF extension python packages in C:\Users\TangTao\esp\Espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe ...

The system cannot find the path specified.
Command failed: "C:\Users\TangTao\esp\Espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe" -m pip install --upgrade --constraint "c:\Users\TangTao\.vscode\extensions\espressif.esp-idf-extension-1.6.3\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\TangTao\.vscode\extensions\espressif.esp-idf-extension-1.6.3\requirements.txt"
The system cannot find the path specified.
Command failed: "C:\Users\TangTao\esp\Espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe" -m pip install --upgrade --constraint "c:\Users\TangTao\.vscode\extensions\espressif.esp-idf-extension-1.6.3\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\TangTao\.vscode\extensions\espressif.esp-idf-extension-1.6.3\requirements.txt"
The system cannot find the path specified.
```

### More Information.

操作系统是Win11

dobairoland, 2023-06-13:
It looks like you have trouble with your Internet connection. I'd suggest to use the offline installer from https://dl.espressif.com/dl/esp-idf/ and then follow https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md to set up vscode (with the existing ESP-IDF option instead of downloading it again).

woshiwstt50123, 2023-06-13:
> It looks like you have trouble with your Internet connection. I'd suggest to use the offline installer from https://dl.espressif.com/dl/esp-idf/ and then follow [...]

我需要5.1版本的idf来编写esp32C6的程序,5.0版本的idf不兼容C6,而5.1版本没有offline installer.

dobairoland, 2023-06-13:
I'm afraid that in that case you will have to wait for the final release of v5.1. The offline installer is usually available after 2-3 days after the ESP-IDF release. https://github.com/espressif/esp-idf/releases/tag/v5.1-rc1 is already available, so the final release will come soon.

woshiwstt50123, 2023-06-13:
好的.不过这样一来我们这边的项目也得跟着延后了.

closed 2023-06-13 by woshiwstt50123

dobairoland, 2023-06-13:
There are other options as well.

- You can start with v5.0 targeting some other chip and update to v5.1 when it is released. 
- You can develop with docker until the release (https://hub.docker.com/layers/espressif/idf/release-v5.1/images/sha256-c50dc40bef6be397c9c4701e7bb9e0d1cc47a6126d4a499b411c5209a1d0689e?context=explore). https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html
- Use VPN so the listed pages would be accessible at least during the installation.

xiaguangbo, 2024-03-11:
临时关闭pip共享缓存：export PIP_NO_CACHE_DIR=1