espressif/esp-idf #11657: cmake error

prabhu012, 2023-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>idf.py build
Executing action: all (aliases: build)
Running cmake in directory C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant"...
-- AUDIO_BOARD_PATH not defined
-- Configuring done
CMake Error:
  The detected version of Ninja () is less than the version of Ninja required
  by CMake (1.3).

CMake Warning:
  Manually-specified variables were not used by the project:

    CCACHE_ENABLE
    ESP_PLATFORM
    PYTHON_DEPS_CHECKED

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1

C:\esp_idf\components\esp-va-sdk\examples\google_voice_assistant>

### What is the actual behavior?

i am new in esp idf 

### Steps to reproduce.

1. Step
2. Step
3. Step
...

9305128621, 2023-06-14:
> -- AUDIO_BOARD_PATH not defined
-- Configuring done
CMake Error:
  The detected version of Ninja () is less than the version of Ninja required
  by CMake (1.3).

CMake Warning:
  Manually-specified variables were not used by the project:

    ESP_PLATFORM
    PYTHON_DEPS_CHECKED

prabhu012, 2023-06-24:
Thankyou Zikalino for replying

I have found it
ESP_PLATFORM:UNINITIALIZED=1
PYTHON_DEPS_CHECKED:UNINITIALIZED=1

Can you please explain in brief, what parameter I have to change to solve this error? I am new to this.

Thank you.

sudeep-mohanty, 2023-06-28:
@9305128621 @prabhu012 Please try the solution suggested [here](https://github.com/espressif/esp-idf/issues/11730#issuecomment-1610946303)

sudeep-mohanty, 2023-07-04:
Closing this issue with the resolution suggested [here](https://github.com/espressif/esp-idf/issues/11730#issuecomment-1610946303). Please feel free to open a new ticket for further reporting any new problems.

closed 2023-07-04 by sudeep-mohanty