espressif/esp-idf #11658: Watchdog reset after esp_restart()

ddomnik, 2023-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP-S3

### Power Supply used.

External 3.3V

### What is the expected behavior?

Simple RTC_SW_SYS_RST  after calling esp_restart()

### What is the actual behavior?

Unidentifiable Watchdog reset after esp_restart()

The Watchdog timeout is set to 30ms, I know this is fast, but I still would like to find the exact root cause if possible.
Strangely, the log already prints RTC_SW_SYS_RST from the esp_restart() but then prints a RTCWDT_RTC_RST. 

### Steps to reproduce.

Both on core 0 and 1 we run multiple RTOS tasks, with heavy load, such as WiFi, Modem, TWAI, ...

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038314e
0x4038314e: _frxt_setup_switch at C:/Espressif/frameworks/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:60

SPIWP:0xee
mode:DIO, clock div:�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3980,len:0x1ff8
load:0x403c9700,len:0x1038
load:0x403cc700,len:0x3760
entry 0x403c9994
```
```

ESP-Marius, 2023-06-26:
Hi @ddomnik, sorry for the slow response.

> The Watchdog timeout is set to 30ms,

Which watchdog are you talking about here? The interrupt WDT (`ESP_INT_WDT_TIMEOUT_MS`)?

Some questions:
1. Does the issue occur even if you dont configure the WDT to 30ms?
2. Do you in any way touch the RTC WDT in your code?
3. Could you attach your `sdkconfig`-file?

If you have any way for us to reproduce the issue then that would help a lot! Currently it is a bit hard to tell what exactly is happening here.

ddomnik, 2023-06-26:
Hey, 
we utilize `wdt_hal_init(&rtc_wdt_ctx, WDT_RWDT, 0, false);` and `wdt_hal_config_stage(&rtc_wdt_ctx, WDT_STAGE0, stage_timeout_ticks,WDT_STAGE_ACTION_RESET_RTC);`

no the WDT does not trigger if I reset the WDT via esp_task_wdt_reset() or change the timeout via wdt_hal_config_stage to more than 30ms directly before the esp_restart().

This is our sdkconfig for WDT, but please note that we set it to 30ms via code.
[image]

I would provide a sample or precise reproduce steps but it's not that easy as our project is very big. But I am wondering how a WDT can be triggered after a esp_restart (RTC_SW_SYS_RST) in the logs

ESP-Marius, 2023-06-27:
I'm still a bit confused as you are mixing the different WDTs. You can read [WDTs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/wdts.html) for more info about them.

> `wdt_hal_init(&rtc_wdt_ctx, WDT_RWDT, 0, false);`

Any specific reason why you are using the RTC WDT in your code? And what is the timeout here?

> no the WDT does not trigger if I reset the WDT via esp_task_wdt_reset() 

`esp_task_wdt_reset() ` is for the task WDT which is unrelated to both the RTC WDT and the interrupt WDT, so i cant see how this should affect anything?

> or change the timeout via wdt_hal_config_stage to more than 30ms directly before the esp_restart().

Can you copy-paste the code that does this and show it to me? I'm not able to confirm exactly which WDT you are configuring here.

> But I am wondering how a WDT can be triggered after a esp_restart (RTC_SW_SYS_RST) in the logs.

What I suspect is happening is that you are running into issues due to using the RTC WDT, `esp_restart()` will not clear RTC registers, thus when you reboot from the bootloader your RTC WDT just keeps running, but without enough time left to actual boot the app.

My suggestion for you:

1. Decide if you actual need the RTC WDT. The reason why there is no public API for this ,just HAL functions is that we've considered the interrupt WDT and the task WDT to be sufficient for users. If there is any important use-cases that we've missed that forces you to use the RTC WDT then we would like to know.
2. If you decide to keep using the RTC WDT then simply disable it before calling `esp_restart`

ddomnik, 2023-11-06:
Thanks I didn't know that WDT timers aren't reset at a chip reset.

closed 2023-11-06 by ddomnik