espressif/esp-idf #11659: xQueueSendFromISR sending more signals to the Queue than supposed to

krupis, 2023-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I am learning about FreeRTOS and Queue. 

I have 5 IR sensors and I add isr handler for each IR pin:

```

static void IRAM_ATTR IR_sensor_handler(void* arg)
{
    BaseType_t pxHigherPriorityTaskWoken = pdFALSE;
    uint32_t gpio_num = (uint32_t) arg;

    xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
    if( pxHigherPriorityTaskWoken )
    {
        portYIELD_FROM_ISR ();
    }
}

void sensor_GPIO_setup(){
        //OUTPUTS INIT
    //zero-initialize the config structure.
    gpio_config_t io_conf;
    //disable interrupt
    io_conf.intr_type = GPIO_INTR_ANYEDGE;
    //set as output mode
    io_conf.mode = GPIO_MODE_INPUT ;
    //bit mask of the pins that you want to set,e.g.GPIO18/19
    io_conf.pin_bit_mask = GPIO_SENSOR_INPUT_SEL;
    //disable pull-down mode
    io_conf.pull_down_en = GPIO_PULLDOWN_ENABLE;
    //disable pull-up mode
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    //configure GPIO with the given settings
    gpio_config(&io_conf);

    gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
    //start gpio task
    xTaskCreate(Sensor_detection_task, "Sensor_detection_task", 2048, NULL, 10, NULL);

    gpio_install_isr_service(0);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(IR_RIGHT, IR_sensor_handler, (void*) IR_RIGHT);
    gpio_isr_handler_add(IR_LEFT, IR_sensor_handler, (void*) IR_LEFT);
    gpio_isr_handler_add(IR_MIDDLE, IR_sensor_handler, (void*) IR_MIDDLE);
    gpio_isr_handler_add(IR_MIDDLE_RIGHT, IR_sensor_handler, (void*) IR_MIDDLE_RIGHT);
    gpio_isr_handler_add(IR_MIDDLE_LEFT, IR_sensor_handler, (void*) IR_MIDDLE_LEFT);

}

void Sensor_detection_task(void* arg)
{
    
    uint32_t io_num;
    for(;;) {
        if(xQueueReceive(gpio_evt_queue, &io_num, portMAX_DELAY)) { 
            if(io_num == IR_RIGHT){
                printf("Right sensor toggled state = %u\n",gpio_get_level(io_num));
                sensor_right_state = gpio_get_level(io_num);
            }
            else if(io_num == IR_MIDDLE_RIGHT){
                printf("Middle Right sensor toggled state = %u \n",gpio_get_level(io_num));
                sensor_mid_right_state = gpio_get_level(io_num);
            }
            else if(io_num == IR_MIDDLE){
                printf("Middle sensor toggled state = %u \n",gpio_get_level(io_num));
                sensor_mid_state = gpio_get_level(io_num);
            }
            else if(io_num == IR_MIDDLE_LEFT){
                printf("Middle Left sensor toggled state = %u \n",gpio_get_level(io_num));
                sensor_mid_left_state = gpio_get_level(io_num);
            }
            else if(io_num == IR_LEFT){
                printf("Left sensor toggled state = %u \n",gpio_get_level(io_num));
                sensor_left_state = gpio_get_level(io_num);
            }
            vTaskDelay(100/portTICK_PERIOD_MS);
        }
    }
}

```

I have also connected a logic analyzer to my IR sensor output pin. When the IR sensor is toggled, it will pull the pin LOW for as long as it detects obstacle in front of it. When I move my hand away from the sensor, the signal will move UP to its default position. See the logic analyzer below:

[image]

However, the following is being printed on the serial debug terminal :

[image]

Why does it toggle so many times? Since I have GPIO_INTR_ANYEDGE interrupt, I would only expect it to toggle twice (Once when the output went LOW, and again when the output went HIGH.

Dazza0, 2023-06-13:
@krupis Could add a `esp_rom_printf()` or increment an integer to verify whether it is the ISR running multiple times or the queue sending more times than expected.

krupis, 2023-06-14:
@Dazza0 
I have added esp_rom_printf in the IR handler:
```
static void IRAM_ATTR IR_sensor_handler(void* arg)
{
    BaseType_t pxHigherPriorityTaskWoken = pdFALSE;
    uint32_t gpio_num = (uint32_t) arg;

    xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
    esp_rom_printf("IR sensor handler toggled \n");
    if( pxHigherPriorityTaskWoken )
    {
        portYIELD_FROM_ISR ();
    }
}
```

I did the following test:
Put my hand in front of the IR sensor and hold it there for a couple of seconds, then move the hand away. The results:
[image]

```
IR sensor handler toggled 
IR sensor handler toggled 
IR sensor handler toggled 
IR sensor handler toggled 
IR sensor handler toggled 
IR sensor handler toggled 
Right sensor toggled state = 0
Right sensor toggled state = 0
Right sensor toggled state = 0
Right sensor toggled state = 0
Right sensor toggled state = 0
Right sensor toggled state = 0
IR sensor handler toggled 
IR sensor handler toggled 
IR sensor handler toggled 
IR sensor handler toggled 
IR sensor handler toggled 
IR sensor handler toggled 
Right sensor toggled state = 1
Right sensor toggled state = 1
Right sensor toggled state = 1
Right sensor toggled state = 1
Right sensor toggled state = 1
Right sensor toggled state = 1
```

I cannot wrap my head around why would that happen? Why would it toggle multiple times if the signal hadnt changed?

Dazza0, 2023-06-14:
@krupis What sampling frequency are you running you're logic analyzer at? Any chance the edges are debouncing?

krupis, 2023-06-14:
> @krupis What sampling frequency are you running you're logic analyzer at? Any chance the edges are debouncing?

In to Logic software I use the 24 MS/s sampling frequency:
[image]

Dazza0, 2023-06-14:
@krupis Would be helpful to use an oscilloscope just to rule out debouncing as the source of the issue.

krupis, 2023-06-14:
I did a scope measurement with a decent scope OWON XDS3204E

[image]

krupis, 2023-06-16:
@Dazza0 
Are there any updates regarding this? Keep in mind that this is not only the case for the IR sensor. The same thing happens if I wire the simple push button to the interrupt pin. Multiple signals will be generated even though I only click the button once.

Dazza0, 2023-06-19:
@krupis Could you let us know which ESP chip you are using (e.g., ESP32, ESP32-S2, ESP32-C3 etc)

krupis, 2023-06-19:
For this experiment I am using just a regular ESP32.

Dazza0, 2023-06-22:
@krupis I can't recreate the issue. The following example was run on an ESP32 WROVER Kit where I use a TX GPIO to trigger edge interrupts on multiple RX GPIOs.

<details>
<summary>Example code</summary>

```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_rom_sys.h"

#define OUTPUT_NO 21
#define GPIO_TX_SEL     (1 << OUTPUT_NO)
#define GPIO_SENSOR_INPUT_SEL   ((1 << 22) | (1 << 19))

static void IRAM_ATTR gpio_isr(void* arg)
{
    esp_rom_printf("gpio_isr %d\n", (int)arg);
}

void app_main(void)
{
    printf("Hello World\n");

    /* Setup TX GPIO */
    gpio_config_t tx_conf;
    tx_conf.intr_type = GPIO_INTR_DISABLE;
    tx_conf.mode = GPIO_MODE_OUTPUT;
    tx_conf.pin_bit_mask = GPIO_TX_SEL;
    tx_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    tx_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    gpio_config(&tx_conf);
    gpio_set_level(OUTPUT_NO, 0);

    /* Setup RX GPIO */
    //zero-initialize the config structure.
    gpio_config_t rx_conf;
    //disable interrupt
    rx_conf.intr_type = GPIO_INTR_ANYEDGE;
    //set as output mode
    rx_conf.mode = GPIO_MODE_INPUT;
    //bit mask of the pins that you want to set,e.g.GPIO18/19
    rx_conf.pin_bit_mask = GPIO_SENSOR_INPUT_SEL;
    //disable pull-down mode
    rx_conf.pull_down_en = GPIO_PULLDOWN_ENABLE;
    //disable pull-up mode
    rx_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    //configure GPIO with the given settings
    gpio_config(&rx_conf);

    gpio_install_isr_service(0);
    //hook isr handler for specific gpio pin
    gpio_isr_handler_add(19, gpio_isr, (void *)19);
    gpio_isr_handler_add(22, gpio_isr, (void *)22);

    bool level = false;
    while (1) {
        gpio_set_level(OUTPUT_NO, (uint32_t)(!level));
        level = !level;
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
```
</details>

Could modify the example to your hardware/selected GPIOs to see if you can isolate the issue?

Dazza0, 2023-11-13:
@krupis Closing as I can't recreate the bug on an ESP32 WROVER Kit using the example code I posted above (uses 2 RX pins).

One last (but simple) thing to check is if the various pins chosen for LEFT/MIDDLE/RIGHT aren't shorted at the hardware level.

Please feel free to reopen if you have any further developments/findings.

closed 2023-11-13 by Dazza0