espressif/esp-idf #11660: I2S  MCLK is not correct

zhou88mang, 2023-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.02

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-DevKitCV4

### Power Supply used.

USB

### What is the expected behavior?

I have run the example of i2s std, it worked on DUPLEX mode ,  when i fixed the mclk io to gpio0,  and set sample rate to 16k hz,   I hope that the output mclk was  16000 * 256 = 4096000 hz .

### What is the actual behavior?

the actual mclk frequency was about 95 ~ 100 khz , They're too different .

### Steps to reproduce.

```/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */

#include <stdint.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2s_std.h"
#include "driver/gpio.h"
#include "esp_check.h"
#include "sdkconfig.h"

/* Set 1 to allocate rx & tx channels in duplex mode on a same I2S controller, they will share the BCLK and WS signal
 * Set 0 to allocate rx & tx channels in simplex mode, these two channels will be totally separated,
 * Specifically, due to the hardware limitation, the simplex rx & tx channels can't be registered on the same controllers on ESP32 and ESP32-S2,
 * and ESP32-S2 has only one I2S controller, so it can't allocate two simplex channels */
#define EXAMPLE_I2S_DUPLEX_MODE         CONFIG_USE_DUPLEX

#if CONFIG_IDF_TARGET_ESP32
    #define EXAMPLE_STD_BCLK_IO1        GPIO_NUM_4      // I2S bit clock io number
    #define EXAMPLE_STD_WS_IO1          GPIO_NUM_5      // I2S word select io number
    #define EXAMPLE_STD_DOUT_IO1        GPIO_NUM_18     // I2S data out io number
    #define EXAMPLE_STD_DIN_IO1         GPIO_NUM_19     // I2S data in io number
    #if !EXAMPLE_I2S_DUPLEX_MODE
        #define EXAMPLE_STD_BCLK_IO2    GPIO_NUM_22     // I2S bit clock io number
        #define EXAMPLE_STD_WS_IO2      GPIO_NUM_23     // I2S word select io number
        #define EXAMPLE_STD_DOUT_IO2    GPIO_NUM_25     // I2S data out io number
[... 246 lines cut ...]
Read Task: i2s read 2048 bytes
-----------------------------------
[0] 0 [1] 0 [2] 0 [3] 0
[4] 0 [5] 0 [6] 0 [7] 0

Write Task: i2s write 2048 bytes
Read Task: i2s read 2048 bytes
-----------------------------------
[0] 0 [1] 0 [2] 0 [3] 0
[4] 0 [5] 0 [6] 0 [7] 0

Write Task: i2s write 2048 bytes
Read Task: i2s read 2048 bytes
-----------------------------------
[0] 0 [1] 0 [2] 0 [3] 0
[4] 0 [5] 0 [6] 0 [7] 0

Write Task: i2s write 2048 bytes
Read Task: i2s read 2048 bytes
```

### More Information.

i found that the log showed mclk is 4096000hz , but the actual  frequency is 95 ~ 100khz measured by logic analyzer.
[image]

zhou88mang, 2023-06-14:
Doesn't that happen to anyone?

L-KAYA, 2023-06-14:
I tested with your code, but my mclk is correct (the small error is from my logic analyzer)
[image]

Normally, when the BCLK and WS signal is correct, the mclk signal is unlike to go wrong. As there seems no error in the code, I guess there might be another signal connected to GPIO0 on your dev board, you can check the GPIO0 without initialize I2S.

zhou88mang, 2023-06-14:
> I tested with your code, but my mclk is correct (the small error is from my logic analyzer) [image]
> 
> Normally, when the BCLK and WS signal is correct, the mclk signal is unlike to go wrong. As there seems no error in the code, I guess there might be another signal connected to GPIO0 on your dev board, you can check the GPIO0 without initialize I2S. Or you can also change to another pin like GPIO1 or GPIO3

Thank for your reply ,  i have test gpio3 , it worked  well ! but gpio0  was same as before, it could be a hardware problem with the board.

L-KAYA, 2023-06-14:
Another possible reason might be that, there is another gpio signal attached to GPIO0, which overwrite the I2S mclk signal. For this case, you might need to go through your codes and find where else may initialized GPIO0.

Thanks for your report. I'm going to close this issue, please feel free to reopen it if there are still issues on **I2S**.

closed 2023-06-14 by L-KAYA