espressif/esp-idf #11661: HTTP request URI not empty when using websockets

higaski, 2023-06-13:
### Is your feature request related to a problem?

Currently the _.uri_ member of an _httpd_req_t_ instance is just an empty string instead of the initially specified endpoint when using websockets. I wonder if there is any specific reason for this? It makes it a little cumbersome to write generic websocket handlers which simply forward a request to some listener.

E.g., imagine an application which uses two different websockets. One endpoint could be "/stream" and one endpoint could be "/update". Although those endpoints might do vastly different things some precondition checks might be shared.

```cpp
esp_err_t wsHandler(httpd_req_t* req) {
  if (req->method == HTTP_GET)
    return ESP_OK;

   // Perform some general checks here which would be equal for both handlers
   if (/*some error*/)
     return...

   // Finally pass the request to somewhere else...
   return asyncResponse(req);
}
```

Not knowing the endpoint inside the handler makes it impossible to pass the request on.

### Describe the solution you'd like.

It would be great if the _.uri_ member could be set for websockets as well, just like for normal HTTP requests.

ESP-YJM, 2023-06-27:
@higaski I think when you register the handler `wsHandler`, it is assoicated with your uri. You can register the different handler with corresponding uri. So in different uri handler, you no need care about uri.

higaski, 2023-06-27:
Yes I know, you kinda missed the point about handlers being "generic". If the actual endpoint destination is lost before the handler then writing a generic one isn't possible.

mahavirj, 2023-08-28:
@higaski 

Sorry for slow followup here. Do you mean if you log the `req->uri` in the handler below then it comes out to be empty?

https://github.com/espressif/esp-idf/blob/82cceabc6e6a0a2d8c40e2249bc59917cc5e577a/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L74

At-least this part is not reproducible at my end and I can get the correct `.uri` member for websocket case too (as it was registered using `httpd_register_uri_handler`). Please share some code snippet and we will evaluate it further.

higaski, 2023-08-28:
> @higaski
>
> Sorry for slow followup here. Do you mean if you log the `req->uri` in the handler below then it comes out to be empty? [...]

Yes, last time I checked the behavior was identical. The URI is lost though once the connection has been established. The member is empty on every actual WebSocket transmission

mahavirj, 2023-08-29:
> The URI is lost though once the connection has been established.

Can you please share your patch on top of `ws_echo_server` example? Please mention expected and observed behavior.

higaski, 2023-08-29:
Here you go, https://github.com/higaski/ws_echo_server
The python script in the main folder opens a WebSocket and sends some random strings to the webserver.

```sh
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

# ...
# Remove unnecessary output until actual WebSocket log
# ...

I (12981) ws_echo_server: uri /ws  # uri set until handshake
I (12981) ws_echo_server: Handshake done, the new connection was opened
I (12991) ws_echo_server: uri EMPTY!  # uri empty when receiving data
I (12991) ws_echo_server: frame len is 1024
I (12991) ws_echo_server: Packet type: 2
I (13181) ws_echo_server: uri EMPTY!  # always empty from now on...
I (13181) ws_echo_server: frame len is 1024
I (13181) ws_echo_server: Packet type: 2
I (13231) ws_echo_server: uri EMPTY!
I (13241) ws_echo_server: frame len is 1024
I (13241) ws_echo_server: Packet type: 2
I (13291) ws_echo_server: uri EMPTY!
I (13291) ws_echo_server: frame len is 1024
I (13301) ws_echo_server: Packet type: 2
```

ESP-YJM, 2023-08-30:
@higaski I think you can use `user_ctx` field to record your uri. Now the first time, you can get `req->uri` and then next data comes `req->uri` is NULL. Becase the `req->uri` will be clear after websocket handshake is over. And then the next data comes, the esp_http_server will not parse the `uri` and only use the previous socket database. So no one will re-copy the `req->uri` to structure req. 
But you can follow this code to make `uri` can obtain in your handler function.
```
static char ws_uri[] = "/ws";
static const httpd_uri_t ws = {
        .uri        = "/ws",
        .method     = HTTP_GET,
        .handler    = echo_handler,
        .user_ctx   = NULL,
        .is_websocket = true,
        .user_ctx = ws_uri
};

/*
 * This handler echos back the received ws data
 * and triggers an async send if certain message received
 */
static esp_err_t echo_handler(httpd_req_t *req)
{
    if (strlen((char *)req->user_ctx))
        ESP_LOGI(TAG, "uri %s", (char *)req->user_ctx);
    else
        ESP_LOGI(TAG, "uri %s", "EMPTY!");
    ...
}
```

higaski, 2023-08-30:
That's actually a pretty decent workaround which is good enough for me. Thanks!

closed 2023-10-18 by ESP-YJM