espressif/esp-idf #11663: ESP32C3/ESP32S3 use SPI0 and SPI1, Concurrency Constraints for flash or SPI?

WangShuoran, 2023-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The SPI function in the ESP32C3 and ESP32S3 has a Flash Concurrency Constraints section, which mentions that operations on the SPI1 peripheral can have a significant impact on the overall system:
https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/api-reference/peripherals/spi_flash/spi_flash_concurrency.html
https://docs.espressif.com/projects/esp-idf/zh_CN/v5.0.2/esp32s3/api-reference/storage/spi_flash_concurrency.html
https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.5/esp32s3/api-reference/storage/spi_flash_concurrency.html

However, in https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/peripherals/spi.html. It is mentioned that ESP32 has 4 groups of SPI, of which SPI0 connects to flash where the program is stored, SPI1 connects to PSRAM, SPI0 and SPI1 share a set of GPIO interfaces that are occupied by default; the remaining two groups, SPI2 and SPI3, are general-purpose SPI that can be freely used by customers.

So my first question based on the previous introduction: is it possible that the concurrency constraint of SPI1 is wrong and should be a concurrency constraint for SPI0?

Not done yet, I did the following search again, hoping to find as many resources as possible. I found the following additional information:
https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.4/esp32s3/api-reference/peripherals/spi_master.html
https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.4/esp32s3/api-reference/storage/spi_flash.html

There is a description in the SPI host that the SPI1 function cannot be used, but there is a mention in the SPI_Flash chapter that SPI0 is connected to the Cache, and the ESP_Flash driver is used through SPI1 and SPI2, which can access the main Flash and external Flash chips with SPI1 and SPI2. And introduce the cache Cache can't range to operate its functional range of external Flash or address range, including mmap, encryption read/write, execution code or Flash variables.

So based on the above information, I would like to ask:
Is the Cache introduced above the main Flash mapping opened in RAM, and is the SPI0 bus used to access the main Flash chip through the Cache?
Is it true that the main Flash in ESP-IDF can be operated in SPI0 and also in SI1; SPI0 mainly generates Cache and SPI1 mainly adds, deletes and checks data?

The functions I need to implement are (mainly because of the large project pin requirements and the project for scalability to save future interface considerations)
1. using SPI0/1 to use external Flash (NOR or NAND), the initial selection is W25Qxx and W25N01 chips; if we can we are not using SPI1 for initialization operations, is there a case?
2. use SPI0/1 to drive the slave using the SPI host function, this function is used separately from the previous functions, and it would be better if it can be overlapped; if it is possible, do we use SPI0 for the initialization operation, is there any case?
3. when using the above function, there will not be too much interference with the use of the chip itself, and the lower the performance loss, the better

A few more related questions:
Is it correct that the use of PSRAM is only used to SPI1 driver?
In the Master document SPI1 Flash concurrency constraint chapter introduced when enabling the flash erase automatically suspend, looks very similar to the CONFIG_SPI_FLASH_AUTO_SUSPEND in menuconfig, but because it is the Master document, and our project development are using the official version of the development, so the effect How is it?
And I see that this feature is based on the XMC chip, and XMC can also have a low-voltage restart CONFIG_SPI_FLASH_BROWNOUT_RESET_XMC option that other chips do not have can be configured, so is XMC more suitable for ESP32 to carry? What is the increased efficiency of Flash compared to Winbond?

以下是中文意思，我使用的母语是中文：
ESP32C3和ESP32S3中的SPI功能中有Flash并发约束章节，该章节提到了SPI1外设上的操作会对整个系统造成显著的影响。
https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/api-reference/peripherals/spi_flash/spi_flash_concurrency.html
https://docs.espressif.com/projects/esp-idf/zh_CN/v5.0.2/esp32s3/api-reference/storage/spi_flash_concurrency.html
https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.5/esp32s3/api-reference/storage/spi_flash_concurrency.html
但在https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/peripherals/spi.html介绍中提到ESP32 有 4 组 SPI，其中 SPI0 连接储存程序的 flash， SPI1 连接 PSRAM，SPI0 和 SPI1 共用一组 GPIO 接口，默认被占用；剩下的两组 SPI2 和 SPI3 为可供客户自由使用的通用 SPI。

所以我的根据前面的介绍的第一个疑问：是不是并发约束的SPI1是错的，应该是对SPI0的并发约束？

还没完，我又进行了如下的搜索，希望尽可能找到更多的资源。我又找到以下资料：
https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.4/esp32s3/api-reference/peripherals/spi_master.html
https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.4/esp32s3/api-reference/storage/spi_flash.html
在SPI主机中有介绍不能使用SPI1功能，但在SPI_Flash章节中有提到SPI0是连接到Cache的，而ESP_Flash驱动时通过SPI1和SPI2使用的，可以用SPI1和SPI2访问主Flash和外部Flash芯片。并且介绍高速缓冲Cache无法范围操作其功能范围的外部Flash或地址范围，包括mmap，加密读写，执行代码或Flash变量。

那么基于上面的资料，我想问：
上面介绍的Cache是不是在RAM中开辟的主Flash映射，通过Cache访问主Flash芯片是不是用SPI0总线？
是不是主Flash在ESP-IDF可以在SPI0操作，也可以在SI1操作；SPI0主要生成Cache，SPI1主要增删改查数据？

我需要实现的功能有（主要因为项目引脚需求较大，而且项目为扩展性节约以后接口考虑）：
1. 使用SPI0/1使用外部Flash（NOR或NAND），初步挑选为W25Qxx和W25N01芯片；如果可以我们是不是使用SPI1进行初始化操作，有案例吗？
2. 使用SPI0/1使用SPI主机功能驱动从机，这个功能和前面的功能分离使用，如果能叠加更好；如果可以我们是不是使用SPI0进行初始化操作，有案例吗？
3. 在使用上面功能时，不会对芯片本身的使用有太大的干扰，而且性能损失越低越好

多问几个相关的问题：
在PSRAM的使用是不是只用到SPI1驱动，是否正确？
在Master文档SPI1 Flash并发约束章节中介绍到了当使能 flash 擦除的自动暂停，看起来和menuconfig中的CONFIG_SPI_FLASH_AUTO_SUSPEND非常相似，但因为是Master文档，而我们项目开发都是使用正式版开发，所以效果如何？
而且我看到这个功能是基于XMC芯片，而且XMC也可以有其他芯片不具备的低压重启CONFIG_SPI_FLASH_BROWNOUT_RESET_XMC选项可以配置，那么是不是XMC更适合ESP32搭载？相比于Winbond这种Flash增加的效率是多少？

Icarus113, 2023-06-14:
> 但在https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/peripherals/spi.html 介绍中提到ESP32 有 4 组 SPI，其中 SPI0 连接储存程序的 flash， SPI1 连接 PSRAM，SPI0 和 SPI1 共用一组 GPIO 接口，默认被占用；剩下的两组 SPI2 和 SPI3 为可供客户自由使用的通用 SPI。

This doc is wrong. Sorry for this misleading and we will update this doc then.
SPI0 and SPI1 are used for communicating with external physical memory, including SPI Flash and PSRAM (on ESP32). These two SPIs are call MSPI as well. MSPI (SPI0, SPI1) are two different hardwares, but they share a same set of bus. The only IO difference is that, CS1 is for PSRAM, CS0 is for SPI Flash.

Here're few ways to access the Flash:
- By default most of the ESP-IDF program is placed in the Flash. Accessing code placed in Flash is done by hardware (SPI0). 
- And we provide the [ESP FLASH driver](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/esp_flash.h) for you to access the Flash for other purpose (e.g. storage). This is via SPI1. 
- We provide [SPI FLASH MMAP driver](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/spi_flash_mmap.h) and [MMU MMAP driver](https://github.com/espressif/esp-idf/blob/master/components/esp_mm/include/esp_mmu_map.h) for you to access (via SPI0, e.g. by a pointer) those you write to the Flash via ESP FLASH driver.

> 使用SPI0/1使用外部Flash（NOR或NAND），初步挑选为W25Qxx和W25N01芯片；如果可以我们是不是使用SPI1进行初始化操作，有案例吗？

Yes we're using SPI1 to do the Flash chip initialisation. This contains multiple stages, if you use the ESP-IDF project, these initialisations are done for you already.

About the Flash chips you select, I cannot guarantee these Flash chips are compatible for now. You may contact [here](https://www.espressif.com/en/contact-us/technical-inquiries) and query us officially.

> 使用SPI0/1使用SPI主机功能驱动从机，这个功能和前面的功能分离使用，如果能叠加更好；如果可以我们是不是使用SPI0进行初始化操作，有案例吗？

I would suggest you using SPI2/SPI3 as general purpose SPI usage. Yes on ESP32 you can use [SPI Master driver](https://github.com/espressif/esp-idf/blob/master/components/driver/spi/include/driver/spi_master.h) for SPI1. But this may influence the SPI0 (as I said before, SPI0/1 share the same bus) communication, thus influence the application efficiency. During SPI1 communication, we'll suspend all other tasks and non-IRAM ISRs

> 在PSRAM的使用是不是只用到SPI1驱动，是否正确？

THis is wrong, accessing PSRAM is via the SPI0 hardware. On ESP32, there're two drivers that may use SPI1, [SPI Master driver](https://github.com/espressif/esp-idf/blob/master/components/driver/spi/include/driver/spi_master.h), and [ESP FLASH driver](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/esp_flash.h). 

For PSRAM, we don't give driver APIs to access it. Instead, we suggest you using [heap](https://github.com/espressif/esp-idf/blob/master/components/heap/include/esp_heap_caps.h) to malloc a piece of memory on PSRAM via this [MALLOC_CAP_SPIRAM](https://github.com/espressif/esp-idf/blob/master/components/heap/include/esp_heap_caps.h#L39) flag.

> 在Master文档SPI1 Flash并发约束章节中介绍到了当使能 flash 擦除的自动暂停，看起来和menuconfig中的CONFIG_SPI_FLASH_AUTO_SUSPEND非常相似，但因为是Master文档，而我们项目开发都是使用正式版开发，所以效果如何？

ESP32 doesn't have the capability to use the auto suspend feature yet. For more details, check this [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html#auto-suspend-resume) for more details.

ginkgm, 2023-06-14:
> In the Master document SPI1 Flash concurrency constraint chapter introduced when enabling the flash erase automatically suspend, looks very similar to the CONFIG_SPI_FLASH_AUTO_SUSPEND in [...]

Please note that, Auto Suspend is an optional feature. See https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html. The range of flash chips that you can use will be significantly smaller if you enable this feature. Hence you need to pay more attention to supply issues. Please contact our business team to make sure either you are using the correct Flash model, or Espressif is delivering modules to you with correct Flash models.

> And I see that this feature is based on the XMC chip, and XMC can also have a low-voltage restart CONFIG_SPI_FLASH_BROWNOUT_RESET_XMC option that other chips do not have can be configured, so is XMC more suitable for ESP32 to carry? What is the increased efficiency of Flash compared to Winbond?

It's the workflow recommended by XMC. So far we havn't receive such suggestion from other venders. I don't think it will make performance difference.

WangShuoran, 2023-06-15:
Thank you for your quickly reply!

I want to know the info is related to ESP32S3 and ESP32C3, the next is ESP32.

> SPI0 and SPI1 are used for communicating with external physical memory, including SPI Flash and PSRAM (on ESP32). These two SPIs are call MSPI as well. MSPI (SPI0, SPI1) are two different hardwares, [...]

Your send CS1 is for PSRAM, CS0 is for SPI Flash, the content is really samely with I send SPI0 connects to flash where the program is stored, SPI1 connects to PSRAM. And I need to confirm the just mentioned is for ESP32C3/ESP32S3. Is it correct?

> Accessing code placed in Flash is done by hardware (SPI0).

I can understand it.

> And we provide the [ESP FLASH driver](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/esp_flash.h) for you to access the Flash for other purpose (e.g. storage). This is via SPI1.

I can not understand it, because in the last only SPI0 use to connect Flash. Why the SPI1 can operation Flash?

> We provide [SPI FLASH MMAP driver](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/spi_flash_mmap.h) and [MMU MMAP [...]

I think this is similar with the frist your send. Or this is another thing?

>>使用SPI0/1使用外部Flash（NOR或NAND），初步挑选为W25Qxx和W25N01芯片；如果可以我们是不是使用SPI1进行初始化操作，有案例吗？
>Yes we're using SPI1 to do the Flash chip initialisation. This contains multiple stages, if you use the ESP-IDF project, these initialisations are done for you already.

I do not understand why, the question similar with the second your send(Why the SPI1 can operation Flash).

使用SPI0/1使用SPI主机功能驱动从机，这个功能和前面的功能分离使用，如果能叠加更好；如果可以我们是不是使用SPI0进行初始化操作，有案例吗？

> I would suggest you using SPI2/SPI3 as general purpose SPI usage. Yes on ESP32 you can use [SPI Master [...]

You send "Yes on ESP32 you can use [SPI Master driver](https://github.com/espressif/esp-idf/blob/master/components/driver/spi/include/driver/spi_master.h) for SPI1.". I can not understand why is SPI1? Because In the document SPI1 is not support:

[image]

> THis is wrong, accessing PSRAM is via the SPI0 hardware. On ESP32, there're two drivers that may use SPI1, [SPI Master [...]

So In the ESP32S3 the PSRAM is also via the SPI0 hardware. But what is the last sentence mean? Then content is not same with your last send.

Icarus113, 2023-06-19:
> Your send CS1 is for PSRAM, CS0 is for SPI Flash, the content is really samely with I send SPI0 connects to flash where the program is stored, SPI1 connects to PSRAM. And I need to confirm the just mentioned is for ESP32C3/ESP32S3. Is it correct?

Your statement about "the content is really samely with I send SPI0 connects to flash where the program is stored, SPI1 connects to PSRAM" is not entirely correct. Thanks to the hardware, we don't need to write a dedicated driver to SPI0 and we can still access the Flash and the PSRAM. With the help of the CS0 and CS1, we can multiplex the usage of the bus, and access the Flash / PSRAM respectively, at different time slice.

SPI0 / SPI1 are different peripherals and they share the bus, and **both of them** can access the Flash / PSRAM. However, we don't write a driver for SPI1 to access the PSRAM, we only provide a driver for SPI1 to access the Flash (for usages other than the XIP Flash, e.g. storage)

--- 

About the  [SPI FLASH MMAP driver](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/spi_flash_mmap.h), [MMU MMAP driver](https://github.com/espressif/esp-idf/blob/master/components/esp_mm/include/esp_mmu_map.h), by default the firmware (application) is mapped by the ESP-IDF system. By using these two drivers (actually also [partition_mmap driver](https://github.com/espressif/esp-idf/blob/master/components/esp_partition/include/esp_partition.h#L373)), you can map some extra stuff (e.g. you use the esp_flash driver (the one attached to SPI1) to store some data on Flash, and map them to certain addresses, and access these data via the addresses (pointers)).

> You send "Yes on ESP32 you can use [SPI Master driver](https://github.com/espressif/esp-idf/blob/master/components/driver/spi/include/driver/spi_master.h) for SPI1.". I can not understand why is SPI1? Because In the SPI Master document SPI1 is not support:

This support is only on ESP32, we don't support this on other chips.

> In the SPI Master, can I use SPI0, and you send SPI1 is incorrect?(focus on ESP32S3/ESP32C3)

This question is not very clear, you may rephrase this. If you're referring to using [SPI Master Driver](https://github.com/espressif/esp-idf/blob/master/components/driver/spi/include/driver/spi_master.h) to access the SPI, for ESP32S3/ESP32C3, the answer is no, we don't support this on these chips.

Other question I think I've answered them in above paragraphs, feel free to ask if there's still something you're not clear, or something that I didn't explain well.

Icarus113, 2023-06-29:
@WangShuoran  Hi, may I know if there's any updates? Or we can close this issue then.

Icarus113, 2023-07-04:
I'll close this issue as there's no more comments.

Feel free to reopen this issue and leave your questions/comments.  @WangShuoran

closed 2023-07-04 by Icarus113