espressif/esp-idf #11664: Remove binary blobs from .git when generating the zip archive

stkw0, 2023-06-13:
### Is your feature request related to a problem?

When downloading the esp-idf following the link in the archive it contains all the contents of .git directory. This directory alone contains ~800 MB of data, this is represents ~60% of the size of the zip file. 

### Describe the solution you'd like.

Remove useless .git contents. I guess that this directory is included to grab the current commit of esp-idf, but then at least I think .git/modules (contains 757.2 MB of data) could be removed?

igrr, 2023-06-20:
We do include the entire `.git` directory so that it is possible to switch between IDF versions, for instance to compare the behavior between releases or to do `git bisect`.

We could remove it, but then the instructions for switching between IDF versions in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-esp-idf would have to cover the case if the .git directory doesn't exist.

Could you please check if doing a shallow clone (`git clone --depth=1 --recurse-submodules --shallow-submodules ...`) would work for you instead of downloading the .zip file?

stkw0, 2023-06-20:
In some cases a git clone may work although there are different use cases for which I use he zip file:
- In CI/CD scripts to build the firmware and run some tests
- On my system I install esp-idf from the package manager (which downloads the zip file, in this case a git clone won't work)
- On some development machines

In some cases a git clone and a checkout to a tagged version may work, but in the cases it won't would be nice to reduce the size of the zip file (maybe by moving the `.git/modules` directory to a different zip file for debugging?). Although I don't know if there is much interest in that.

filipmnowak, 2024-12-29:
hello Ivan

> We do include the entire `.git` directory so that it is possible to switch between IDF versions, for instance to compare the behavior between releases or to do `git bisect`.

i am not insisting that it's a single best way of handling this issue, however in general it's common that software releases are size/format optimized for typical usage, what you are describing is related to `espressif/esp-idf` troubleshooting/development?

> We could remove it, but then the instructions for switching between IDF versions in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-esp-idf would have to cover the case if the .git directory doesn't exist.

perhaps easiest way to handle this issue, is to point user to GitHub releases, and just ask them to download and unpack version they want or latest stable release?

i feel like manipulating Git repo is probably more useful to `espressif/esp-idf` developers, than to folks using it to put their own code on Espressif SoC-powered device.

> Could you please check if doing a shallow clone (`git clone --depth=1 --recurse-submodules --shallow-submodules ...`) would work for you instead of downloading the .zip file?

that of course works very well:

| name/version | clone type      | size | clone time |
|-------------------|-------------------|-------|--------------|
| esp-idf-v5.3.2 | full | 2.2G | 3m8.277s |
| esp-idf-v5.3.2 | shallow | 578M | 1m25.328s |

stkw0, 2026-02-20:
Is there some update regarding this? Arch and Gentoo packages are affected by this and seems there are other related opened issues.