espressif/esp-idf #11666: ds18b20 example broken

su-Koch, 2023-06-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

includes available

### What is the actual behavior?

includes broken ds18b20.h 

### Steps to reproduce.

run example https://github.com/espressif/esp-idf/blob/213504238f77e01073f668e5e8f87e3b3cc02a8f/examples/peripherals/rmt/onewire/main/onewire_example_main.c#L11

suda-morris, 2023-06-14:
@su-Koch what do you mean by broken? Can't it compile on your side?

The DS18B20 is a managed component: https://components.espressif.com/components/espressif/ds18b20

It's imported in the example by https://github.com/espressif/esp-idf/blob/master/examples/peripherals/rmt/onewire/main/idf_component.yml#L2

For how to use the component manager, please read: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html

suda-morris, 2023-06-15:
I checked the example can run very well on my side. So close. Feel free to reopen if you meet other issue.

closed 2023-06-15 by suda-morris