espressif/esp-idf #11667: error:check_lock_nonzero locks.c:300 (lock != NULL && "Uninitialized lock used")

LHYYSY, 2023-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

start running

### What is the actual behavior?

repeated restart

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
assert failed: check_lock_nonzero locks.c:300 (lock != NULL && "Uninitialized lock used")
Core  0 register dump:
MEPC    : 0x40380926  RA      : 0x4038ae40  SP      : 0x3fcd1110  GP      : 0x3fc94e00
0x40380926: panic_abort at D:/ESP_IDF_5_0/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/panic.c:423

0x4038ae40: __ubsan_include at D:/ESP_IDF_5_0/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/ubsan.c:313

TP      : 0x3fc954a0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x0000006e  S1      : 0x00000001  A0      : 0x3fcd114c  A1      : 0x3fc96af9
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fca2000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fcd1279
S4      : 0x3fc96af8  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
3fcd1110: 0xa5a5a5a5 0xa5a5a5a5 0x3c11344c 0x40391ae4 0x3fc96c80 0x3c11344c 0x3fc96ffc 0x3c11335c
0x40391ae4: newlib_include_assert_impl at D:/ESP_IDF_5_0/Espressif/frameworks/esp-idf-v5.0.1/components/newlib/assert.c:90

3fcd1130: 0x3fc96c90 0x3fcd1144 0x3fc96c94 0x3c113318 0x3fc96af8 0x00303033 0xa5a5a5a5 0x65737361
3fcd1150: 0x66207472 0x656c6961 0x63203a64 0x6b636568 0x636f6c5f 0x6f6e5f6b 0x72657a6e 0x6f6c206f
3fcd1170: 0x2e736b63 0x30333a63 0x6c282030 0x206b636f 0x4e203d21 0x204c4c55 0x22202626 0x6e696e55
3fcd1190: 0x61697469 0x657a696c 0x6f6c2064 0x75206b63 0x22646573 0xa5a50029 0xa5a5a5a5 0xa5a5a5a5
3fcd11b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd11d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd11f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1210: 0xa5a5a5a5 0x00000000 0x00000000 0xa5000000 0xa5a5a5a5 0xa5a5a5a5 0x3fc9816c 0x3fca5fc8
3fcd1230: 0x3c115900 0x3c115870 0x00000000 0x40381a26 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x40381caa
0x40381a26: lock_acquire_generic at D:/ESP_IDF_5_0/Espressif/frameworks/esp-idf-v5.0.1/components/newlib/locks.c:111

0x40381caa: __retarget_lock_acquire_recursive at D:/ESP_IDF_5_0/Espressif/frameworks/esp-idf-v5.0.1/components/newlib/locks.c:313

3fcd1250: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0x00000003 0x420fd9dc
0x420fd9dc: _vfprintf_r at ??:?

3fcd1270: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1290: 0xa5a5a5a5 0xa5a5a5a5 0x00000001 0xa5a5a5a5 0xa5a5a5a5 0x3c1486b8 0xa5a5a5a5 0xa5a5a5a5
3fcd12b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd12d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd12f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3fcd16ec 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1310: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1330: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1350: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1370: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1390: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd13b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd13d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd13f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1410: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1430: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1450: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1470: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd1490: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd14b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd14d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd14f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 0000000000000000
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 2c54834c44241548f2906a78ed654226cbea6fd0c920055cb600f75379993d5f
```

closed 2023-07-17 by LHYYSY

ferm10n, 2023-07-23:
What causes this error to occur? How was it resolved here?

zikalino, 2023-10-02:
I don't have enough information to find out why this kind of problem could occur.

@ferm10n if you are encountering similar problem, please feel free to reopen this ticket and add some more details on your scenario.
@LHYYSY we would appreciate if you could add some more details on the resolution of the problem.