espressif/esp-idf #11668: Does LEDC share the channels between different timers?

Kampi, 2023-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have an issue with my camera and flash application on an ESP32 platform. If I use the following configuration for the LEDC for the flashlight and the camera my code runs without an issue:

```
static ledc_timer_config_t _Camera_Flash_LEDC_Timer = {
    .speed_mode       = LEDC_LOW_SPEED_MODE,
    .duty_resolution  = LEDC_TIMER_8_BIT,
    .timer_num        = LEDC_TIMER_2,
    .freq_hz          = 10000,
    .clk_cfg          = LEDC_AUTO_CLK
};

static ledc_channel_config_t _Camera_Flash_LEDC_Channel_White = {
    .gpio_num       = PIN_LED_WHITE_FLASH,
    .speed_mode     = _Camera_Flash_LEDC_Timer.speed_mode,
    .channel        = LEDC_CHANNEL_1,
    .intr_type      = LEDC_INTR_DISABLE,
    .timer_sel      = _Camera_Flash_LEDC_Timer.timer_num,
    .duty           = 0,
    .hpoint         = 0,
    .flags          = 0,
};

...

    camera_config_t CameraConfig;

    CameraConfig.ledc_channel = LEDC_CHANNEL_0;
    CameraConfig.ledc_timer = LEDC_TIMER_1;
    CameraConfig.pin_d0 = PIN_Y2_GPIO_NUM;
    ...
 ```
 
 But if I change the configuration for the LEDC of the white flash to `LEDC_CHANNEL_0` and `LEDC_TIMER_2` the camera isn´t able to capture an image.
 
 Why does this happen? My understanding is that each channel of each timer is independent, but it looks like channel 0 is connected between the timers.
 
 Can someone explain it to me, please?

wuyuanyi135, 2023-06-14:
[image]

According to esp32s3 TRM pg. 1271, the timers are mutiplexed to the ledc channels. So I believe you should not use the same channel for different timers.

Kampi, 2023-06-14:
Hi @wuyuanyi135,

thank you! That explains my issue and confirmed my thoughts.