espressif/esp-idf #11669: When runing http-server with async send frame i get error in send: 11

Mowglli, 2023-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.3-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

I expected it to run with no issues. Right now it is hosted via USB-C, configured as RNDIS device. I have also tried WIFI, where i get the same error.

### What is the actual behavior?

It can sometimes run for hours, before the errors comes. After the errors have occured, the socket is closed, and i can not get a new handshake.

I send data from the websocket every 25 ms, 
and a async from the http-server every 25 ms also. 

### Steps to reproduce.

1. Configure ESP32-S3 to RNDIS Device. with a static ip
2. Configure http-server  
3. Configure async message function : 

```
void ws_async_send_struct(const WebSocketMessage_t* msg) {
	websock_instance *dev = NULL;
	while ((dev = ldeCore_ItemNext(socks, dev)) != NULL) {
		if (!dev->delete) {
			httpd_ws_frame_t ws_frame = {
					.payload = (uint8_t *)msg,
					.len = sizeof(WebSocketMessage_t),
					.type = HTTPD_WS_TYPE_BINARY,
					.final = true,
					.fragmented = false,
			};

			//if (httpd_ws_get_fd_info(dev->handle, dev->descriptor)	== HTTPD_WS_CLIENT_WEBSOCKET) {

			if(httpd_ws_send_frame_async(dev->handle, dev->descriptor,&ws_frame) != ESP_OK){
				dev->delete = 1;
			}
			//}
		}
	}
}
```
4. Create a service task that control the async message type:
```
void homePage_task(void *pvParameters) {

	list_spiffs_files();
	msg.pct = 0;
	msg.size = 0;
	msg.cold = 0;
	while (1) {
		vTaskDelay(25);

		if (1 ) {

			uint32_t t = esp_timer_get_time() / 1000;

			strcpy(msg.builddate, __DATE__);
			strcpy(msg.buildtime, __TIME__);
			msg.time = t;

			websock_instance *dev = NULL, *delDev = NULL;
			ws_async_send_struct(&msg);

			//Some other code is based here aswell
	}
}
```

### Debug Logs.

```plain
W (69912) httpd_txrx: httpd_sock_err: error in send : 11
W (69912) httpd_ws: httpd_ws_send_frame_async: Failed to send WS payload
W (75413) httpd_txrx: httpd_sock_err: error in send : 11
W (75413) httpd_ws: httpd_ws_send_frame_async: Failed to send WS payload
W (75628) httpd_txrx: httpd_sock_err: error in recv : 104
W (77825) httpd_txrx: httpd_sock_err: error in recv : 104
W (1232860) httpd_txrx: httpd_sock_err: error in recv : 113
W (1232860) httpd_ws: httpd_ws_get_frame_type: Failed to read header byte (socket FD invalid), closing socket now
W (1232868) httpd_txrx: httpd_sock_err: error in recv : 113
W (1232872) httpd_ws: httpd_ws_get_frame_type: Failed to read header byte (socket FD invalid), closing socket now
```

ESP-YJM, 2023-06-21:
The errno 11 is due to the peer not ack in specified time. And then the peer send RST to the device. Could you please help capture the packet between the websocket client and server.

Mowglli, 2023-06-22:
Hello, Here is the capture of when it happens:
[image]
I have also attached the full capture.

The async send is called in a task every 1ms for testing, in a real use case i would like it to be atleast send with around 25hz.
Right now the async is as follows: 
```
#define max_clients 10

void ws_async_send_struct(const WebSocketMessage_t* msg) {
	websock_instance *dev = NULL;
	//while ((dev = ldeCore_ItemNext(socks, dev)) != NULL) {

			httpd_ws_frame_t ws_frame = {
					.payload = (uint8_t *)msg,
					.len = sizeof(WebSocketMessage_t),
					.type = HTTPD_WS_TYPE_BINARY,
					.final = true,
					.fragmented = false,
			};
			size_t clients = max_clients;
			       int    client_fds[max_clients];
			       if (httpd_get_client_list(server, &clients, client_fds) == ESP_OK) {
			           for (size_t i=0; i < clients; ++i) {
			               int sock = client_fds[i];
			               if (httpd_ws_get_fd_info(server, sock) == HTTPD_WS_CLIENT_WEBSOCKET) {
			                   //ESP_LOGI(TAG, "Active client (fd=%d) -> sending async message", sock);
			                   struct async_resp_arg *resp_arg = malloc(sizeof(struct async_resp_arg));
			                   resp_arg->hd = server;
			                   resp_arg->fd = sock;
			                   if(httpd_ws_send_data_async(resp_arg->hd, resp_arg->fd, &ws_frame, my_transfer_complete_cb, NULL) != ESP_OK){
			                   				ESP_LOGI("","nw");
			                   //				dev->delete = 1;
			                   				ready = 0;
			                   			}
			                   if(resp_arg) free(resp_arg);

			               }
			           }
			       } else {
			           ESP_LOGE(TAG, "httpd_get_client_list failed!");
			           return;
			       }
//			if(httpd_ws_send_data_async(dev->handle, dev->descriptor, &ws_frame, my_transfer_complete_cb, NULL) != ESP_OK){
//				ESP_LOGI("","nw");
//				dev->delete = 1;
//			}
			if(++tempD % 1000 == 0){
				ESP_LOGI("","Send");
			}
			//}
	//}
}
```

Mowglli, 2023-06-22:
Another capture here: 
[image]

ESP-YJM, 2023-06-25:
Do `2.2.2.3` is the client and `2.2.2.2` is server? Could you please share the source capture file?

Mowglli, 2023-06-25:
Yes exactly, i have attached the capture
[error11.zip](https://github.com/espressif/esp-idf/files/11860320/error11.zip)

ESP-YJM, 2023-06-27:
@Mowglli I didn't see any unusual packet in your attached packet. It is ok for websocket TCP flow. If the packet is only for errno 11, i think it is a normal issue that upper send data too fast. It will not broke the TCP connection.

Mowglli, 2023-06-27:
@ESP-YJM 
The problem is that it Aldi haloens if i slog Down the send speed. When it ocurs, there is no Way back, i have to unplug device and reconnect to the webserver. Refresh and opening of a Mew Windows is not enough

ESP-YJM, 2023-06-29:
Could you share the packet when you refresh or opening of a new window. I want to check whether the server response or not.
Also i think you can try this config to enable so_linger https://github.com/espressif/esp-idf/blob/v4.4.3/components/esp_http_server/include/esp_http_server.h#L190 and set linger_timeout to 0. Besides, you still need merge keep-alive (https://github.com/espressif/esp-idf/commit/311b50498b52d3072361466899a29384507fd31e) to the esp_http_server. The http server need use TCP keep-alive to detect abnormal connection and to close it and use so_linger to recycle the resource immediately.

Mowglli, 2023-06-29:
I have hereby attach a wireshark capture of, when the error occurs and when i try to open a window and connect to the webserver.
[refresh_window_with error.zip](https://github.com/espressif/esp-idf/files/11905458/refresh_window_with.error.zip)
I will try enable so_linger as well as the timeout and the merging of keep alive.

ESP-YJM, 2023-06-30:
@Mowglli If I understand correctly, when you re-open the web page on the client, the client will re-establish a TCP connection with the server. At this time, the client should connect to port 80 of the server, but according to the packet capture, the client has been making connection requests with port 53 of the server. This is problematic, and port 53 is used to monitor DNS services.
[image]

And in following picture, the server send TCP keep-alive and the client with no response, so the server will send RST to close TCP connection(Due to enable SO_LINGER). 
[image]

Mowglli, 2023-06-30:
Its because i have enabled mdns as well. So the webserver Can be accessed via A1d.local

Venlig hilsen
Jesper Egsgaard Nielsen

Mowglli, 2023-08-08:
Hello again, i have looked at it again, and made a minor change in the websocket client to the following, to ensure port 80: 
ws = new WebSocket('ws://' + location.hostname + ":80" + '/ws');
Right now im sending the async every 40 hz(25 ms). 

i have made an other capture in wireshark, which i have attached. 
[capture_http.zip](https://github.com/espressif/esp-idf/files/12288793/capture_http.zip)

ESP-YJM, 2023-08-09:
@Mowglli I not see any unusual  packet.Could you please describe your issue?

ESP-YJM, 2023-10-23:
@Mowglli Any update for this issue? If not, i will close it and if you have any update, feel free to re-open it. Thanks.

closed 2023-10-23 by ESP-YJM

Pr47h4m, 2025-03-15:
Hi everyone,

I am encountering the same issue as described in this thread, with the following errors logged:

```
httpd_txrx: httpd_sock_err: error in recv : 104  
httpd_ws: httpd_ws_send_frame_async: Failed to send WS payload  
```

## Environment Details:
Hardware: ESP32-S3
Framework: ESP-IDF v5.3.1
Application: WebSocket server

## Problem Description:
My ESP32-S3 is connected to a sensor that generates data frames at a high frequency. Each frame is 255 bytes, and the sensor operates at a frequency of 1000 Hz (i.e., 1000 frames per second). To reduce the transmission frequency, I have implemented a solution where I combine two consecutive sensor messages into a single frame, effectively halving the frequency to 500 Hz before sending the data over the WebSocket connection.

However, despite this approach, I am still encountering the aforementioned errors, and the WebSocket connection disconnects intermittently.

## Current Exploration:
I am exploring alternative network layer protocols, such as TCP or UDP sockets, to handle the real-time transmission of data to connected clients more reliably. My goal is to ensure stable and efficient data transfer given the high data rate and frame size.

## Request for Assistance:
If anyone has encountered a similar issue or has suggestions for improving my approach, I would greatly appreciate your insights. Specifically, I am looking for:

Recommendations on whether WebSockets are suitable for this high-frequency, real-time data transfer use case, or if TCP/UDP sockets would be more appropriate.
Best practices or optimizations for handling high-speed data streams on the ESP32-S3 within the ESP-IDF framework.
Any potential configurations or adjustments to the WebSocket server implementation that could mitigate the disconnect issue (e.g., buffer sizes, timeout settings, etc.).

Thank you in advance for your assistance and suggestions!

Best regards,
pr47h4m