espressif/esp-idf #11672: Core  0 panic'ed (Cache disabled but cached memory region accessed) with HttpsOTAUpdate

0x0fe, 2023-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32DWDR2

### Power Supply used.

USB

### What is the expected behavior?

Being able to perform the OTA.

### What is the actual behavior?

Crashes 

### Steps to reproduce.

Start A2DP sink (no need to connect a device to it)
Start an MQTT secured connection
Starts HttpsOTAUpdate

### Debug Logs.

```plain
link up
IP: 192.168.1.81
mac: xxxxxxxxxxxxxxxxx
uuid: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Firmware Version: 0x92 = v2.0
ts: 421683993
key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
connecting mqtt
[  2362][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 117768
[  2362][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[  2422][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator
[  2423][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS structure...
[  2429][V][ssl_client.cpp:178] start_ssl_client(): Loading CA cert
[  2447][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...
[  2448][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...
[  3433][V][ssl_client.cpp:290] start_ssl_client(): Verifying peer X.509 certificate...
[  3433][V][ssl_client.cpp:298] start_ssl_client(): Certificate verified.
[  3438][V][ssl_client.cpp:313] start_ssl_client(): Free internal heap after TLS 117284
[  3446][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 74 bytes...
connected
sub to XXXXXX/5443b26ed0a8/XXXXXXX/DOWNLOAD/+
[  3525][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 47 bytes...
sub to XXXXXX/5443b26ed0a8/XXXXXXX/UPDATE
[... 1678 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT is not set
# CONFIG_SUPPORT_TERMIOS is not set
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

```

0x0fe, 2023-06-14:
and so to make sure the problem actually comes from my sdkconfig, i made a test firmware with arduino, where the sdkconfig has less optimisation, and sure enough, the OTA works here, it could perform the OTA and reboot.

```
Stack Space: 7260
Used PSRAM: 0
init_sd
Type: SDHC
Total: 7678MB
Used: 128MB
WiFi init
.
link up
IP: 192.168.1.81
fw rev 1.0.0
Stack Space: 6396
Used PSRAM: 29336
mac: 5443b26ed0a8
uuid: c3e11dd29084d45c09cfa75650b1778e
ts: 421686965
key: ibola5Xe6WKm3NX+wxmxJAzsb2oUQVDSO08gMXetDW4=
connecting mqtt
connected
sub to XXXXXXXX/5443b26ed0a8/XXXXXXX/DOWNLOAD/+
sub to XXXXXXXX/5443b26ed0a8/XXXXXXX/UPDATE
Starting OTA
Http Event On Connected
Http Event Header Sent
Http Event On Header, key=Content-Length, value=1319440
Http Event On Header, key=Content-Type, value=application/octet-stream
Http Event On Header, key=Server, value=Microsoft-HTTPAPI/2.0
Http Event On Header, key=api-supported-versions, value=1.0
Http Event On Header, key=Content-Disposition, value=attachment; filename=firmware.bin; filename*=UTF-8''firmware.bin
Http Event On Header, key=Strict-Transport-Security, value=max-age=15724800; includeSubDomains
Http Event On Header, key=Date, value=Wed, 14 Jun 2023 13:04:26 GMT
Http Event Disconnected
Http Event Disconnected
OTA success, rebooting
[ 27917][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
ets Jul 29 2019 12:21:46
```

0x0fe, 2023-06-14:
i tried to unset 

```
# CONFIG_SPI_MASTER_ISR_IN_IRAM is not set
# CONFIG_SPI_SLAVE_ISR_IN_IRAM is not set
```

But it did not help. I must precise i have the following flags set (between others)

```
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_ESP32_ECO3_CACHE_LOCK_FIX=y
CONFIG_ESP32_REV_MIN_3=y
CONFIG_ESP32_REV_MIN=3 
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY=y
CONFIG_ESP_EVENT_POST_FROM_ISR=y
CONFIG_ESP_EVENT_POST_FROM_IRAM_ISR=y
CONFIG_ESP_SYSTEM_RTC_EXT_XTAL=y
CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH=y
CONFIG_FREERTOS_ENABLE_TASK_SNAPSHOT=y
CONFIG_FREERTOS_PLACE_SNAPSHOT_FUNS_INTO_FLASH=y
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN=y
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=16384
CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN=4096
CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
CONFIG_MBEDTLS_DYNAMIC_FREE_PEER_CERT=y
```

0x0fe, 2023-06-14:
when i modified the  heap caps malloc to 4096

heap_caps_malloc_extmem_enable(4096);

The OTA suceeded, however this is problematic because seting the heap cap malloc is what helped most to keep enough free IRAM to be able to run A2DP sink, HTTPS, MQTTS and MP3 decoding/I2S working together. With 4096 i think it wont be possible and thus i think something shoudl be changed in the HTTPS OTA to eb able to run normally when heap caps malloc is lower than 4096.

chegewara, 2023-06-14:
It is not easy to handle so many memory consuming features on embedded system with limited ram.
Maybe you should consider to disable mqtt, mp3 or a2dp during OTA?

0x0fe, 2023-06-14:
@chegewara 
Yes, it is close to impossible. 
Sadly we cannot release the memory of A2DP sink https://github.com/espressif/esp-idf/issues/11642
(I have to admit i dont understand the logic behind this).

> Maybe you should consider to disable mqtt, mp3 or a2dp during OTA?

This is what i tried, to no avail, i suspect disconnecting MQTT does not release ressources.

```
      mqtt.disconnect();
      a2dp_sink.end(true);
```

0x0fe, 2023-06-14:
I am evaluating the code since several days and performed tens of memory map analysis, actually. 
A2DP cannot free ressources, this is a flaw of the SDK and so far there is nothing i can do about it, if i free the ressources it cannot be restarted, this is documented.
Regarding the other tasks, MQTT does not run on a separate task, it runs in the idle task which also performs other jobs like the adkeys handling and some other event driver actions. The file downlaod task for example is started when needed and stopped at the end of the download. The audio playback is currently in the idle task. So there is no task overhead. You know, simply using HTTPS and A2DP consumes most of the IRAM with default SDKconfig settings. They are the biggest consumers by very far.

idea--list, 2023-06-16:
@0x0fe 
OTA takes quite some memory and it can not be released without stopping the service, however it needs to be running after the system has started to do the FW verification. In my case i can just simply stop that at a specific event.
Despite that i was still facing the symptom you describe. In my case the solution was that i simply restart the ESP32 regularly when there is no user activity so it does not affect anything in my case. Not sure if this would fit your case.

0x0fe, 2023-06-16:
@idea--list 
this has been solved by keeping the heap caps malloc to 4096, and stripping further things in the sdkconfig. 
I cannot "restart the system regularly" on a production firmware. Are you sure the OTA is working without releasing its ressource after system reset? If so it is lauched on its own, because on my side there is nothing starting any OTA function after a reset, the OTA is started manually when it needed.

closed 2023-07-06 by Alvin1Zhang