espressif/esp-idf #11673: Issue with esp_tls_conn_http_new_sync()

bmkuter, 2023-06-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Are there any known issues with `esp_tls_conn_http_new_sync`? I am using it per the example code, and it keeps returning -1 when I use Amazon's most recent PEM on `https://www.howsmyssl.com/a/check`. I am connected to the internet on the esp at this point I believe, since the local logging I use is over UDP. 

Any ideas?

Here is my example skeleton version of the function:
```
#define WEB_URL "https://www.howsmyssl.com/a/check"
const char **alpn_protos = {"h2", NULL };

void tls_connection(void *parameters)
{
    BaseType_t          result;
    unsigned char       local_rx_buffer[UART_BUF_SIZE];
    uint32_t            actual_payload_size;
    uint32_t            temp_uin32t = 0;
    int                 temp_int = 0;
    
    vTaskDelay(5000/portTICK_PERIOD_MS);

    // Init TLS subsystem
    esp_tls_t *tls = esp_tls_init();
    udp_print("esp_tls_init done: 0x%x", tls);

    // Connect to new TLS
    esp_tls_cfg_t cfg = {
        .cacert_buf = amazon_root_PEM,
        .cacert_bytes = sizeof(amazon_root_PEM)
        // .clientcert_buf = public_key_PEM,
        // .clientcert_bytes = strlen((char *)public_key_PEM),
        // .clientkey_buf = private_key_PEM,
        // .clientkey_bytes = strlen((char *)private_key_PEM)
    };

    temp_int = esp_tls_conn_http_new_sync(WEB_URL, &cfg, tls); //aws
    udp_print("esp_tls_conn_http_new_sync: %d", temp_int);

    vTaskSuspend( NULL );

    while (1)
    {
        // Read TLS
        ssize_t  ret_size = esp_tls_conn_read(tls, local_rx_buffer, sizeof(local_rx_buffer));
        udp_print("Recv TLS data of size %d", ret_size);

        // Send to K70 for processing
        deposit_uart0_tx_message('K', local_rx_buffer, ret_size);

        memset(local_rx_buffer, 0, UART_BUF_SIZE);
        if ( (result = xQueueReceive(tls_response_queue, (void *)&(local_rx_buffer), 5000/portTICK_PERIOD_MS ) == pdTRUE) ) // Wait for 5000 ms before timing out. 
        {
            // uint64_t send_to_queue_end = esp_timer_get_time();
            xQueueReceive(tls_response_size_queue, (void *)&temp_uin32t, 500/portTICK_PERIOD_MS );
            
            // Respond to TLS
            esp_tls_conn_write(tls, local_rx_buffer, temp_uin32t);
        }
        else if (result == pdFALSE)
        {
            // Cleanup?
            udp_print("Response from Cyclone timed out!");
        }
    }

    // Free resources
    esp_tls_free_global_ca_store();
    esp_tls_conn_destroy(tls);
}
```

AdityaHPatwardhan, 2023-06-15:
The API returning `-1` means there is failure in the connection.
I don't understand why you are using Amazon root CA for `howsmyssl.com`, it should be the root certificate for `howsmyssl.com` (i.e. the ca of the target endpoint that you want to connect).

Just to get more information about the error, Can you try getting the last error obtained in esp-tls.
It can be done with calling
https://github.com/espressif/esp-idf/blob/17d6768e65b6e8d8ef040e20d888bd6d573bc3cb/components/esp-tls/esp_tls.h#L637

and then 

https://github.com/espressif/esp-idf/blob/17d6768e65b6e8d8ef040e20d888bd6d573bc3cb/components/esp-tls/esp_tls.h#L609

bmkuter, 2023-06-17:
Ah I misunderstood how to handle the keys. Thanks!

closed 2023-06-20 by bmkuter